MarlinFirmware/Marlin #6706: This doesn't explain DISTINCT_E_FACTORS

ruggb, 2017-05-12:
I think this explanation leaves a little to be desired.
/**
 * With this option each E stepper can have its own factors for the
 * following movement settings. If fewer factors are given than the
 * total number of extruders, the last value applies to the rest.
 */
//#define DISTINCT_E_FACTORS

I.E. why would I need to uncomment that??
Are you trying to say that simply adding those factors does not yield that result?
This line
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]

and what you wrote above implies that it is automatic if you add those factors.
You may want to make this a little clearer.

tcm0116, 2017-05-15:
What it's trying to say is that if DISTINCT_E_FACTORS is not enabled, then Marlin will use the E0 factor for all extruders. If DISTINCT_E_FACTORS is enabled, then it will use the E0 factor the extruder 0, E1 for extruder 1, and so on. However, if you have more extruders than E factors, then the last E factor defined will apply to that extruder and the remainder. For example, if you have 4 extruders, but have only defined E0 and E1, then E0 will apply to extruder 0, and E1 will apply to extruders 1, 2, and 3.

ruggb, 2017-05-15:
@tcm0116 THAT is a great explanation.
 How about replacing the current explanation with the  following?
/*
 *if DISTINCT_E_FACTORS is enabled, Marlin will use the E0, E1, E2, etc. factors for E0, E1, E2, etc. extruders.
 * If the number of factors does not equal the number of extruders, the last factor will be used for the remaining extruders.
 * If DISTINCT_E_FACTORS is not enabled, then Marlin will use the E0 factor for all the extruders regardless of the number of factors listed.
*/

Bob-the-Kuhn, 2017-05-15:
There's now a check to ensure that the number of factors is not too large.    It doesn't check if there are too few.

ruggb, 2017-05-15:
OK, make this change
```
/*
 * If DISTINCT_E_FACTORS is enabled, Marlin will use the E0, E1, E2, etc. factors for E0, E1, E2, etc. extruders.
 * If the number of factors **is less than** the number of extruders, the last factor will be used for the remaining extruders.
 * If DISTINCT_E_FACTORS is not enabled, then Marlin will use the E0 factor for all the extruders regardless of the number of factors listed.
*/
```

AirRaidX, 2017-05-17:
I have yet to get this feture to work, anytime i enable it the printhead moves off the platform when it switches to second extruder. If i disable it everything works fine when printing multiple colors through the diamon head. So im stuck with only using 2 colors at the moment due to the extruder configuration.

thinkyhead, 2018-02-06:
Presuming your issue is fixed, @AirRaidX. And the description isn't so bad, really.

closed 2018-02-06 by thinkyhead