MarlinFirmware/Marlin #6708: Another serious bug with bed level

batata004, 2017-05-12:
Hi,

Do this in your printer: start printing something and after 1 minute of printing time click STOP PRINT. Wait 30 seconds and go do PREPARE -> BED LEVEL -> BED LEVEL.

The printer will start bed leveling BUT in a very weird position, it starts bed leveling right in the position it is and keep moving around to probe the GRID (I am using linear bed leveling with servo) in weird spots, some even outside the print area!

I realized that if turn on your printer and you go PREPARE -> BED LEVEL -> BED LEVEL IT WILL HOME FIRST AND THEN, ONLY AFTER THAT, IT WILL START bed leveling.

BUT if you are in the middle of a print and you interrupt it (STOP) and try to bed level again some very dangerous behaviour will happen. In my case the servo deployed outside the printing area and because of that the nozzle almost broke que glass of my heated surface. This glass looks like steel cause so many times I almost broke it, but this time it was a lot but luckly the glass didnt break.

Tannoo, 2017-05-13:
Sounds like there is an issue.

Note: I usually hit the kill button if the nozzle comes too close to the bed while doing any probing.

You don't?

batata004, 2017-05-13:
I usually Trust electronics I dont use to lay around watching If it's
working correctly ALL the time  . Any way I think the BED LEVEL procedure
should always be preceded by HOMING dont you think?

Tannoo, 2017-05-13:
Normally, yes. But, not everyone's setup works the same.

Tannoo, 2017-05-13:
Well, the program is not usually the cause of crashes. If I envoke something via the lcd, I am going to watch it. I'm also watching most of the prints I do. That is only because it is a machine. Anything can happen. I may not beable to save a print, but I can usually save the printer from fubaring itself.

batata004, 2017-05-13:
I would like to know why this question was marked as a support question and not as a bug? To me it's a disaster waiting to happen if you dont add some conditional in your code of FIRST HOME if the user still didnt home when trying to BED LEVEL.

Please do as I said in the first post, test it yourself. You will see that clearly there is a problem!

mtowara, 2017-05-14:
Which kind of leveling do you use? Roxys tag says bilinear but you dont mention this.

Homing might not be necessary if the printer still knows where it is. Invalidating that data might still be a good idea because the printer might have skipped steps during the print.

Does the same thing happen when you dont use the lcd but trigger G29 manually?

Tannoo, 2017-05-14:
>Please do as I said in the first post, test it yourself. You will see that clearly there is a problem!

How about fixing it and submitting a PR?

Tannoo, 2017-05-14:
>Which kind of leveling do you use? Roxys tag says bilinear but you dont mention this.

Well, it could be mesh, linear, or bilinear. Those are the only ones that have "level bed" in the prepare menu.

Tannoo, 2017-05-14:
>(I am using linear bed leveling with servo)

mtowara, 2017-05-14:
oh, reading certainly helps :)
I looked into it yesterday night a bit but I find the heap of code in ultralcd.cpp quite confusing.
But if its actually linear all that stuff should be irrelevant as the LCD just issues an G29 command.

mtowara, 2017-05-14:
I can reproduce. For me it moves to the right positions, but z-height is messed up (probe does not move up before probing). Also after aborting the steppers are of, so the axis positions are not reliable.

G29 according to the code should fail when the axis is not homed.
However Marlin axis_unhomed_error will return false when the axis has been homed but the position of the axis is not known.
@thinkyhead would it be wise to excchange

> bool axis_unhomed_error(const bool x, const bool y, const bool z) {
>     const bool xx = x && !axis_homed[X_AXIS],
>                yy = y && !axis_homed[Y_AXIS],
>                zz = z && !axis_homed[Z_AXIS];

with

> bool axis_unhomed_error(const bool x, const bool y, const bool z) {
>     const bool xx = x && !axis_known_position[X_AXIS],
>                yy = y && !axis_known_position[Y_AXIS],
>                zz = z && !axis_known_position[Z_AXIS];

or might there be unintended side effects?

batata004, 2017-05-14:
I am using bilinear but also happens with linear. Sometimes it loses track of Z and more frequently with X and Y. You just need to abort a print and go do BED LEVEL -> BED LEVEL. You will see the effect right away.

EVEN if if off for just a bit it's still a bug cause when calculatin the mesh (using BILINEAR) it will certainly calculate the "rectangle surfaces" in the wrong spots.

thinkyhead, 2017-05-14:
> `G29` according to the code should fail when the axis is not homed.

I agree it should also make sure steppers haven't gone to sleep, since position is unknown. I will add that constraint. I'll check the other calls to `axis_unhomed_error` and make sure they're okay with the change — or add a bool parameter for those that want "known" as well.

closed 2017-05-15 by thinkyhead