MarlinFirmware/Marlin #6709: Bed leveling Z_MIN_PROBE_ENDSTOP

ruggb, 2017-05-12:
I am confused about this - I have been trying to create a suggested rewrite of it, but it is that confusing that I cannot.

This should be moved from the endstop section to a logical place under Z-Probe Options
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.
=======================================
/**
 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

/**
 * Z_MIN_PROBE_ENDSTOP
 *
 * Enable this option for a probe connected to any pin except Z-Min.
 * (By default Marlin assumes the Z-Max endstop pin.)
 * To use a custom Z Probe pin, set Z_MIN_PROBE_PIN below.
 *
 *  - The simplest option is to use a free endstop connector.
 *  - Use 5V for powered (usually inductive) sensors.
 *
 *  - RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
 *    - For simple switches connect...
 *      - normally-closed switches to GND and D32.
 *      - normally-open switches to 5V and D32.
 *
 * WARNING: Setting the wrong pin may have unexpected and potentially
 * disastrous consequences. Use with caution and do your homework.
 *
 */
#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN Z_MAX_PIN
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe. == proposed move
========================================

The confusing points starting with:
// See http://marlinfw.org/configuration/probes.html
  ==== cannot be reached

1. Z_MIN_PROBE_ENDSTOP - would read a lot easier and less confusing if it were Z_PROBE_MIN_ENDSTOP
2. #define Z_MIN_PROBE_ENDSTOP_INVERTING false - currently in an illogical place
3. (By default Marlin....) is unclear if that is because you have defined it at the bottom or it is coded elsewhere
4. There are 3 options = MIN, MAX, & Custom but since it is scattered and disconnected, I can't figure out what I need to change.
5. The Ramps 1.4 board has no D32 on Aux4. Z_MAX_ENDSTOP is on D19
6. Talking about using 5v, GND, and Aux4->D32 makes no sense. For Z_MAX_ENDSTOP on Ramps you would use pins 1 & 2 of ZMAX connector.
7. Interpreting the "simple switches connect" is not so simple.

I have a simple NO nozzle-based probe switch. I have the pullup on ZMAX enabled. Connecting the probe to 5v and D(whatever) doesn't work.

If you can explain to me what you mean by all this, I will do a rewrite for you.
#define Z_MIN_PROBE_ENDSTOP == Can we change this to Z_PROBE_MIN_ENDSTOP? Why is this enabled? Should it be by default or only if you are using this option?

#define Z_MIN_PROBE_PIN Z_MAX_PIN == is this the default you talk about? Can we add a note to change the pin to whatever custom you need.

#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe == better in this section

I can rewrite the connect section if you agree there is no D32 on Aux4

Roxy-3D, 2017-05-12:
> Z_MAX_PIN

This is the pin defined for your Mother Board Electronics so that the firmware can watch for the Z-Max Endstop getting pressed.

> Z_MIN_PROBE_ENDSTOP_INVERTING

If your Z-Probe is connected to the Z-Min pin of your electronics,  this controls whether the signal needs to be inverted to be used.

ruggb, 2017-05-13:
@Roxy-3D re-reading your answer adds even more confusion - this should be fixed
"Z_MIN_PROBE_ENDSTOP_INVERTING
If your Z-Probe is connected to the Z-Min pin of your electronics, this controls whether the signal needs to be inverted to be used."
It appears that the title has even confused you. I am interpreting that name as == invert whatever input is the Z_PROBE input. What you said is = if the Z_PROBE is connected to ZMIN invert it. == which would not really make any sense.

Roxy-3D, 2017-05-13:
Well...   It does get confusing.   The Arduino has pull up resistors that are typically enabled for an endstop switch.   One end of the switch is connected to Ground.   So if the switch is 'Triggered', it will short the pin to 0 volts.   And when that pin is read, it needs to be inverted to get the state of the switch being triggered.

ruggb, 2017-05-13:
@Roxy-3D  I understand all of that, but you are missing the big picture. I believe that  Z_MIN_PROBE_ENDSTOP_INVERTING does NOT refer to Z_MIN_ENDSTOP. It refers to Z_PROBE_MIN_ENDSTOP => which you are calling Z_MIN_PROBE_ENDSTOP_INVERTING - and that is even confusing to you.
If you can address my 7 points, maybe I can understand it and rewrite it so other users will also.

closed 2018-02-06 by thinkyhead