MarlinFirmware/Marlin #6711: About auto bed leveling calculate with unprobed point extrapolate

Jolly2015, 2017-05-13:
I had a delta printer and I used a mechanical switch under the nozzle as auto bed leveling probe.
I believe the algorithm was no problem ,but some result was not match to actual，does those data should be filtering?
I attachment the probe data below(marlin rc8 version),
<           0            1           2         3        4          5         6          7           8
<  0 +131.38 +157.77 -27.98 -27.78 -27.69 -27.50 -27.39 +131.08 +127.17
<  1 +130.98 -28.14 -28.03 -27.90 -27.74 -27.55 -27.35 -27.15 +130.62
<  2 -28.27 -28.24 -28.12 -28.01 -27.84 -27.68 -27.44 -27.28 -27.09
<  3 -28.41 -28.37 -28.29 -28.17 -27.99 -27.78 -27.59 -27.41 -27.19
<  4 -28.61 -28.54 -28.47 -28.31 -28.15 -27.94 -27.77 -27.53 -27.30
<  5 -28.71 -28.67 -28.60 -28.45 -28.29 -28.09 -27.89 -27.69 -27.48
<  6 -28.94 -28.82 -28.71 -28.60 -28.48 -28.28 -28.10 -27.82 -27.60
<  7 +129.41 -28.86 -28.84 -28.73 -28.60 -28.42 -28.20 -27.98 +129.13
<  8 +129.05 +129.07 -28.70 -28.80 -28.68 -28.51 -28.25 +129.22 +129.15

Roxy-3D, 2017-05-13:
You have a 28mm correction in the middle of your bed?

Jolly2015, 2017-05-13:
@Roxy-3D  Yes,it is, does it not allowed  to correction so much?But it's useful to me, I used nozzle  hit before , but it's bad for printing,because the  extruder can be active at whole print job. and the BL touch too expensive:).so I mounted a switch under the nozzle, and I modified  the wrong data manually,now it seems ok, but I know this method not regular , so I hope you can fix this issue.

Roxy-3D, 2017-05-13:
In theory...  It should work just fine.  But I think it would work better to get your Z_PROBE_OFFSET_FROM_EXTRUDER set such that the nozzle just touches the glass when you have the height set to 0.00 mm.   

You can shift the mesh up or down with G29 P6 C 1.23

thinkyhead, 2017-05-20:
@Jolly2015  What form of bed leveling are you using?

Jolly2015, 2017-05-22:
@thinkyhead #define AUTO_BED_LEVELING_BILINEAR

closed 2017-11-03 by thinkyhead