MarlinFirmware/Marlin #6712: No Z-axis moves with soft limits enabled after upgrading to 1.1.0-1

jeremyherbert, 2017-05-13:
Hi all,

I have a problem with the soft limits on my printer (Maker Gear M2, heavily modified) after upgrading to the latest release (1.1.0-1). 

I have the firmware hard-coded to home the Z axis towards the maximum, but to set the home position manually to 0,0,0 (`#define MANUAL_X_HOME_POS`, etc) rather than 0,0,~155. Then, using the `M206` command I set a Z offset which I calibrate myself: `M206 Z155`. This is so that I can adjust the offset without reflashing the firmware. 

This worked fine with my old version of marlin (maybe an early 2016 build?) but it locks up now with the new build. Now, with soft limits enabled, if I run:

`G28
G91
G1 Z-30`

the printer does not move at all. Using `G1 Z30` shows `echo:endstops hit:  Z:155.00` which is expected.

If I disable the soft limits and run the same code, the bed moves up by 30mm (which is what I would expect). 

Coordinates from M114 with soft limits enabled:
before G1: `X:0.00 Y:0.00 Z:155.00 E:0.00 Count X:0 Y:0 Z:62000`
after G1: `X:0.00 Y:0.00 Z:155.00 E:0.00 Count X:0 Y:0 Z:62000`

Coordinates with soft limits disabled:
before G1: `X:0.00 Y:0.00 Z:155.00 E:0.00 Count X:0 Y:0 Z:62000`
after G1: `X:0.00 Y:0.00 Z:125.00 E:0.00 Count X:0 Y:0 Z:50000`

Has anyone else experienced this? My Z axis minimum is set as 0, and maximum as 200.

jeremyherbert, 2017-05-13:
After learning about and using `M211`, I get this: `echo:Soft endstops: On   Min: X0.00 Y0.00 Z155.00  Max: X205.00 Y255.00 Z355.00`

It seems that the code adjusts my software limits by the home offset. I can see this in the code too. Is there a way to disable this, or to set a home offset which does not affect the soft limits?

Roxy-3D, 2017-05-13:
I know this subject title isn't true.    I just made a complicated print with these settings:
```cpp
// If enabled, axes won't move below MIN_POS in response to movement commands.
#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS
```

jeremyherbert, 2017-05-13:
So I have been trying to write a patch which changes this behaviour (as controlled by a define) but there is something which is confusing me. In the `update_software_endstops` function, it has the line:

`workspace_offset[axis] = offs;`

Unfortunately, modifying this global variable does more than just recomputing the soft limits; the workspace offsets show up in several other places, for example this line in `set_axis_is_at_home` which is used to compute the coordinates for the home position:

`current_position[axis] = LOGICAL_POSITION(base_home_pos(axis), axis);`

(`LOGICAL_POSITION` is defined as a function of the workspace offsets). So commenting out the `update_software_endstops` calls doesn't appear to do the same thing as disabling software endstops (the former does more), and I'm not really sure why it should be like this? Could someone please help me understand why the code is organised in this way?

jeremyherbert, 2017-05-13:
`LOGICAL_Z_POSITION` (which calls `LOGICAL_POSITION`) also shows up in the `do_probe_raise` function. So adjusting the workspace offset in the soft endstop function can change probing too.

jeremyherbert, 2017-05-13:
I've added a pull request which appears to fix this problem for me. I also refactored out the workspace offset calculations to an appropriately named function.

Comments are welcome.

jeremyherbert, 2017-05-27:
I had some explanation from thinkyhead in the PR, and I see that marlin doesn't really work in this way. Closing.

closed 2017-05-27 by jeremyherbert

vithprodlet, 2017-06-12:
maybe this is a solution for you  ? https://www.youtube.com/watch?v=abFfEi2mIJs