MarlinFirmware/Marlin #6714: SD Uploads fail

Grogyan, 2017-05-13:
RAMPS 1.4
RRD Full Graphic Display
MAX31855 Thermocouple
Marlin 1.1x
OctoPrint 1.32
Brand Spanking new, quality SD card

Uploading a gcode file to the SD card via the host causes the printer to shutdown reading back
`
Recv: Error:MINTEMP triggered, system stopped! Heater_ID: 0
Changing monitoring state from 'Sending file to SD' to 'Error: MINTEMP triggered, system stopped! Heater_ID: 0\x0a'
Recv: Error:Printer halted. kill() called!
`

SDSUPPORT is enabled in the firmware

Tannoo, 2017-05-13:
Try another sd card.

Grogyan, 2017-05-13:
This is the 3rd SD card.

Tannoo, 2017-05-13:
Bad reader?

Tannoo, 2017-05-13:
Try another SD card reader with those cards.

Tannoo, 2017-05-13:
Try writing locally at the pc (from a reader on the pc), then read it from the lcd (after inserting into the lcd's reader).

Take Octoprint out of the loop.

Tannoo, 2017-05-13:
If that works, then try with the IDE serial console.

Grogyan, 2017-05-13:
OctoPrint is not the issue
Have also tried with Repetier host with the same result
SD card works fine on PC.

Just not able to remotely upload to the SD card

Tannoo, 2017-05-13:
Yeah, I'm trying to help.
I tried it this afternoon myself and having issues.

Tannoo, 2017-05-13:
I tried uploading to SD from Repetier and it says it uploaded, but the file size is 0. I will delete that 0 byte file.

So, I tried from Octopi...well, it says it's uploading...forever.

Tannoo, 2017-05-13:
I'll have to look at it more in the am.

Grogyan, 2017-05-14:
Have re flashed the firmware from 1.1.x BugFix as of an hour ago
Same result, still unable to upload gcode files to SD through, OctoPrint, Repetier Host, Pronterface

Tannoo, 2017-05-14:
I have been trying to narrow this down to find when and where it got broken.

I have tried all backups I have... all do the same. Cannot upload any files from any host.
I have downgraded the IDE from 1.8.2 to 1.6.7 and re-flashed... same.

I have reduced the comm speed from 250000 to 115200 and to 57600... same.

I then went back to IDE 1.8.2, put comm speed back to 250000 and reflashed. Still the same.

This time I tried to create a file manually... I sent:
`M28 testing.gco`
then
`G28`
then
`G0 X100 Y100`
then
`G28`
then
`M29`

All looked good.
Then I sent:
`M23 testing.gco`
then
`M24 testing.gco`

and the printer did exactly what that file contained.

So, why can any host not send a file anymore? I don't know.

We need to enlist some smarter people on this.

Bob-the-Kuhn, 2017-05-14:
I gave up trying to upload files directly to the SD card via Repetier Host.  The new version doesn't seem any better

When I need to upload a file to the SD card I actually load it into Repetier first (via the LOAD button) and then go to the SD card section and select Upload Current G-Code.  After entering the Uploaded Filename then just click in the File box.  Now if you click OK it'll upload.

[image]

Bob-the-Kuhn, 2017-05-14:
File extension MUST be 'g' - nothing else works.

Tannoo, 2017-05-14:
I will try that later.. I was using `.gco`.

Grogyan, 2017-05-14:
I can't verify as I am at work.
A quick search in the code shows in CardReader.cpp
'
void CardReader::checkautostart(bool force) {
if (!force && (!autostart_stilltocheck || ELAPSED(millis(),
next_autostart_ms)))
return;
autostart_stilltocheck = false;
if (!cardOK) {
initsd();
if (!cardOK) return; // fail
}
char autoname[10];
sprintf_P(autoname, PSTR("auto%i.g"), autostart_index);
for (int8_t i = 0; i < (int8_t)strlen(autoname); i++) autoname[i] = tolower
(autoname[i]);
'

Which shows the *.g as the only upload option.  which is kinda a problem,
as some programs export to *.gco and others *.gcode (Cura)

Bob-the-Kuhn, 2017-05-15:
I lied to you.  Looks like about any file extension that starts with a 'g' will do when writing a file to the SD.

The code that was cited is actually for the autostart function.  That's disabled by default.  The only way to enable it is to edit Marlin_main.cpp.

Tannoo, 2017-05-15:
Repetier is creating the filename and is able to delete it. The file will only be larger than 0 bytes if I select to include the start and end codes.

When I do that, it will print that file which now only contains `M84`... And the lcd says print complete.

I am baffled as to why it will not upload a sliced file... Even a freshly sliced file and uploaded as you suggested results in a 0 byte file.

Tannoo, 2017-05-15:
Even a very small single layer designed and sliced, will not upload.

Tannoo, 2017-05-15:
Do I need to reset eeprom? I now think I remember having this trouble once before and was told to re-int eeprom.

I'll try that in the am.

Tannoo, 2017-05-15:
The other option I'm going to try is all caps for the Sdcard filename.

Bob-the-Kuhn, 2017-05-15:
The process I use with Repetier Host is:
1. click on the Load button on the main screen
2. select the file & click Open
[image]
3. click on G-Code Editor to make sure something got there
4. click on the SD Card button
5. click on the upload button
[image]
6. clear the two "Include .. " flags & select Upload Current G-Code
7. type a DOS 8:3 file name in the "Uploaded Filename:" box - extension must start with a 'g' - doesn't matter if the letters are upper or lower case (lower case is automatically converted to uppercase)
8. do NOT enter anything into the "File:" box.  On the older Repetier Host you had to click in the box but not enter anything).
9 click OK
[image]

Grogyan, 2017-05-15:
What about *. GCO and *. GCODE extensions?

Bob-the-Kuhn, 2017-05-15:
Those are acceptable when using the Load button to move a g-code file into the Repetier Host's buffer.   You may have to click on the extension box and select ALL FILES to see your file.
[image]

The .GCO is acceptable when uploading from the buffer to the SD card.  The .GCODE has too many letters in it to upload from the buffer to the SD card.

Grogyan, 2017-05-15:
Okay, I have tried both *.gco and *.g extensions
OctoPrint, Repetier Host and Printerface all keep doing the same thing
`
Send: M28 /pi3_bo~1.gco
Recv: echo:Now doing file: /pi3_bo~1.gco
Recv: Writing to file: /pi3_bo~1.gco
Changing monitoring state from 'Operational' to 'Sending file to SD'
Recv: ok
Send: N3 M190 S60.000000*115
Recv: Error:MINTEMP triggered, system stopped! Heater_ID: 0
Changing monitoring state from 'Sending file to SD' to 'Error: MINTEMP triggered, system stopped! Heater_ID: 0\x0a'
Recv: Error:Printer halted. kill() called!
`

Tannoo, 2017-05-15:
I haven't gotten any mintemp errors.

Is your mintemp still default of 5?

Grogyan, 2017-05-15:
-3 
MAX31855 Thermocouple

Tannoo, 2017-05-15:
I have confirmed that uploading from Repitier is fine if you upload freshly sliced gcode and only in 8.3 file format. It has to "print" it to the SD card. The file uploading doesn't work. I believe that is a host issue.

Tannoo, 2017-05-15:
It took several tries, but I got Octopi to send it also.

Tannoo, 2017-05-15:
>-3
MAX31855 Thermocouple

That is the thermocouple, not the MINTEMP.

Tannoo, 2017-05-15:
I don't have issues with MINTEMP or MAXTEMP when getting these files sent to SD.
I cannot set the thermistor to -3 and be able to do anything. Doing so throws a MAXTEMP error right at startup that I cannot get around.

Grogyan, 2017-05-15:
The SD card and the thermocouple operate on the same SPI bus, where the firmware asserts the CS pin of which device it is talking to. 

So, what it looks like is that the SD CS pin is being asserted for too long when the thermocouple is being read. 
This would definitely give the MINTEMP error. 
But I didn't have this issue way back in RC6

closed 2018-02-06 by thinkyhead