MarlinFirmware/Marlin #6718: manage_heater() during segmented moves?

oldmcg, 2017-05-13:
@thinkyhead, @Roxy-3D 
I was looking at the G2_G3 arc code with respect to how it interacts with UBL, and I noticed this in the top of the plan_arc per-segment loop:

```
    for (uint16_t i = 1; i < segments; i++) { // Iterate (segments-1) times

      thermalManager.manage_heater();
      if (ELAPSED(millis(), next_idle_ms)) {
        next_idle_ms = millis() + 200UL;
        idle();
      }
```
The ordinary non-UBL delta kinematic segmentation loop does NOT call manage_heater() or idle(), so neither does the UBL segmentation loop.

There could be 1000 _buffer_line segments in a 100mm line.  Should the UBL segmentation loop (and/or the non-UBL kinematic segmentation loop) periodically call manage_heater() and/or idle()?

Roxy-3D, 2017-05-13:
Maybe...   It is important that gets called often enough it can make intelligent decisions.   But I think I would argue calling it every iteration of that loop is way too often and going to waste a lot of CPU cycles.    I think I would  have a counter and  maybe only call it 1 or 10 or 1 out 20 iterations of the loop.

oldmcg, 2017-05-13:
For UBL_DELTA I could probably call it once per mesh cell transition.  I just didn't know if it was handled by interrupt or still required.

Looks like idle() calls manage_heater() already, so pointless to call both separately -- don't know why plan_arc calls both.

I could also check millis() periodically (once per mesh cell) and then call idle() when past a time threshold, but how often?  200ms?  100ms?  

But the original current non-UBL delta segmentation routine doesn't do any of this so I don't have a good model or threshold to follow.

Roxy-3D, 2017-05-13:
If you went the millis() route...  I would think 100ms would be OK.  The heaters don't change the temperature super fast.

> But the original current non-UBL delta segmentation routine doesn't do any of this so I don't have a good model or threshold to follow.

Probably they are assumming the whole move can be broken up into smaller moves within a quarter second.   Or, it maybe that for deltas, the number of small sections that get schedule totally fill the planner buffer, and the planner is going to have to wait anyway to finish the move.   And while the planner waits...  It will be calling idle().

I don't know the answer.   But it would not surprise me if everything is OK if you don't call it just because one of two things is going to happen.   Either the move being broken up fills the move buffers and the planner blocks waiting for space and calls idle().   Or the move finishes being broken apart and idle() gets called before the next command gets processed.

closed 2017-11-01 by thinkyhead