MarlinFirmware/Marlin #6720: Endstop Trigger issue (Marlin-1.1.x)

TunaLatte, 2017-05-13:
Hello,
Thank you for the efforts on making this decent firmware.

I dusted off my Printrbot Jr, having Printrboard Rev.D I was able to compile and flash the firmware successfully with Arduino-1.8.2 ide, teensyduino 1.36, (flashing with dfu-programmer), and adding #include"Arduino.h" and "macros.h" to serial.h for "customserial" issue.
My Configuration.h diff:

```
+ #define MOTHERBOARD BOARD_PRINTRBOARD
- #define MOTHERBOARD BOARD_RAMPS_14_EFB
+ #define DEFAULT_AXIS_STEPS_PER_UNIT   { 63.36,63.36,2272.72,590 }
- #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500 }
+ #define DEFAULT_MAX_FEEDRATE          { 60, 60, 2, 14 }
- #define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }
+ #define DEFAULT_MAX_ACCELERATION      { 2000,2000,30,10000 }
- #define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }
+ #define INVERT_X_DIR true
- #define INVERT_X_DIR false
+ #define EEPROM_SETTINGS
- //#define EEPROM_SETTINGS
+ #define SDSUPPORT
- //#define SDSUPPORT"
```
I have also tried reverting to stock firmware, and compiling a commit of first March(while i was trying to solve the serial issue), non have introduced this issue.
```
19:19:14.883 : N20 M119*24
19:19:14.906 : Reporting endstop status
19:19:14.906 : x_min: open
19:19:14.907 : y_min: TRIGGERED
19:19:14.907 : z_min: open
```
I have also tried inverting with M119 Y1 1 (or 0) did not change anything.

Thank you

Bob-the-Kuhn, 2017-05-13:
M119 only reports the state of the endstops.  You can't change anything with it.

You'll need to go into your **Configuration.h** file and change **Y_MIN_ENDSTOP_INVERTING** .

TunaLatte, 2017-05-13:
Hey thank you for the response!
But the issue is, even with the endstop is physically triggered, it is stuck at "TRIGGERED", so inverting is useless.

TunaLatte, 2017-05-13:
Compiling and flashing Marlin 1.0.x:
```
21:01:17.324 : Reporting endstop status
21:01:17.324 : x_min: TRIGGERED
21:01:17.324 : y_min: TRIGGERED
21:01:17.324 : z_min: TRIGGERED
21:01:29.454 : N26 M119*30
21:01:29.609 : Reporting endstop status
21:01:29.609 : x_min: TRIGGERED
21:01:29.609 : y_min: open
21:01:29.610 : z_min: TRIGGERED
```

Bob-the-Kuhn, 2017-05-13:
On 1.0.2 all endstops toggle but on 1.1,x the Y_MIN is stuck.

Do you have the SD card enabled?

In 1.02 the Y_MIN endstop was always pin 8.  In 1.1.0 it moves to pin 37 if the SD card is enabled.
```
#if ENABLED(SDSUPPORT)
  #define Y_STOP_PIN       37 // Move Ystop to Estop socket
#else
  #define Y_STOP_PIN        8 // Ystop in Ystop socket
```

TunaLatte, 2017-05-13:
omg I just disabled SDSUPPORT and it worked, and **then** I see your comment, arr.
Is there a workaround with sd enabled? Should I just change pin number?
Thanks a bunch.

Bob-the-Kuhn, 2017-05-13:
It sounds like the change is to move the Y_MIN cable from the Y_STOP connector to the E_STOP connector.

TunaLatte, 2017-05-13:
[image]
[image]
For reference ^

It works, thank you. Closing issue.

Bob-the-Kuhn, 2017-05-13:
FANTASTIC!!

closed 2017-05-13 by TunaLatte