MarlinFirmware/Marlin #6721: How to move the nozzle following the bed

batata004, 2017-05-13:
Hi,

I need some help maybe it's possible to be done with GCODE. I am creating a "standard" GCODE file that I will use in many prints.

To avoid having to bed level before every print I usually do this: turn on the printer, LCD -> BED LEVEL -> BED LEVEL and level the bed automatically (using servo). After that I can do 1 or 10 prints without needing to BED LEVEL.

But sometimes, after finishing one print and starting another the heated bed cools down completely and if I request it to print another file the printer sits there for 2 to 3 minutes doing nothing but heating.

So my question is this: is there any GCODE I can insert in my start script that does this:

IF (TARGET_TEMPERATURE - CURRENT TEMPERATURE > 20) {

     bedLevel();

}

Is it possible? Instead of keepking my printer doing nothing I would request it to start auto leveling again if it has nothing else to do while waiting. 

Is it possible to do this with some GCODE trick?

Bob-the-Kuhn, 2017-05-13:
The g-codes supported by Marlin do not have that capability.

If you've saved the bed leveling values to EEPROM (via M500) then you don't need to level the bed again unless something has mechanically changed.

The "standard" GCODE file items usually go into the slicer's start and finish scripts/settings.

If you don't want to turn off the heaters after a print then modify the slicer's  finish scripts/settings.

batata004, 2017-05-13:
I dont like saving GCODE values to epprom cause when I remove my pieces from the printer, sometimes they stick too much and I shake everytjing to remove the part. So i think it's better doing a new bed level once in a while.

I want to turn off the heaters cause sometimes I go out and take hours to come back. Isnt there any workaround so?

emartinez167, 2017-05-14:
This is a question for the reprap forum, not a coding problem. Please post your question there.

Regards,

Ernesto.

closed 2017-05-14 by batata004