MarlinFirmware/Marlin #6722: Marlin configurator

akaJes, 2017-05-13:
Introduce new version 
installation: `sudo npm -g i marlin-conf`
then right click in Marlin folder with git repository and choose Open in Terminal
and type `mct`

auto detection of release from git (show which options changed)
**instant** changing files 
* Marlin/Configuration.h
* Marlin/Configuration_adv.h

documentation used from https://github.com/MarlinFirmware/MarlinDocumentation

[image]
(It is a extension for previous version https://github.com/MarlinFirmware/Marlin/issues/6613)

akaJes, 2017-05-14:
some improvements
to update `sudo npm -g i marlin-conf`
i think it can be good tool for beginners to explore Marlin firmware

[image]

if anyone have some suggestions to add any features welcome

akaJes, 2017-05-14:
added:
* button 'Issue' - for quick issue creation with your configuration
* added in hints panels,alerts and tables

changed:
* hints only if present
* all librares is local

Roxy-3D, 2017-05-14:
It would be nice if it could take an existing Configuration.h file set and map those old parameters to the current release.   Many users, if they have source, got it as a .ZIP file from their vendor.   There is no easy way to know where they are in the branch process....

But it is that group of users that is probably the largest and having the most difficulty getting caught up to v1.1.0

akaJes, 2017-05-15:
I see two ways how to map old user's config over new release
* Simple: just export all options from old then import to new deployed release
* Harder: 
 1. try to seek for nearest possible release from all git tags and/or to give for user ability choose it manually
 2. Export only changes between chosen git tag and user config then import these changes to new release

Which would be prefer for your mind?

akaJes, 2017-05-16:
Hi @Roxy-3D @thinkyhead 
updated version  with ability to load users old configurations
full documentation https://github.com/akaJes/marlin-config#readme

[image]

thinkyhead, 2017-05-17:
Exciting! I look forward to (1) giving it a try and (2) linking to it from our website. It would be good to be able to point users to this resource. In fact, I will post a link on Twitter now….

akaJes, 2017-05-17:
Hi
You can try it now - first done
Second is ready but not present in web interface,
it gets files from cloned git repository instead website

bgort, 2017-05-17:
This looks really fantastic.  Thanks very much for doing it!

akaJes, 2017-05-17:
thank you @bgort 
i think about ability PlatformIO implementation for quickest a firmware compilation and uploading it - maybe serial console too

bgort, 2017-05-17:
I think both of those are great ideas.  

I use and really like PIO.  It's very simple, clean, and easy to use for those new to this kind of thing..

Roxy-3D, 2017-05-17:
> I see two ways how to map old user's config over new release
> Simple: just export all options from old then import to new deployed release
> Harder:
>    try to seek for nearest possible release from all git tags and/or to give for user ability choose it manually

> Which would be prefer for your mind?

For the purpose of helping customers with very old versions of Marlin (that they might have gotten from the machine's maker)...   The first version makes sense.

It is my belief we have hardly ever (maybe never) taken an option name and re-purposed it to another meaning.    But we have taken names and changed them to make more sense with the current software base and thinking.

I think what I'm hearing you say is you can scan for **_ANY_** keyword option name that has been used in the past and then map that name onto what the current option set names are.   If so....   For sure people will love this tool!

bgort, 2017-05-17:
> (2) linking to it from our website

Is this something that will be online continuously for others to use?  If so, is it something that will be hosted on marlinfw.org, or?  Might I suggest a subdomain like configurator.marlinfw.org, or something along those lines?  

If the marlinfw.org server can't host it for any reason, I'd be happy to donate server space and bandwidth -- I have a number of cloud servers that are significantly underutilized.  Happy to help in that way.  Just let me know.

akaJes, 2017-05-17:
@thinkyhead @bgort  this tools was designed as application and works with users' file system for instant files changing
> (2) linking to it from our website

for deploying it on the site it not suitable well and needs some rework.

bgort, 2017-05-17:
Ah, I understand!

thinkyhead, 2017-05-18:
Now that we have all the configuration settings established, I will produce a document (Google sheet) that shows how to map every old option to every new option, then anyone who wants to make such a converter can use it as reference.

From that I should be able to quickly produce a small Javascript singlet that can take dropped 1.0.x configuration files and produce 1.1.x configurations from them. I've already got code that can produce a ZIP right in Javascript (as part of my old currently-broken [configurator](https://github.com/thinkyhead/MarlinConfigurator)).

akaJes, 2017-05-19:
hi @bgort 
embedding of platformio done and console almost done
how pio detects default device ports?
is important to specify port manually?

bgort, 2017-05-19:
If only one device is connected, it detects automatically (and correctly), but the upload port can be specified in platformio.ini or on the commandline.

See:
- http://docs.platformio.org/en/latest/projectconf.html#id83; and 
- http://docs.platformio.org/en/latest/userguide/cmd_run.html#options

It'd be nice to have an option to specify the port, but generally I think it's best that it choose automatically, generally, as I imagine most only have one device connected at a time and those who have more can pretty quickly figure out how to specify the port...

akaJes, 2017-05-22:
i have built version with  pio support and consoles manager
for upgrade `sudo npm -g i marlin-conf`

thinkyhead, 2017-05-22:
Will give it a try soon…
```
npm WARN deprecated isarray@2.0.1: Just use Array.isArray directly
```

thinkyhead, 2017-05-22:
First note: I've been sticking JSON into the config files above the options that they apply to. You can use that JSON to produce a drop-down, radio buttons, or select list.

thinkyhead, 2017-05-22:
I added the @section markers with the intent of making each of those a separate tab, so that each config section is smaller and easier to digest.

fiveangle, 2017-05-22:
Looking forward to giving this a whirl - thx @akaJes !

akaJes, 2017-05-23:
Hi @thinkyhead 
I released new one with navigation bar and fixes

> `npm WARN deprecated isarray@2.0.1: Just use Array.isArray directly`

this warning becomes from dependence libraries
> `@section`

new version has right sided navigate bar
> sticking JSON

in last version a few of these helpers
```
Configuration.h: * :[0, 1, 2, 3, 4, 5, 6, 7]
Configuration.h: * :[2400, 9600, 19200, 38400, 57600, 115200, 250000]
Configuration.h:// :[1, 2, 3, 4, 5]
Configuration.h: * :{ 0:'No power switch', 1:'ATX', 2:'X-Box 360' }
Configuration.h: * :{ '0': "Not used", '1':"100k / 4.7k - EPCOS", '2':"200k / 4.7k - ATC Semitec 204GT-2", '3':"Mendel-parts / 4.7k", '4':"10k !! do not use for a hotend. Bad resolution at high temp. !!", '5':"100K / 4.7k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '6':"100k / 4.7k EPCOS - Not as accurate as Table 1", '7':"100k / 4.7k Honeywell 135-104LAG-J01", '8':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT", '9':"100k / 4.7k GE Sensing AL03006-58.2K-97-G1", '10':"100k / 4.7k RS 198-961", '11':"100k / 4.7k beta 3950 1%", '12':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT (calibrated for Makibox hot bed)", '13':"100k Hisens 3950  1% up to 300°C for hotend 'Simple ONE ' & hotend 'All In ONE'", '20':"PT100 (Ultimainboard V2.x)", '51':"100k / 1k - EPCOS", '52':"200k / 1k - ATC Semitec 204GT-2", '55':"100k / 1k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '60':"100k Maker's Tool Works Kapton Bed Thermistor beta=3950", '66':"Dyze Design 4.7M High Temperature thermistor", '70':"the 100K thermistor found in the bq Hephestos 2", '71':"100k / 4.7k Honeywell 135-104LAF-J01", '147':"Pt100 / 4.7k", '1047':"Pt1000 / 4.7k", '110':"Pt100 / 1k (non-standard)", '1010':"Pt1000 / 1k (non standard)", '-3':"Thermocouple + MAX31855 (only for sensor 0)", '-2':"Thermocouple + MAX6675 (only for sensor 0)", '-1':"Thermocouple + AD595",'998':"Dummy 1", '999':"Dummy 2" }
Configuration.h:// :{ 0:'Low', 1:'High' }
Configuration.h:// :[-1,1]
Configuration.h: * :{ 'en':'English', 'an':'Aragonese', 'bg':'Bulgarian', 'ca':'Catalan', 'cn':'Chinese', 'cz':'Czech', 'de':'German', 'el':'Greek', 'el-gr':'Greek (Greece)', 'es':'Spanish', 'eu':'Basque-Euskera', 'fi':'Finnish', 'fr':'French', 'gl':'Galician', 'hr':'Croatian', 'it':'Italian', 'kana':'Japanese', 'kana_utf8':'Japanese (UTF8)', 'nl':'Dutch', 'pl':'Polish', 'pt':'Portuguese', 'pt-br':'Portuguese (Brazilian)', 'pt-br_utf8':'Portuguese (Brazilian UTF8)', 'pt_utf8':'Portuguese (UTF8)', 'ru':'Russian', 'tr':'Turkish', 'uk':'Ukrainian', 'test':'TEST' }
Configuration.h: * :['JAPANESE', 'WESTERN', 'CYRILLIC']
Configuration_adv.h:// :[0, 2, 4, 8, 16, 32, 64, 128, 256]
```
but I will use soon it when will rework of input field types

akaJes, 2017-05-25:
**UPDATE**
added Gcode help
added conditions for enabled options/gcodes

[image]

akaJes, 2017-05-26:
[marlin-conf v2.5.3](https://github.com/akaJes/marlin-config#readme)

Some updates:
- swig parser improved for a gcode help
- navigation reworked
- more condensed version for tables
- added links to options for gcodes requirements
- added boolean and select fields

> sticked JSON

some troubles was with these not standard JSON (only `"` must be used for strings! )
```
                var json=def.select.trim();
                if (json[0]=='[')
                  json=json.replace(/'/g,'"');
                else
                  json=json.replace(/'?([\w-]+)'?\:/g,'"$1":').replace(/\:\s*\'/g,':"').replace(/(\')\s*(,|\})/g,'"$2'); 
                try{
                  def.select=JSON.parse(json);
                }catch(e){
                  def.type='numeric';
                }
```
but it were solved

akaJes, 2017-05-27:
PlatformIO compiler console support colors now (with my new npm library [**tty2html**](https://github.com/akaJes/tty2html) )
improved tty console:
- saving previous port speed and EOL type
- dropdown with history for send field
- auto scrolling

[image]
[image]

akaJes, 2017-05-29:
I have build new standalone version for x64 Win and Linux https://github.com/akaJes/marlin-config/releases/tag/2.6.0

vMeph, 2017-05-29:
is the marlin-conf.Setup.2.6.0.exe for windows?
i have tryed to run and i get this
[image]

akaJes, 2017-05-30:
there were no win32 version in this setup - you can try again

vMeph, 2017-05-30:
can you be more detailed in how to install on windows cause i have try again using the marlin-conf.Setup.2.6.0.exe and i get same error the other files appimage no idea what they and how to open

thanks

akaJes, 2017-05-31:
it is my fault - i have built cross-compiled version and serial port component has fail
i released new version 2.6.1  and have tested on windown x64 - but it with no serial consoles

vMeph, 2017-06-01:
now say error
dependencies
application needs GIT to be installed!

ts4s, 2017-06-01:
on the the get hub page  there's a section titled "Installation prebuilt with Electron versions" and it has a sub section "Dependencies". there's a link to git. follow the link, download, and install it. everything should work once installed.

vMeph, 2017-06-01:
@ts4s im not able to find the link could send link?

thanks

ts4s, 2017-06-01:
[image]

vMeph, 2017-06-01:
@ts4s  thanks mate got it to work :)
@akaJes great job you have done it is very cool, i think this is the way to go for less experience people in looking into coding, maybe in future could get even better if it add examples with explanations in how to set things and how they work,  like end stops, Bluetooth, RGB leds etc.. with diagrams of wiring etc..
very impressive the job you did

thanks

akaJes, 2017-06-03:
reworked toolbar and works console under windows
https://github.com/akaJes/marlin-config/releases/tag/v2.6.3x

megic1, 2017-06-06:
hey akajes - cool tool. exactly what i was looking for. i just have a few problems:
- the configurator doesn't see my own pid-settings, but instead shows only the defaults in the ui.
- interestingly, the hotend-pid-settings are wrong in the ui, but are correctly transferred to the config-file (under the headline makergear, but ok... :-).
- the pid settings for the bed are wrong in the ui AND don't appear in the config-file.
i'm coming from marlin 1.1.1 and tried to go to 1.1.3. i somehow can't attach my config files to this post...
any-ideas?
best wishes, megic1

akaJes, 2017-06-06:
Hi @megic1, post your config file at free service like this https://uploadfiles.io, it's hard to say some about your issue without your config

megic1, 2017-06-06:
sure. tried to upload as a zip, but that didn't work. have a look here. https://ufile.io/lnp5l

megic1, 2017-06-06:
oops. that was only adv. here the "normal" one ;-) https://ufile.io/2tdks

akaJes, 2017-06-07:
i publish new 2.6.4 version to npm

thinkyhead, 2017-06-07:
It's coming along well… I will give the latest version a test soon.

akaJes, 2017-06-19:
@thinkyhead would you consider an option to include the default configuration files for delta/scara machines, alike as configuration_scara.h etc? I mean only additional options and include these in configuration.h

akaJes, 2017-06-20:
new improvement 
coordinate helper:
moving head, regulate flow,speed,fan

[image]

akaJes, 2017-06-21:
updated npm package
new feature - ability to change base configuration files from Examples

thinkyhead, 2017-06-22:
Regarding SCARA, are you able to use the SCARA example files for that porpoise?

akaJes, 2017-06-22:
@thinkyhead i suppose that scara is some old fashioned :)
and i found another way to use scara/delta files as base - i have added switch for changing files from whole folder with examples

have published **2.7.0 version** - with the snippets for stepper motors step values calculations

akaJes, 2017-06-23:
new update 2.7.1
added snippets with default and custom images

vMeph, 2017-06-23:
@akaJes just one question
Sense there is new updates of marlim all the time, and every time you update, you got to redo the settings all over again, was wondering if this software does or if is possible compare the new update and old settings and automatic write them for you?

akaJes, 2017-06-23:
@vMeph You mean about an ability to change the releases, without the configuration reset and uploading your old configuration again?
I have the Configuration* files saved for each revision(checked on hardware)  - and I drag it into page after changing the release

thinkyhead, 2017-06-23:
Some new config options will be added to 1.1.4, so do a compare with the 1.1.2 / 1.1.3 configs to see what's changed. Version 1.1.4 should be the last bug-fix patch for 1.1.x before the 2.0.x branch is ready to take over main development…. Aiming for July 1.

akaJes, 2017-06-24:
new feature:
snippet for editing custom **Boot Screen** image

[image]

akaJes, 2017-07-08:
new update on npm package:
- you can share now your current marlin-conf page
- all changes from all pages synchronized 
- open public UPnP port for external access
- ability to connect to marlin-conf a phone as webcam and get access from anywhere

[image]

akaJes, 2017-07-15:
released new version https://github.com/akaJes/marlin-config/releases

fixed: console 3d Command menu
added:
- external upnp access
- ability to use phone as web cam
- simultaneously working
- store current config
- upload stored config

akaJes, 2017-07-27:
released new version https://github.com/akaJes/marlin-config/releases
fixed:
- examples bug
- serial port for macos electron
- serial port polling for all OS
- ability to select enviroments for PlatformIO

akaJes, 2017-07-27:
i have made https://www.instructables.com/id/Marlin-config/ too

akaJes, 2017-11-09:
new version with ability to publicate your configuration
https://github.com/akaJes/marlin-config/releases

fiveangle, 2017-11-10:
On my phone so can’t check now, but if this is what it claims to be, wow 😮, just… wow. #gamechanger ?

Jartza, 2018-03-08:
This is interesting, I've been thinking for web-based configurator for ages, but also with the ability to compile online. I'm making new bootloader for atmega2560 that still works with avrdude, but can also update firmware from SD-card, if there is new one - my proto bootloader already works, unfortunately only for .bin -files, not .hex. Needs some optimisations and such to fit in the .hex -parser together with CRC check.

[4 more replies]

closed 2018-11-05 by thinkyhead