MarlinFirmware/Marlin #6723: Printing from SDcard fails - open failed, File:

khcassidy, 2017-05-13:
Hello,
After upgrading to the latest Marlin 1.1.1 I can no longer print from the SD card (launched from Pronterface & tried simplify3d).

echo:SD card ok
Begin file list
CAL_S.G 1696271
CAP.G 433195
JUSPIN~1.GCO 1717198
End file list
SD printing byte 0/0
echo:Now fresh file: cap.g 433195
**open failed, File: cap.g 433195.**
SD printing byte 0/0
SD printing byte 0/0

I have also tried the latest bugfix branch with the same results.

When I try reloading the old RC Bugfix firmware Printing works fine

echo:SD card ok
Begin file list
CAL_S.G
CAP.G
JUSPIN~1.GCO
End file list
echo:Now fresh file: cal_s.g
**File opened: cal_s.g Size: 1696271**
File selected
Print started at: 20:00:43

I have multiple different files and models.

Any ideas?

Thanks!

Tannoo, 2017-05-13:
Will it print the SD file from the LCD?

Grogyan, 2017-05-13:
Is this related to #6714  ?

Tannoo, 2017-05-13:
Not exactly. I can print just fine from SD via LCD, Octoprint, and Repetier....even after any failed uploads.

Tannoo, 2017-05-13:
Btw, I am running `bugfix-1.1.x` from this afternoon.

khcassidy, 2017-05-14:
I have not got an LCD to test with.

khcassidy, 2017-05-14:
Hi Tannoo,Yes plugged  in an LCD, and it worked better,I was able to pick a file and the printer started. So yes from the LCD controls I can select a file. (I do not want to use an LCD because it slows my delta down).

khcassidy, 2017-05-16:
My Board is a Megatronics v3.0 btw.

khcassidy, 2017-05-17:
Tried todays fix but I still have the issue. Thought it may have to do with the EEPROM fix in bugfix-1.1.x

Roxy-3D, 2017-05-17:
> Hi Tannoo,Yes plugged in an LCD, and it worked better,I was able to pick a file and the printer started. So yes from the LCD controls I can select a file. (I do not want to use an LCD because it slows my delta down).

OK!  I hear you!  But the LCD code is much easier on the processor than it used to be.   And you can change this number in ultralcd.h to 
```cpp
 #define LCD_UPDATE_INTERVAL 100
```
something like
```cpp
 #define LCD_UPDATE_INTERVAL 1000
```
At that point...  The LCD panel is hardly getting any CPU cycles...  And it will be kind of sluggish to navigate through the menus on it.  But it will work....

khcassidy, 2017-05-18:
Thanks Roxy-3D. I can use it as a work around.

Is there any logging I can enable  to help out with troubleshooting? 

In real life I have little use for the LCD really since my PC sits next to the printer. And the issue does not exist in the 1.1.x RC's or previous releases. Just wanted to guage if Im the only one which would point to a miss configuration some where or a hardware abnormality.

For the time being Ill stick with the RC, its working resonable well for me. Maybe someone with better troubleshooting skillz will run into the issue.

Arcari55, 2017-07-02:
I also encountered this problem using Simplify3D and Marlin 1.1.x...

The error is caused by the addition of "File Size" to the information returned by M20 (Command issued by "Print From SD Card" in Simplify3D).

When Marlin responds to M20 with the list of file names (now with size included) it confuses Simplify3D because 3D is expecting a list of files, so it treats everything "Part1.G 452854" as the file name.  When you select the file, 3D sends the M23 open command with "Part1.G 452854" and Marlin errors out with file not found "open failed".

The quick fix is to remove this additional file size information from the Marlin response.  

In File **cardreader.cpp**, line 137:

case LS_SerialPrint:
          createFilename(filename, p);
          SERIAL_PROTOCOL(prepend);
          SERIAL_PROTOCOL(filename);
          SERIAL_PROTOCOLCHAR(' ');
          SERIAL_PROTOCOLLN(p.fileSize);
          break;

**Needs to be updated to:**

case LS_SerialPrint:
          createFilename(filename, p);
          SERIAL_PROTOCOL(prepend);
          **SERIAL_PROTOCOLLN(filename);
          //SERIAL_PROTOCOLCHAR(' ');
          //SERIAL_PROTOCOLLN(p.fileSize);**
          break;

The Marlin team should consider removing this in the next bugfix or release **or** make it an option in config.h for downstream software that can handle it.

Bob-the-Kuhn, 2017-07-02:
Could Simplify3D be made to send `M20 S0` to get a directory listing without the file size?

Arcari55, 2017-07-02:
Yes, the commands linked to the GUI buttons are user-defined in the Firmware Configuration/SDcard menu of Simplify3D.  Its M20 by default for the "RepRap (Marlin/Repetier/Sprinter)" profile.

The 'S' and 'P' options would have to be added to the M20 command in Marlin to accommodate this approach.  Simplify3D would also have to add this setting to a new default "Marlin" profile so their software works out-of-the-box (would introduce a new profile that deviates from RepRap, Sprinter etc).

I am not sure if a standard exists for the M20 'S' option.  If I were to write that code, I would make M20 with no arguments return file name only.  The 'S' option would give the user different levels of detail if desired.  This would make Marlin backwards compatible with existing printers and provide new 'S' options available to software that can process additional information.

Bob-the-Kuhn, 2017-07-02:
Strange - the reason we changed to "defaulting to name with file size" is to be compatible with how Repetier does it.

The RepRap gcode Wiki indicates that the S parameter specifies the output style.  

Strange, the Wiki doesn't mention file size.

Bob-the-Kuhn, 2017-07-03:
In case you want to test ...

Here's the four files I changed so the M20 S0 will give a file list without sizes.
https://www.dropbox.com/s/1nuoh2825n2brlp/M20_S0.zip?dl=0

Here's the entire Marlin directory.
https://www.dropbox.com/s/ap5znlyxpmq3sy5/M20_S0_all.zip?dl=0

Arcari55, 2017-07-03:
Did some experimenting...

Flashed Rambo with latest Repetier and Sprinter to see how they responded to M20 / M23:

**Repetier** - Replies to M20 with "File Name + File Size".  Sending M23 from Simplify3D "M23 Test.G 235675" results in no error and file is opened/selected. (Explains no forum activity on this issue)

**Sprinter** - Replies to M20 with "File Name" only.  No issues with M23.

Neither respond to any 'S' (0-5) options.

It appears that the way Repetier parses commands, its sending M23 to a case structure, then _only_ the following argument for the file name, which is Test.G.  The 235675 gets dropped and ignored because it has no command associated with it.

So I guess a complete solution would be to add similar behavior to Marlin M23, ignoring anything after .g | .gco | .gcode

I could not find any info on the wiki about what the 'S' options correspond to.  If there is no accepted standard, I guess you get to define it.  So S0 = names only, S1 = names + size?, S2 = long names?, S3 = long names + size?  dates and time?       Overthinking this would be no problem.

Bob-the-Kuhn, 2017-07-03:
Can you get Simplify3D to send the `M20 S0 `command?

Marlin's parsing treats all strings the same.  In some situations having a blank in the string is normal.

If it's not feasible to change the way Simplify3D works then it's time to look at changing M23 to ignore anything after a blank.  In the 8.3 DOS file naming system a blank is not allowed in a file name.

Roxy-3D, 2017-07-03:
Shouldn't some of the GCode Nazi's speak up now and say  "You shouldn't but a G, M, N or T in a file name  because that violates the GCode standard.  And you shouldn't have more than 1 of the same letter in any filename because that also violates the GCode standard."

Bob-the-Kuhn, 2017-07-03:
Here's a patch where M23 truncates the file name at the first blank.  That should get you limping along.
https://www.dropbox.com/s/tkfzqv138zcjpjv/M23%20mod%20Marlin_main.zip?dl=0

It's definitely NOT ready for prime time.

There's some strange stuff in the M23 processing.  There's a note deep in the bowels about 8.3 but the syntax checking doesn't strictly enforce it. 
- More than 8 characters before the period works.
- The file extension is limited to 3 places.
- A space in the file name doesn't work

M20 will report file names with spaces in it if the first letter of the extension is "G".

Bottom line is we'll only get someone interested in doing this properly if there's no way to make Simplify3D happy any other way.

Arcari55, 2017-07-03:
I tested the S0 code you posted, and changed the Simplify3D option to send M20 S0.  That combo worked correctly.  The only problem with that solution is that its a silent fix (its still an error until everyone understands the fix and updates their simplify3D settings).   We should send this info to the developers of Simplify3D and put it in the Marlin documentation.

[image]

I patched my copy early this morning (when I first posted), so I'm good to go.

Just keeping the conversation alive for KHCassidy (who first asked) and whoever else is interested.

Thanks for your help.

khcassidy, 2017-07-03:
Thanks Arcari55, 
Tested the workaround last night. Editing the cardreader.cpp file this fixed it.
Happy days.
Thanks Lots.

Bob-the-Kuhn, 2017-07-03:
The patch is still preliminary.  Before making this public I'd like to get a pull request into Marlin to be sure that this is acceptable to the general population.

Bob-the-Kuhn, 2017-07-03:
I've opened PR #7227 for this issue.

closed 2018-02-09 by thinkyhead