MarlinFirmware/Marlin #6729: Switches question

Radiu, 2017-05-14:
hi i have some questions about switches
i was wondering where you install your switches X switch on left side, Y switch on the back part, Z switch on bottom of machine?
what would be best switches, mechanical or optical?
do you use 2 switches for each axis?

Roxy-3D, 2017-05-14:
Most people use 1 switch and usually they put it as a X, Y or Z Min switch.    I prefer micro-switches but a lot of people prefer optical.   This is the RepRap world...  You can do what ever you think makes sense.

Bob-the-Kuhn, 2017-05-14:
I personally prefer the A1125 Hall effect sensor.  Has same pinout as most endstop connectors, doesn't care about dirt/dust & lasts forever.

Tannoo, 2017-05-14:
I did use 3 microswitches. X, Y and Z.
I have since stopped using the Z in favor of a proximity probe.

Tannoo, 2017-05-14:
My X limit is installed on the X carraige rail on the drive end.
My Y limit is attached to the Y rail opposite the drive end.

teemuatlut, 2017-05-14:
For X and Y it's mostly up to your own reference.
Personally I use Sensorless Homing (cleaner with no extra wires) and hall effect sensors for the second printer. Optical would be my next choice, but I don't think I'll ever go back to microswitches. In the end they all work the same.
For Z axis many have upgraded to either inductive sensors or a BLTouch sensor.
Generally there's no need to use two sensors for a single axis.

mtowara, 2017-05-14:
@teemuatlut can you point me to references for Sensorless Homing?
Is that a feature of your stepperdrivers?

teemuatlut, 2017-05-14:
Yes, TMC2130 or Trinamic TRAMS required.
https://www.trinamic.com/technology/adv-technologies/stallguard/

closed 2017-06-02 by Radiu