MarlinFirmware/Marlin #6730: [Feature Request] Motorised ABL (3 Z-Steppers)

siggi60, 2017-05-14:
Hello @all,
if i remember right you announced M-ABL will be implemented in Marlin 1.1.x
Are there any plans when that will come?
In the moment i'm using Repetier, but i miss some coole Marlin stuff and so i look several times each day to read something like "we are proud to present our own M-ABL":)

Can you tell me good news or do i have to wait a long Time?

Unfortunately i don't know C++ so i can't present my own solution :(

Best regards
Siggi - who is hoping his english isn't too bad to understand :)

alexborro, 2017-05-14:
What is a Motorised ABL?

siggi60, 2017-05-14:
Simple to describe 
There are three Z-Steppers. While homing Z they work together.
Other while ABL: One Stepper is reference and the other two are going up/down during probing to reach a "non"tilted Bed.
ABL is starting -> Probing1 -> Probing2 -> Probing3. Firmware stores Probing3 and Probing1+2 going up/down ->Probing1 -> Probing2 -> Probing3 .... repeating the steps until a given tolerance is reached or cancels probing while can't reaching the tolerance.
The reason to cancel can be wrong steps, bad probe...
I hope you understand. If not: have a look on Sparkcube i.e.

To have this you must buy a CNC-Shield, RUMBA-Board or something similar to drive more Steppers.

alexborro, 2017-05-14:
And what is the advantage over the regular Bed Auto Leveling? Not to
mention the UBL because there is no mesh compensation.

Tannoo, 2017-05-14:
I see the videos... Auto traming of the bed. Well, that does require at least three Z axis steppers.

Does what Marlin's 3-point mesh leveling does, but on the hardware end of things.

Tannoo, 2017-05-14:
https://youtu.be/76bUTPKNloM

siggi60, 2017-05-14:
Yes, three point leveling. So simply add the code to control the additional steppers and all is ok^^
But i think there is a little bit more to do.

kunok, 2017-05-14:
What about also adding this also to the manual bel leveling?
I will explain myself.
After you do a 3 point bed leveling it would be interesting that the firmware calculate how much the bed support points (info that need to be added in a the config manually) need to be raised/lowered to be flat at the z = 0 level, and show that in the LCD or on the computer
Basically the same that in the video, but with info in the lcd so you can do it manually, not needing three steepers.
If something of this already exist forgive me

alexborro, 2017-05-14:
It is not that easy...

And my question is: WHY?

I mean, I see no advantages at all using a motorized system to level the
print surface when you are able to use a "regular" bed leveling.. The old
fashioned 3-points leveling do the same job.. And the new mesh leveling,
that maps the entire surfaces and compensate for a not flat surface do a
way better job...

Tannoo, 2017-05-14:
I agree. Plus, the software leveling has less parts to wear out.

kunok, 2017-05-14:
It will be always better to have the bed most leveled possible, my suggestion is just a help to manual bed leveling, even if after that you use the mesh bed leveling to compensate for a non flat surface, it would be an improvement.
And is not just calculating how is the difference in z height that are in the three support points in the calculated plane in AUTO_BED_LEVELING_3POINT and putting that in the LCD?
I don't see why would be more difficult than that.

Tannoo, 2017-05-14:
I believe that if one has a tripple z axis stepper system, Marlin should have an option to utilize it. Just like all of the other optional features.

siggi60, 2017-05-14:
Yes, an option to use it will be fine. 
And maybe there are less people using it, it's like the other stuff: Use it or not. You decide :)

Roxy-3D, 2017-05-14:
> And what is the advantage over the regular Bed Auto Leveling?

@siggi60   Alex has a very valid point.   The big problem is unless there is a very real advantage, you may be the only person that builds the hardware to do this.   And this is going to be a difficult piece of code to write and maintain.  (especially if it is solved for the general case where the 3 motors can be located anywhere on the bed.)   Even with the motorized bed leveling, if you have a print bed of any size you are still going to need some mesh based compensation system helping you.

Also...   In order to develop this kind of a system, each developer would need a machine they can code against.   So the developers will need to be convinced there is enough of a benefit to justify trying to modify a machine to do the work.  (Unless you will contribute by doing that and getting the machines into the developer's hands.)

This is an interesting problem.  But right now...   I've got my hands full tilting the UBL's Mesh based leveling system in firmware.   From my perspective...  No special hardware is required.   The code will run on anybody's printer.    And soon, it won't even require a Z-Probe.   So if I'm trying to help people, it is a lot easier to go that route.

siggi60, 2017-05-14:
You're right Roxy. Not many people use this system outside the Sparkcube Universe. But most people have to change to Repetier and then missing the sweet little things Marlin offers.
My Vulcanus isn't ready now so i have no chance to test software, but i'm working to finalise him :)
Maybe one day one of the developers is asking for testers then i'm here :)

kunok, 2017-05-14:
@Roxy-3D and what about the basic info to help bed leveling that I was suggesting? Or is already implemented somewhat? (My PSU is currently dead, so I couldn't still try the new 1.1)

That would help anybody that uses manual or automatic bed leveling systems, no more hardware required

I really want to use the new manual bed leveling to help me to get my bed as flat as possible, and then doing a ubl to compensate my non perfect flat surface

Should I made a new issue instead of talking that here?

alexborro, 2017-05-14:
@kunok , check this out:
http://marlinfw.org/docs/features/unified_bed_leveling.html

Roxy-3D, 2017-05-14:
> I really want to use the new manual bed leveling to help me to get my bed as flat as possible, and then doing a ubl to compensate my non perfect flat surface

Of course I'm biased...   But to physically level my print beds...   I just do a G29 P1 O in UBL.    I get to see the data points and decide how to raise or lower an edge or corner.    I haven't done it...  But you could also do a G29  J5 V4   and see what the plane normal vector looks like and tilt the bed to get that straight up.

Tannoo, 2017-05-14:
G29 J5 V4. -  another option in the LCD UBL menu?

Roxy-3D, 2017-05-14:
The V4 only makes sense if you have a host program like PronterFace or Repetier Host connected to the printer.   And if you are running things from the LCD Panel, that is very likely not the case.

closed 2018-02-06 by thinkyhead