MarlinFirmware/Marlin #6732: Issue: FAST_PWM_FAN compile error

TunaLatte, 2017-05-14:
Hello,

Compiling the latest Marlin with arduino ide, printrboard, with FAST_PWM_FAN enabled results in this:
```
Marlin_main.cpp: In function 'void setPwmFrequency(uint8_t, int)':
Marlin_main.cpp:11691: error: 'TIMER0A' was not declared in this scope
         case TIMER0A:
              ^
Marlin_main.cpp:11692: error: 'TIMER0B' was not declared in this scope
         case TIMER0B:
              ^
Marlin_main.cpp:11697: error: 'TIMER1A' was not declared in this scope
         case TIMER1A:
              ^
Marlin_main.cpp:11698: error: 'TIMER1B' was not declared in this scope
         case TIMER1B:
              ^
Marlin_main.cpp:11709: error: 'TIMER2A' was not declared in this scope
         case TIMER2A:
              ^
Marlin_main.cpp:11710: error: 'TIMER2B' was not declared in this scope
         case TIMER2B:
              ^
Marlin_main.cpp:11715: error: 'TIMER3A' was not declared in this scope
         case TIMER3A:
              ^
Marlin_main.cpp:11716: error: 'TIMER3B' was not declared in this scope
         case TIMER3B:
              ^
Marlin_main.cpp:11717: error: 'TIMER3C' was not declared in this scope
         case TIMER3C:
              ^
'TIMER0A' was not declared in this scope 
```
Am I missing something? I have searched all issues and did not find any note wether this is a bug, or e.g not compatible with printrboard or teensy++.
Configuration.h is stock except FAST_PWM_FAN.

Thanks.

TunaLatte, 2017-05-14:
Update, issue fixed with FAN_SOFT_PWM, are there any drawbacks or side effect on performance (cpu cycles etc)?
Thank you, it would be good to include this in sanitycheck with printrboard motherboard selection.

Bob-the-Kuhn, 2017-05-14:
Can you point me to the IDE extension you're using?

If I'm not mistaken the standard printrboard IDE extension is based off the Teensyduino extension.  Teensyduino does some things differently than the other IDEs.  

Try putting the following code at the very beginning of the setPwmFrequency function.
#if !defined(TIMER1B)  // working with Teensyduino extension so need to define the timers 
   #define TIMER0A 0
   #define TIMER0B 1 
   #define TIMER1A 7 
   #define TIMER1B 8 
   #define TIMER1C 9 
   #define TIMER2A 6 
   #define TIMER2B 2 
   #define TIMER3A 5 
   #define TIMER3B 4 
   #define TIMER3C 3 
 #endif 

I'm not sure about timer0A.  May need to make the case that handles timer0a conditional.

----

Here's the translation form to SOFT_PWM_SCALE to frequency of the soft fan pwm.

    // SOFT_PWM_SCALE to frequency:
    //
    // 0: 16000000/64/256/128 =   7.6294 Hz
    // 1:                / 64 =  15.2588 Hz
    // 2:                / 32 =  30.5176 Hz
    // 3:                / 16 =  61.0352 Hz
    // 4:                /  8 = 122.0703 Hz
    // 5:                /  4 = 244.1406 Hz

TunaLatte, 2017-05-15:
I am using the following: https://www.pjrc.com/teensy/td_download.html
Teensyduino, Version 1.36 at this time.
Arduino IDE 1.8.2.
By adding the code your provided, compilation was successful!
Thanks by the way.

Bob-the-Kuhn, 2017-05-15:
FANTASTIC!!

This is strictly a "Teensyduino does things differently" and sometimes the Marlin coders forget about that.

TunaLatte, 2017-05-16:
Hello again, after flashing the compiled firmware, fan is not working, reverting to soft pwm. To confirm I recompiled with soft pwm option to make sure I did not miss anything and it worked again.

Bob-the-Kuhn, 2017-05-16:
You grabbed the code from PR #6737 today and it compiles but the fan is NOT working if the FAST_PWM_FAN option is enabled.

With the earlier work around the fan was functional with the FAST_PWM_FAN option enabled.  Correct?

What pin is your fan connected to?

fiveangle, 2017-05-17:
To get it to compile and FAN to work with the audible Teensy HW PWM included `MarlinConfig.h` into `serial.h` and leave both _FAST_ and _SOFT_ PWM undefined in `Configuration.h` : https://github.com/MarlinFirmware/Marlin/issues/6696#issuecomment-302143233

TunaLatte, 2017-05-17:
To clarify, yes leaving fast and soft_pwm undefined, fan works with disturbing noise on my side (almost like loudspeaker running frequency tone generator, might replace fan but that isnt the root issue) Fan pin is connected normally next to the temperature sensors on Printrboard Rev.D.

Bob-the-Kuhn, 2017-05-18:
OK - got new code uploaded to the PR.

As best I can tell the strange pin mapping results in FAN0 being on a hardware PWM when BOTH FAST_PWM_FAN and FAN_SOFT_PWM are disabled.  Just the opposite of the comments in the pins_xxx.h file.  Using the M43 I command I can move the speed slider in Repetier Host and watch the PWM register change.

Please try this code out.

TunaLatte, 2017-05-23:
Hello,
Cloned and compiled commit  d252d1b from bugfix-1.1.x, didn't modify anything but enable Fast_pwm_fan, worked without any compiler error and fans works. Great work, thank you! Closing issue.

closed 2017-05-23 by TunaLatte

thinkyhead, 2017-05-23:
> As best I can tell the strange pin mapping results in FAN0 being on a hardware PWM when BOTH FAST_PWM_FAN and FAN_SOFT_PWM are disabled.

Teensy only, I assume…? That's an interesting one. I'm pretty sure these two options are incompatible anyway, so maybe a sanity check that only one is enabled will cover all cases.

TunaLatte, 2017-05-23:
Just FYI, previously, when both being undefined fan works with noisy PWM (or shorter frequency), enabling FAST_PWM broke fan control. Enabling SOFT_PWM fixed it though the PWM frequency is way too low (which is expected) Now with the commit I mentioned earlier, enabling FAST_PWM works, frequency is high (almost inaudible, ultrasonic).

Bob-the-Kuhn, 2017-05-23:
@thinkyhead - this is on Printrboard.  The pins_PRINTRBOARD.h file is just plain wrong when it comes to the fan options.