MarlinFirmware/Marlin #6734: Struggling for 3 days with M851 / Z_PROBE_OFFSET_FROM_EXTRUDER

msdavid, 2017-05-14:
Hello, I have tried to read every post but I am still unable to understand those settings and have been struggling for 3 days  (and almost 3 nights) with this issue. 
I am not able to get the nozzle at the right distance from the bed.
I am using the Asteeg X3 and use D7 for an inductive 4mm Z Prob (FIXED). All Endstops including Z Probe are working great as verified with M119. 
```
Recv: x_min: TRIGGERED
Recv: x_max: open
Recv: y_min: TRIGGERED
Recv: y_max: open
Recv: z_min: TRIGGERED
Recv: z_max: open
Recv: z_probe: TRIGGERED
```
After G28 my nozzle seats precisely on top of the bed (touching) while M114 reports: 
`Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0`

My understanding from watching "Tom's Video on Bed leveling" is that the adjustment to the nozzle height is corrected via Z_PROBE_OFFSET_FROM_EXTRUDER or M851.
However, those two setting have no effect for me.

I am new to this and very confused any help or pointers would be much appreciated.
I am using the RCBugFix branch and here are my config files:

[Configuration.h](https://drive.google.com/open?id=0B_caXmKAgEBGcDdncmpXMnY3UVU)
[Configuration_adv.h](https://drive.google.com/open?id=0B_caXmKAgEBGMlBudk9oOTRSZjA)

Thank you for your support.

Roxy-3D, 2017-05-14:
Do you REALLY have an X-Min and X-Max endstop?
Do you REALLY have an Y-Min and Y-Max endstop?
Do you REALLY have an Z-Min and Z-Max endstop?
```cpp
// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
```

msdavid, 2017-05-14:
Thank you, Roxy-3D
Yes, I do on all 6!

Roxy-3D, 2017-05-14:
The first thing to do is make sure each switch toggles.  And that they each report the correct state.   If the nozzle is in the center of the bed all endstops should be 'Open'.    But as you move to the limits, you should be able to see each one go to a 'Triggered' state.   

The same applies to the Z-Probe (which it looks like you have on Pin-7).   You should be able to manually trigger that and see the results reported in M119

If that all works...  We can move on to the next step.

msdavid, 2017-05-14:
Thank you. 
I have confirmed each of the endstops including the Z-Probe. They are all working fine and M119 reports perfectly for each test.

Roxy-3D, 2017-05-14:
Why do you have #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   ???

Almost for sure...  That needs to be a positive number if you are using an inductive probe.

msdavid, 2017-05-14:
I tried every possible value, but it makes no difference. the Probe (Sensing) is positioned at 2mm from the nozzle (physically measured). I tried setting that value too. it just happens that at the time of sharing the config file I tried back 0

Roxy-3D, 2017-05-14:
Are you aware the machine's configuration is loaded from EEPROM at startup?   You need to do a M502 to restore the machine's numbers to the Configuration.h default values.   And then do a M500 to write the values to EEPROM.    If you don't do that...  You change the Configuration.h file back and forth but you won't see any changed behavior.

msdavid, 2017-05-14:
I kept EEPROM disabled to avoid this issue while testing.
`//#define EEPROM_SETTINGS`
Is this correct?

Roxy-3D, 2017-05-14:
OK...  That should insulate you from the problem.   I think you should change the Z_PROBE_OFFSET_FROM_EXTRUDER number back to 2.0  and flash the firmware.   When you power up and connect to the printer...  Some of the fine print will tell you what that number is.    You can also do a M503 to print out system configuration information.

msdavid, 2017-05-14:
Will try once again. I just realized that I checkout RCBugfix and it seems to be RC4 for some reason. I have now check out RC8. Will reconfigure and report back.

Thank you so much for your support.

msdavid, 2017-05-14:
Same thing, the nozzle still remains touching the bed after G28 Z0.
Here is what I get when I connect to it:

```
Attempting connection at /dev/ttyUSB2...
READ: start
  Connected to machine!
READ: echo:Marlin 1.1.0-RC8
Marlin 1.1.0-RC8
READ: 
READ: echo: Last Updated: 2016-12-06 12:00 | Author: (none, default config)
Last Updated: 2016-12-06 12:00 | Author: (none, default config)
READ: Compiled: May 15 2017
READ: echo: Free Memory: 5319  PlannerBufferBytes: 1168
Free Memory: 5319  PlannerBufferBytes: 1168
READ: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X88.89 Y88.89 Z88.89 E500.00
M92 X88.89 Y88.89 Z88.89 E500.00
READ: echo:Maximum feedrates (mm/s):
Maximum feedrates (mm/s):
READ: echo:  M203 X10000.00 Y10000.00 Z10000.00 E25.00
M203 X10000.00 Y10000.00 Z10000.00 E25.00
READ: echo:Maximum Acceleration (mm/s2):
Maximum Acceleration (mm/s2):
READ: echo:  M201 X1200 Y1200 Z1200 E10000
M201 X1200 Y1200 Z1200 E10000
READ: echo:Accelerations: P=printing, R=retract and T=travel
Accelerations: P=printing, R=retract and T=travel
SENT: T0
READ: echo:  M204 P1000.00 R1000.00 T1200.00
M204 P1000.00 R1000.00 T1200.00
READ: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
READ: echo:  M205 S0.00 T0.00 B20000 X16.00 Y16.00 Z16.00 E5.00
M205 S0.00 T0.00 B20000 X16.00 Y16.00 Z16.00 E5.00
READ: echo:Home offset (mm)
Home offset (mm)
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:PID settings:
PID settings:
READ: echo:  M301 P22.20 I1.08 D114.00
M301 P22.20 I1.08 D114.00
READ: echo:Filament settings: Disabled
Filament settings: Disabled
READ: echo:  M200 D3.00
M200 D3.00
READ: echo:  M200 D0
M200 D0
READ: echo:Z-Probe Offset (mm):
Z-Probe Offset (mm):
READ: echo:  M851 Z2.00
M851 Z2.00
READ: echo:Active Extruder: 0
Active Extruder: 0
```

msdavid, 2017-05-14:
and this is what I get after G29
```
READ: Bilinear Leveling Grid:
READ:       0     1     2
READ:  0 +3.00 +3.44 +4.05
READ:  1 +2.91 +3.23 +3.70
READ:  2 +2.84 +3.08 +3.43
READ: 
READ: X:500.00 Y:406.00 Z:1.65 E:0.00 Count X: 44445 Y:36089 Z:444
```

thinkyhead, 2017-05-14:
> `M851 Z2.00`

If the sensor triggers when the nozzle is at Z=2 then you should use `M851 Z-2.00`. This number reflects the distance from the nozzle tip to the probe trigger point.

**NEGATIVE**: Nozzle is above the glass when the probe triggers.
**POSITIVE**: Nozzle is pressing down on the glass when the probe triggers.

thinkyhead, 2017-05-14:
P.S. You should be using version `1.1.1` or, better, `bugfix-1.1.x` for testing. `1.1.0-RC8` should work in this area, but it is getting long-in-the-tooth at this point.

Bob-the-Kuhn, 2017-05-14:
You're definitely NOT running RC8.  The config file has items in it that were added later.

I think you should try [Marlin 1.1.1](https://github.com/MarlinFirmware/Marlin/releases) or [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x).  They contain some bug fixes that could be the cause of your problem.  You may need to copy the machine specific items over to the new config files.

The Z offset should be negative because the probe triggers when the nozzle is above the bed.

When I see positive numbers in the bilinear compensation matrix it means the probe is triggering above the Z0 position,

I think the real problem is the Z_MIN endstop is physically too low.  The nozzle is contacting the bed before the Z_MIN endstop triggers.  Please move the endstop up until there is clearance between the bed and the nozzle after G28.  It doesn't need to be set precisely.  5-15 mm is a normal clearance.

After switching to the latest firmware and moving the endstop please do a G28 and then a G29 and post the results.

Bob-the-Kuhn, 2017-05-14:
My suggestion for setting the Z_MIN endstop is to use the **M43 E1**  endstop monitoring utility.  This software reports to the host whenever an endstop changes state and toggles the system LED.

To use it do the following:
1. enable PINS_DEBUGGING in configuration_adv.h
2. issue **M43 E1** command
3. move the head to about 5-10 mm above the bed
4. move the Z_MIN endstop up until the LED changes state
5. move the nozzle up & down a few times and confirm that Z_MIN triggers before the nozzle touches the bed

thinkyhead, 2017-05-14:
> You're definitely NOT running RC8.

I know we did update the version number for 1.1.1 and `bugfix-1.1.x` so maybe flashing is failing…
```
Attempting connection at /dev/ttyUSB2...
READ: start
  Connected to machine!
READ: echo:Marlin 1.1.0-RC8
Marlin 1.1.0-RC8
```

Bob-the-Kuhn, 2017-05-14:
Sorry - I hadn't notice that **Marlin 1.1.0-RC8** existed.

What's it's purpose in life?

msdavid, 2017-05-15:
Thank you all for your responses and apologies for the late reply. (I am in Singapore TZ)
I will now try with 1.1.1 and report.

msdavid, 2017-05-15:
Thank you for the great suggestion on the Z_MIN config Bob-the-Kuhn. I am trying now. Will post results.

msdavid, 2017-05-15:
Thank you so much for your great support. 
Following your instructions, I was able to get the ABL working.

closed 2017-05-15 by msdavid

thinkyhead, 2017-05-16:
> Sorry - I hadn't notice that Marlin 1.1.0-RC8 existed. What's its purpose in life?

It was out in the public eye for review for a few months, giving us a chance to discover new bugs and continue to patch up the RC for release.

pierrealexb, 2017-05-18:
Hey, @msdavid can you give more detail about the solution you use. I have the same problem using Marlin 1.1.0-1. I solved using M851 but Z_PROBE_OFFSET_FROM_EXTRUDER is not working at all...
I prefer to use Z_PROBE_OFFSET_FROM_EXTRUDER because I can add comment detail in the custom firmaware as many people work with me on this project.
Many thanks!

pierrealexb, 2017-05-18:
Hey guys, shame on me, I forgot to reset EEPROM, I really thaugh I did... Now it works well!
Thanks and sorry