MarlinFirmware/Marlin #6739: Issue: Compile Error when enable RGB LED & PWM for Motor driver not work

lacrimosa2k2, 2017-05-15:
The controler fan dosnt aceppt a value of 180, only by 255 it works at full speed

When i enable the rgb like this in the configuration.h:
#define RGB_LED
//#define RGBW_LED
#if ENABLED(RGB_LED) || ENABLED(RGBW_LED)
  #define RGB_LED_R_PIN 44
  #define RGB_LED_G_PIN 64
  #define RGB_LED_B_PIN 59
  #define RGB_LED_W_PIN -1
#endif

the ardouino Compiler give the error:

Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\Marlin_main.cpp: In function 'void gcode_M190()':

Marlin_main.cpp:6940: error: expected 'while' before '(' token

       if (wants_to_cool) {

          ^

Marlin_main.cpp:6940: error: expected ';' before '{' token

       if (wants_to_cool) {

                          ^

Marlin_main.cpp:6944: error: 'temp' was not declared in this scope

           if (old_temp - temp < MIN_COOLING_SLOPE_DEG_BED) break;

                          ^

Marlin_main.cpp:6944: error: break statement not within loop or switch

           if (old_temp - temp < MIN_COOLING_SLOPE_DEG_BED) break;

                                                            ^

Marlin_main.cpp:6946: error: 'temp' was not declared in this scope

           old_temp = temp;

                      ^

sketch\Marlin_main.cpp: At global scope:

Marlin_main.cpp:6950: error: expected unqualified-id before 'while'

     } while (wait_for_heatup && TEMP_BED_CONDITIONS);

       ^

Marlin_main.cpp:6952: error: expected unqualified-id before 'if'

     if (wait_for_heatup) LCD_MESSAGEPGM(MSG_BED_DONE);

     ^

In file included from sketch\Marlin.h:32:0,

                 from sketch\Marlin_main.cpp:223:

ultralcd.h:64: error: expected unqualified-id before ')' token

   #define LCD_MESSAGEPGM(x) lcd_setstatuspgm(PSTR(x))

                                              ^

sketch\Marlin_main.cpp:6952:26: note: in expansion of macro 'LCD_MESSAGEPGM'

     if (wait_for_heatup) LCD_MESSAGEPGM(MSG_BED_DONE);

                          ^

In file included from sketch\Marlin_main.cpp:223:0:

Marlin.h:358: error: expected unqualified-id before 'do'

   #define KEEPALIVE_STATE(n) do{ busy_state = n; }while(0)

                              ^

sketch\Marlin_main.cpp:6953:5: note: in expansion of macro 'KEEPALIVE_STATE'

     KEEPALIVE_STATE(IN_HANDLER);

     ^

Marlin.h:358: error: expected unqualified-id before 'while'

   #define KEEPALIVE_STATE(n) do{ busy_state = n; }while(0)

                                                   ^

sketch\Marlin_main.cpp:6953:5: note: in expansion of macro 'KEEPALIVE_STATE'

     KEEPALIVE_STATE(IN_HANDLER);

     ^

Marlin_main.cpp:6954: error: expected declaration before '}' token

   }

   ^

exit status 1
expected 'while' before '(' token

here my config
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1000636/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1000637/Configuration_adv.txt)

Bob-the-Kuhn, 2017-05-15:
I'm not getting those errors when I drop your files into [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x ) from yesterday.

What flavor of Marlin are you using?

How did you determine that the controller fan only ran at full speed?

FYI - I added **#define PROPORTIONAL_FONT_RATIO 1.0** to configuration_adv.h to get it to compile.

lacrimosa2k2, 2017-05-16:
okay now the led works with the bugfix.

the controler fan at this code
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 5  // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255       // 255 == full speed
#endif
now only spin up at 255
i use marlin on arduion 2650 with ramps 1.4

Bob-the-Kuhn, 2017-05-16:
I found the code that runs the CONTROLLER_FAN.

CONTROLLER_FAN is either off or it's running at the speed set by CONTROLLERFAN_SPEED.

There's no user control of the fan.  It's all automatic.

The M106 & M107 commands apply only to the extruder fans.

lacrimosa2k2, 2017-05-17:
in the rc8 it runs, no M commands, only in the configuration_adv set the speed for the controler fan

lacrimosa2k2, 2017-05-17:
now i load the actual bugfix and it works now :D thanks

closed 2017-05-17 by lacrimosa2k2