MarlinFirmware/Marlin #6740: How to travel over the bed keeping the same height

batata004, 2017-05-15:
Hi,

I made my heated bed very unlevel, I mean, it's very angled and not perfectly horizontal. I started the BED LEVELING procedure and ok, it worked fine.

NOW I want to make my nozzle travel the bed in XY plane keeping the same relative height to the bed. How can I do that?

If I click in the LCD -> MOVE AXIS -> X (after bed leveling), if I increase/decrease the value it just travels the X axis to the right and to the left. It does not compensate the unleveled surface. How do I make my nozzle to travel over the bed keeping the same height so I can check visually that the bed leveling is indeed being done correctly?

Tannoo, 2017-05-15:
Use a bed leveling option. I would use bilinear or UBL.

batata004, 2017-05-15:
@Tannoo  I am of course alreay using it. I am using BAL with servo and even after auto leveing when I move X axis to the right/left it does not try to keep the same relative distance from the nozzle to the bed.

Roxy-3D, 2017-05-15:
In the UBL case...   You need to have the system activated.   G29 A     And you need to have the nozzle below the 'Fade Height' which can be set with G29 F 10.0   (Or if you turn off the Fade Height... it should correct regardless of the height G29 F 0.0 )

batata004, 2017-05-15:
I dont think I made myself clear, sorry guys.

I am not using UBL, I am using auto balacing using servo probe. I am using LINEAR mode.

I would like this: right after I bed level (LCD -> BED LEVEL -> BED LEVEL) I would like to go to PREPARE -> MOVE AXIS -> X and be able to move the carriage at X to the right/left BUT doing it in a way that the nozzle keeps the same distance from the bed.

For example, if I have a bed that is angled at 20 degrees, when the nozzle moves to the right/left it should move in a 20 degree steep.

Roxy-3D, 2017-05-15:
It should do that after the G28 / G29 pair.

batata004, 2017-05-15:
So you are saying that if I send G28 and after that G29 to my printer, if I move my X carriage it should keep the nozzle at the same distance from the bed in every lateral position? Cause that's not happening with me :(

Roxy-3D, 2017-05-15:
Did you get a correction matrix at the end of the G29?   What does it say?

batata004, 2017-05-15:
How do I know if I got a correction matrix?

Also one other qustion: I am using LINEAR, do you think tha using BILINEAR will bring any benefit? I am using a glass that looks preeety flat in my heated bed. Do you still would believe that BILIENAR would have any benefit?

Roxy-3D, 2017-05-15:
> Do you still would believe that BILIENAR would have any benefit?

I never expressed a belief about Bi-Linear in this discussion.

batata004, 2017-05-15:
@Roxy-3D when I get home I will do exactly as you said.

I didnt say that you expressed BILINNEAR is better. I just made a queston, I didnt want to offend anyone :(

closed 2017-05-16 by batata004

thinkyhead, 2017-05-17:
> using BILINEAR will bring any benefit?

It (or UBL) might help. Even if the bed itself is perfectly flat, the machine may not move in perfectly straight lines that always stay at the same Z distance from the bed. Mesh-based leveling —like BILINEAR, MBL, or UBL— has the added benefit of compensating not only for a tilted plane, but for any imperfections of the mechanical system too. So, give it a try!