MarlinFirmware/Marlin #6741: Issue: Compile error in branch bugfix 1.1.x for delta with UBL enabled

joris57, 2017-05-15:
When compiling for a delta with UBL enabled, following error message is generated:

sketch\ubl_motion.cpp: In function 'bool ubl_prepare_linear_move_to(const float*, const float&)':

ubl_motion.cpp:675: error: assignment of read-only variable 'cx'
           cx += dx_seg;
              ^
ubl_motion.cpp:676: error: assignment of read-only variable 'cy'
           cy += dy_seg;
              ^

exit status 1
assignment of read-only variable 'cx'

Roxy-3D, 2017-05-15:
Those variables should NOT be const.   If you change the code to:
```cpp
       const float x0 = pgm_read_float(&(ubl.mesh_index_to_xpos[cell_xi  ])),  // 64 byte table lookup avoids mul+add
                    y0 = pgm_read_float(&(ubl.mesh_index_to_ypos[cell_yi  ])),  // 64 byte table lookup avoids mul+add
                    x1 = pgm_read_float(&(ubl.mesh_index_to_xpos[cell_xi+1])),  // 64 byte table lookup avoids mul+add
                    y1 = pgm_read_float(&(ubl.mesh_index_to_ypos[cell_yi+1]));  // 64 byte table lookup avoids mul+add

float            cx = rx - x0,   // cell-relative x
                    cy = ry - y0;   // cell-relative y
```

Maglin, 2017-05-16:
I'm getting a similar error but it looks like the above changes are already in ubl_motion.cpp

```
Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\ubl_motion.cpp: In function 'bool ubl_prepare_linear_move_to(const float*, const float&)':

ubl_motion.cpp:604: error: 'z_x0y0' was not declared in this scope

               z_x0y0 = ubl.z_values[cell_xi  ][cell_yi  ],  // z at lower left corner

               ^

ubl_motion.cpp:605: error: 'z_x1y0' was not declared in this scope

               z_x1y0 = ubl.z_values[cell_xi+1][cell_yi  ],  // z at upper left corner

               ^

ubl_motion.cpp:606: error: 'z_x0y1' was not declared in this scope

               z_x0y1 = ubl.z_values[cell_xi  ][cell_yi+1],  // z at lower right corner

               ^

ubl_motion.cpp:607: error: 'z_x1y1' was not declared in this scope

               z_x1y1 = ubl.z_values[cell_xi+1][cell_yi+1];  // z at upper right corner

               ^

ubl_motion.cpp:619: error: 'z_xmy1' was not declared in this scope

         const float z_cxy1 = z_x0y1 + z_xmy1 * cx,          // z height along y1 at cx

                                       ^

ubl_motion.cpp:622: error: 'z_cxyd' was not declared in this scope

               float z_cxym = z_cxyd * (1.0 / (MESH_Y_DIST)),  // z slope per y along cx from y0 to y1

                              ^

ubl_motion.cpp:634: error: 'z_cxcy' was not declared in this scope

           z_cxcy += ubl.state.z_offset;

           ^

Multiple libraries were found for "LiquidCrystal.h"
 Used: C:\Users\jerem\OneDrive\Documents\Arduino\libraries\LiquidCrystal
 Not used: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal
exit status 1
'z_x0y0' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Roxy-3D, 2017-05-16:
It is good to see people bringing up the Delta version of UBL.   But seriously...   Keep a close eye on your printer at each step.   We don't have tons of test time on the Delta version yet.   And the Delta stuff is way more complicated than the Cartesian stuff.    

With that said...   UBL was actually designed to help low quality Delta's.   But as it turned out, Cartesian was a much easier place to debug code.    So....  Cartesian happened first.

If you see a problem...   either myself or @oldmcg will try to figure it out and keep you happy!

Maglin, 2017-05-16:
I don't have a fixed probe on my delta so I was really wanting UBL to at a minimum save the bed mesh to EEPROM.  It just won't compile and I'm not savvy on coding.  I'm hoping to be able to compile this soon.  I do enjoy wringing out new code/functions.

Roxy-3D, 2017-05-16:
> It just won't compile and I'm not savvy on coding. I'm hoping to be able to compile this soon. I do enjoy wringing out new code/functions.

I think it compiles now...   I fixed those problems this morning.    But even though you don't have a probe...   If you declare and claim you have a probe...  You should be able to bring up UBL.  Instead of doing a G29 P1 to probe the mesh, you would just do a G29 P0 to set the mesh to all 0.0000's.   And then jump to the G26 Mesh Validation Pattern.    You would want to be ready to press and hold the encoder wheel to stop it if the nozzle gets too close to the glass.   But if you start with the nozzle in the center of the bed and do a G26 P O2.5....  The nozzle should work from the center outward and you should have time to stop it if it gets too close.

Maglin, 2017-05-16:
I'm fixing a probe to the effector just to probe the mesh.  Then I plan to tune the mesh.  Hopefully, it will work out.  I'm not too sure what will happen when it travels outside of the mesh.

Roxy-3D, 2017-05-16:
I believe you will not leave the mesh.  The way oldmcg set things up...  There are some 'wasted' mesh points in the corners of the mesh.   You can't ever get to them.   But the left and right edges of the printable radius should still be within the mesh.

Maglin, 2017-05-16:
So I'm still getting the following error when I'm compiling.
This is doing nothing but moving the Kossel XL config's over and just uncommenting #define AUTO_BED_LEVELING_UBL
```
Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\ubl_motion.cpp: In function 'bool ubl_prepare_linear_move_to(const float*, const float&)':

ubl_motion.cpp:604: error: 'z_x0y0' was not declared in this scope

               z_x0y0 = ubl.z_values[cell_xi  ][cell_yi  ],  // z at lower left corner

               ^

ubl_motion.cpp:605: error: 'z_x1y0' was not declared in this scope

               z_x1y0 = ubl.z_values[cell_xi+1][cell_yi  ],  // z at upper left corner

               ^

ubl_motion.cpp:606: error: 'z_x0y1' was not declared in this scope

               z_x0y1 = ubl.z_values[cell_xi  ][cell_yi+1],  // z at lower right corner

               ^

ubl_motion.cpp:607: error: 'z_x1y1' was not declared in this scope

               z_x1y1 = ubl.z_values[cell_xi+1][cell_yi+1];  // z at upper right corner

               ^

ubl_motion.cpp:619: error: 'z_xmy1' was not declared in this scope

         const float z_cxy1 = z_x0y1 + z_xmy1 * cx,          // z height along y1 at cx

                                       ^

ubl_motion.cpp:622: error: 'z_cxyd' was not declared in this scope

               float z_cxym = z_cxyd * (1.0 / (MESH_Y_DIST)),  // z slope per y along cx from y0 to y1

                              ^

ubl_motion.cpp:634: error: 'z_cxcy' was not declared in this scope

           z_cxcy += ubl.state.z_offset;

           ^

exit status 1
'z_x0y0' was not declared in this scope
```

Roxy-3D, 2017-05-16:
Yikes!!!!   A typo snuck into the code!   Please change this code block to remove the first ;
```cpp
        float cx = rx - x0,   // cell-relative x
              cy = ry - y0,   // cell-relative y
              z_x0y0 = ubl.z_values[cell_xi  ][cell_yi  ],  // z at lower left corner
              z_x1y0 = ubl.z_values[cell_xi+1][cell_yi  ],  // z at upper left corner
              z_x0y1 = ubl.z_values[cell_xi  ][cell_yi+1],  // z at lower right corner
              z_x1y1 = ubl.z_values[cell_xi+1][cell_yi+1];  // z at upper right corner
```
I'll get this updated soon.

Roxy-3D, 2017-05-16:
The fix will be merged in 10 minutes when Travis says "OK!"

oldmcg, 2017-05-16:
My still-pending PR removed these incorrect const qualifiers on those cx/cy variables that someone else added.

Roxy-3D, 2017-05-16:
Oh!  I didn't see that....  I fixed those this morning...  But I fat fingered the key board and turned a , into a ; and those assignments caused scope issues (because they were no longer declarations).   But in a minute or two, that will be fixed too...

oldmcg, 2017-05-16:
PR 6705:
fix fade_height, lost during some previous merge
**fix float cx,cy which are not const**
move repeated z_cxcy calc line inside loop
style fixes and comment fixes/alignment
add UBL_DELTA_ON_CARTESIAN to allow finer grained UBL segmentation on cartesians

Roxy-3D, 2017-05-16:
Is 6705 all checked out and ready to merge?   I'm not sure it can still merge because I made changes in the same areas it did.   What I can do is pull out the pieces I have not already changed and make a new Pull Request for these.    

I knew this Pull Request was happening...  But I stayed out of it because you were working with ThinkyHead.   I guess I assumed everything was already merged.

Maglin, 2017-05-16:
Roxy-3D thank you for your help.  It's compiled and I'm testing it now.  I had gone to Bi-Linear just to check my probe out.  Again thank you.

Roxy-3D, 2017-05-16:
You are very welcome.   In a couple of minutes https://github.com/MarlinFirmware/Marlin/pull/6705  will be merged with more good stuff.   You might want to hold off ten minutes and grab that.

Maglin, 2017-05-16:
I'm loving the Marlin Community....  I'm already a large fan of the RRF guys.  Thank you....

oldmcg, 2017-05-16:
@Maglin, sorry you got caught up in the merge process for UBL on delta.  It's new and I'm very interested to hear how it works or doesn't work for you.  There are still some rough edges -- like the G26 test pattern is not yet delta-tuned, so it will leave some unfinished-looking edges, but it WILL NOT MOVE OR PRINT outside the edge of your DELTA_PRINTABLE_RADIUS no matter what you do.  The mesh dimensions should automatically define a square the encompasses printable radius -- it will not go into the unreachable corners.  I set my UBL_MESH_INSET to zero which lets me get valid mesh points at the extreme N,S,E,W points on the bed.  Use an odd number for GRID_MAX_POINTS_X/Y so you get a point at exactly 0,0.  I use 11x11 grid for my small 70mm radius bed.  It will go up to 15x15 if you want very tight grid and don't mind it taking a long time to probe all the points.  After probing (manual or auto) all the reachable grid points on the bed, use G29 P3 (interpolate unreachable points) several iterations until G29 O shows the mesh is fully defined even to the corners.  I will fix this soon to interpolate all the points with single G29 P3 instead of manual iteration.  Also if you intend to use G29 T (three-point), be sure to define UBL_PROBE_PT_xxx to be valid coordinates within your DELTA_PRINTABLE_RADIUS.  They don't get compile-time validated yet, and if they are off the bed then it simply won't go there and G29 T will produce error message.

oldmcg, 2017-05-16:
@Maglin, recommend you wait for one more important UBL-on-delta fix.

Maglin, 2017-05-17:
@oldmcg Thank you for the help.  I'm totally new to mesh bed leveling and deltas for that matter.  But I can do testing.  I'm just now getting the latest built and will do some testing.  Is there a way to put your mesh grid into a configuration file?  That would make life very easy if you have a fixed bed as I do when changing the fw.

Roxy-3D, 2017-05-17:
Not into a configuration file...   You can save the mesh to a user defined slot in the EEPROM with a G29 S 1 command.  (That saves to slot 1)   But you can also 'save' the mesh to your computer by doing a G29 S -1.  It will print the mesh in a form that you can save it to a file....  And later load and print that file.   

We are trying not to invalidate people's mesh data unless we don't have a choice.  But occasionally, we do tell people to 'Save your mesh to your computer with a G29 S-1 prior to doing this update."

Anyway...  If you get a finely tuned mesh, it is a wise idea to have a copy on your computer so you can get it back any time you want.

Maglin, 2017-05-17:
@Roxy-3D Thank you.  I'll be sure to do that.

Roxy-3D, 2017-05-17:
@oldmcg    I found the Travis commit I was talking about...   

https://github.com/MarlinFirmware/Marlin/pull/6738/commits/dd75d6d8a99863cf0261e956c6c45ecedea5e79b

joris57, 2017-05-18:
Issue resolved.

closed 2017-05-18 by joris57