MarlinFirmware/Marlin #6746: [Discussion] Z-Lift During Travel

tcm0116, 2017-05-15:
I've been thinking about this concept today, and I wanted to throw it out there to get some feedback on if others think it might be useful. Over the weekend, I was playing around with the Z Lift settings in Cura, which are 1mm by default. When attempting a print with that value, my print ended up with stalagmites at each of the retraction points because it was essentially pulling out the molten plastic as it lifted the nozzle.

It seems to me that there are two reasons why people like to use Z Lift:

1. To prevent the nozzle from dragging across the part and possibly knocking small structures over.
2. To prevent ooze from the nozzle from getting cleaned off by an otherwise nice looking perimeter, resulting in zits.

In my case, I was trying to solve the second issue as I was getting zits on my part due to ooze from the nozzle being wiped off on the part. As I was thinking about this, it occurred to me that a compromise might be to perform the Z Lift during the travel move as opposed to before. This would give the benefit of wiping the nozzle during the first part of the travel, but then hopefully being high enough at the end of the travel to prevent any ooze from being wiped off and forming a zit.

From an implementation standpoint, I'm not certain how difficult it would be to accomplish. At a quick glance, it seems like it might be as simple as not calling `prepare_move_to_destination()` in the `retract()` function in **Marlin_main.cpp** when `retracting == true`, which should essentially queue the Z lift as part of the next move. I'd have to try it out any see.

Any thoughts on this?

thinkyhead, 2018-02-06:
I would say this should be left up to the slicer. Sometimes you do want ordinary Z lift, and sometimes you want combing, or combination moves with movement and retract combined. Slicers seem to be getting smarter about this.

tcm0116, 2018-02-06:
Sounds reasonable

closed 2018-02-06 by tcm0116