MarlinFirmware/Marlin #6747: [Feature Request] Invert bed pin logic

fabio-pulcini, 2017-05-15:
hi!
is possible now (or will be possible) invert logic of bed pin? i use a relay for my bed and when arduino start my bed pin is activate and it is a problem..
thanks!

Bob-the-Kuhn, 2017-05-15:
What relay are you using?

How did you connect it to the controller?

fabio-pulcini, 2017-05-15:
i use a relay like this [https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcQIFP7Vj-8W0P1D4bwdDknZ0RrDbCqDYXO1DsBdRYxmzbZH2oPd](url).
i controller my AC bed by digit pin 6 (PWM).
my problem is when arduino start pin 6 is on..

Tannoo, 2017-05-16:
Pics are good enough, can you provide a link to the one you actually bought?

Bob-the-Kuhn, 2017-05-16:
Please ZIP your config files and drop them on your next post.

If you're using a RAMPs based controller then the heated bed could be attached to a different pin.

Actually, you can verify it yourself.  In Configuration_adv.h enable PINS_DEBUGGING and then issue a M43 I command.  That should list all your pins, the signals attached to them and what state they are in.

fabio-pulcini, 2017-05-16:
this is my configuration.
after i will try M43 command!
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1004772/Configuration.zip)

Bob-the-Kuhn, 2017-05-16:
It looks like you're running Marlin 1.0.2.  Is that correct?

fabio-pulcini, 2017-05-16:
it is marlin 1.0.1

Bob-the-Kuhn, 2017-05-16:
It took a LONG time but I finally ported M43 over to 1.0.2.2.

It says that pin 6 is tied to HEATER_BED_PIN, that it's an output and it's set to low unless you turn the heater on.

It's finally dawning on me.  You have one end of the relay tied to +5V and the other tied to pin 6.  When pin 6 is low the relay will be energized and the heater will be on.

Move the relay connection from +5V to ground.  That should fix your problem.

fabio-pulcini, 2017-05-16:
are you sure that with this specific relay i can invert 5v to ground? it has got a small circuit..
i want to put my relay to NO configuration and when my arduino start my pin must to be slow

Bob-the-Kuhn, 2017-05-16:
If you give me the part number of the relay or the link to where you bought it I expect that I can answer that question.

Chances are good that it'll work.  The Mega2560 pins are spec'd to sink & source the same amount of current.

The alternative is to put a FET in between the relay and pin 6.

fabio-pulcini, 2017-05-17:
my relay is like this:
[https://it.aliexpress.com/item/5-pcs-Free-Shipping-1-Channel-Supports-High-and-Low-Trigger-5V-Relay-Module-with-Optocoupler/32454224659.html?spm=2114.010208.3.110.1zh5ch&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10068_5160013_10136_10137_10157_10060_10138_10155_10062_10156_10154_10056_10055_10054_10059_100032_100033_100031_10099_10103_10102_10096_10147_10052_10053_10050_10107_10142_10051_5150013_5180015_10084_10083_10080_10082_10081_10177_10110_10111_10112_10113_10114_10181_10183_10182_10185_10184_10078_10079_10073_10123,searchweb201603_1,ppcSwitch_5&btsid=58e433a7-83c9-4893-8f7e-e218da6672ca&algo_expid=0990aa67-4cf6-40bd-a001-e4fef671cf4f-16&algo_pvid=0990aa67-4cf6-40bd-a001-e4fef671cf4f](url)

Bob-the-Kuhn, 2017-05-17:
Here's the best I could find on that module.
https://drepart.en.alibaba.com/product/60451429598-802072120/Low_Level_Triger_with_Optocoupler_Relay_Module_12V_for_Arduino.html?spm=a2700.8304367.0.0.TiY4mj

It says it's optocoupled so you can drive it directly from any Mega2560 pin.  It also allows you to set the active level as low or high.

I suggest you start looking for a solid state relay.  Mechanical ones like this will eventually wear out.

fabio-pulcini, 2017-05-21:
with my relay you can choose if you want activate it when pin is low or hight (now i know it thanks bob!).
but it will be good if you can do it by software!
thanks!

Bob-the-Kuhn, 2017-05-21:
There isn't a configuration option for this.  Any software solution would require custom code.

The jumper on the module is the only hardware method available to invert the signal.  My earlier comments don't apply to this module.

thinkyhead, 2017-05-21:
IOW — it would need to invert the bed PWM, which could also be done by inverting the duty cycle.

However, the result would be that the Arduino would have to keep pumping 5V into the relay to keep it from heating the bed, and it would be better if it had the opposite fail condition. That is, if it got disconnected, it shouldn't start to heat the bed.

For that reason, I strongly disfavor this feature request and advocate for pebble bed reactors.

boelle, 2019-02-18:
@thinkyhead i agree

@fabio-pulcini have you found a "fix" ?

ellensp, 2020-08-24:
Non supported version of marlin.

closed 2020-08-24 by ellensp