MarlinFirmware/Marlin #6753: Issue: Unexpected behaviour after G28 in branch bugfix 1.1.x for delta with UBL enabled

joris57, 2017-05-16:
When entering G0 Z0 F6000 after G28, no movement happens, although both he display and M114 have the Z coordinate at 0. A subsequent G0 Z0 F6000 moves the nozzle to the correct position.

```
Send: G28
Recv: X:0.00 Y:0.00 Z:270.82 E:0.00 Count X:45611 Y:45611 Z:45611
Recv: ok
[...]
Send: G0 Z0 F3000
Recv: ok BUT NO PHYSICAL MOVEMENT OCCURS (see also 'Count' values)
[...]
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:45611 Y:45611 Z:45611
Recv: ok
[...]
Send: G0 Z0 F3000
Recv: ok
[...]
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:18529 Y:18529 Z:18529
Recv: ok
[...]
```

In another case, the nozzle stops at mid-height and entering the command a second time finishes the move (again note the 'Count' values)

```
Send: G28
Recv: X:0.00 Y:0.00 Z:270.82 E:0.00 Count X:45611 Y:45611 Z:45611
Recv: ok
[...]
Send: G0 X1 Y0 Z1 F3000
Recv: ok
[...]
Send: M114
Recv: X:1.00 Y:0.00 Z:1.00 E:0.00 Count X:27593 Y:27652 Z:27623
Recv: ok
[...]
Send: G0 X1 Y0 Z1 F3000
Recv: ok
[...]
Send: M114
Recv: X:1.00 Y:0.00 Z:1.00 E:0.00 Count X:18585 Y:18673 Z:18629
Recv: ok
[...]
```

oldmcg, 2017-05-16:
I just picked up the latest code and I'm seeing the same thing -- first G0 does nothing after G28.  I'll try to figure out what's going on.

Roxy-3D, 2017-05-16:
@oldmcg   I have the Delta Configuration.h files fixed for M48....   I'm not ready to push them upstream yet...  But you can grab them here:  https://github.com/Roxy-3D/Roxy_BugFix_v1.1.0-v12

oldmcg, 2017-05-16:
@joris57, @Roxy-3D, @thinkyhead, I found the problem with G0 not moving.  

During the ubl_delta merge cleanup, my COPY_XYZE macro usage was replaced by simple COPY macro usage which uses sizeof operator, which for array parameter is only size of one element of an array, not the full size of the array, so it was not copying all the coordinates.  I'll have a PR to fix shortly.

joris57, 2017-05-19:
Issue resolved.

closed 2017-05-19 by joris57