MarlinFirmware/Marlin #6754: Issue: Compile error in branch 1.1.x (and bugfix) when activating M48 for delta

joris57, 2017-05-16:
```
sketch\Marlin_main.cpp: In function 'void gcode_M48()':

Marlin_main.cpp:6275: error: 'DELTA_PROBEABLE_RADIUS' was not declared in this scope

                   DELTA_PROBEABLE_RADIUS / 8, DELTA_PROBEABLE_RADIUS / 3

                   ^

exit status 1
'DELTA_PROBEABLE_RADIUS' was not declared in this scope
```

Roxy-3D, 2017-05-16:
@joris57    I have verified what you said with the Kossel-XL files...   I'm trying to correct it.  But it is mixed up with other changes I have going.    In an hour or two I'll have the corrected file merged for you to grab.

If you are willing to flash a second time in an hour or two, you can just turn of M48 for now knowing soon you can turn it on.

oldmcg, 2017-05-16:
@joris57, for now, in configuration.h, just after #define DELTA_PRINTABLE_RADIUS, add #define DELTA_PROBEABLE_RADIUS DELTA_PRINTABLE_RADIUS.

I will fix this in the delta sample configuration.h files but the final fix might be a little different.

Roxy-3D, 2017-05-16:
@oldmcg    What I just merged is 'Reasonable'.  But you may have better ideas on how it should be done.

joris57, 2017-05-18:
Issue resolved.

closed 2017-05-18 by joris57