MarlinFirmware/Marlin #6755: SD card problem

batata004, 2017-05-16:
Is it just me or any of you are also experiencing problems with SD card in the latest stable version of Marlin?

In the previous version I never had problems with SD card, now I have to precisely move the SD card inside the slot of the LCD. If I move it a little bit it says REMOVED (in the LCD display)...

I took a look at the date of changes in some files of Marlin but I couldnt know for sure if any significant change was made in the SD card detector. Could you please double check if any significant change was made in the SD card detection code?

akaJes, 2017-05-16:
Hi
if your SD card has 1 or 2 class speed.
It can happen with a lot of files on the card
The best way is to use a card with  4 or 10 class speed

batata004, 2017-05-16:
I will check that when I get home but in the previous version I had not this problem.

It's a 256MB SD card. The problem is actually not reading files, it's that the LCD says CARD REMOVED, CARD INSERTED, CARD REMOVED... I cant find a good "insert position" to make sure it shows CARD INSERTED every time.

Tannoo, 2017-05-16:
Possible bad card reader.

Tannoo, 2017-05-16:
Or, something got into the slot and the card cannot quite go in all the way to make good contact.

Roxy-3D, 2017-05-16:
And some of the electronics have the SD-Detect stuff wired up wrong (or not wired up).   It is worth while to do a quick search and see if your electronics have any issues.

Tannoo, 2017-05-18:
So... What did you find?

serhattural, 2017-11-16:
I have exactly the same problem as batata04 mentioned. It started after update Marlin 1.1.6. I was no problem before. I think it is about software. It doesnt interrupted when I made pressure on sides of SD Card and cables. But it pauses on random times. Additionaly one time I saw CARD REMOVED, CARD INSERTED, CARD REMOVED... under a second.

batata004, 2017-11-16:
@serhattural it looks like a problem in the firmware. Two other friends that bought their 3d printer reprap style from other vendores have this exactly same behavirous randomly. I first thought my hardware might be with problem but after my 2 friends reported the same problem I simply had to deal with it since no fix was provided.

Tannoo, 2017-11-17:
I have had a card in my RRFG for quite some time. I haven't seen any randomness.

Try bugfix-1.1.x or bugfix 2.0.x and see if you still have issues.

Tannoo, 2017-11-17:
1.1.6 issues are addressed in bugfix-1.1.x.

closed 2018-02-09 by thinkyhead

WoodrowRobocats, 2018-04-22:
I have a similar issue with 1.17, but in my case I get the error message mid way through large prints sent from Octo-print rather than printing from the SD card. and only when I have NO card in the slot!!!
I would not expect the SD card state change to be monitored/updated when not actually printing from the SD card, and certainly not when printing from USB. I do not remove a card, I just get the random messages, Now, if I insert a card (even half way through a print) the messages stop and the print runs a LOT faster !!!
Will try to find a CR10S binary for 118 to see if it is fixed there,

thinkyhead, 2018-04-23:
There may be electrical noise giving false signals on the SD detect line. You may want to open up the control box and make sure that all the wires are moved away from the PSU and other sources of electrical noise. https://www.youtube.com/watch?v=tP9pVNBAH4k