MarlinFirmware/Marlin #6756: BLTouch not responding to M280 commands - Pin assignments incorrect?

SpeedGP, 2017-05-16:
Hello, I've been struggling with this for about a week now and am out of ideas. The main issue is when the printer is turned on, the BLTouch goes through the self test, but that's it (it will go into alarm mode if I hold the pin). It will not respond to "Reset BLTouch" through the LCD, or any of the M280 The main difference between my setup is I have a Fan Extender using pins D11 and D6, so I've moved the Servo0 pin to pin D0. I've also tried D1, and/or changing the Servo0 to Servo1-3 with no success either. Just to clarify, the command I am currently using is M280 P0 S10 (or S90).

I've enabled #define BLTouch and set NUM_SERVOS 3, Z_ENDSTOP_SERVO_NR 0, set the servo pins
#ifdef IS_RAMPS_13
#define SERVO0_PIN        7 // RAMPS_13 // Will conflict with BTN_EN2 on LCD_I2C_VIKI
#else
#define SERVO0_PIN       1
#endif
#define SERVO1_PIN          0
#define SERVO2_PIN          5
#define SERVO3_PIN          4

Set fan pins (Fan 6 works perfect)

#ifndef FAN_PIN
#define FAN_PIN 6      // IO pin. Buffer needed
#endif

(I don't have Controller fan setup currently, for some reason pin 11 is set to high all of the time, or my Fan Extender board is half burnt out)

#define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
#define CONTROLLERFAN_SECS 60 //How many seconds, after all motors were disabled, the fan should run
#define CONTROLLERFAN_SPEED 255  // == full speed

#ifndef FAN_PIN
  #define FAN_PIN -1
#endif
#ifndef FAN1_PIN
  #define FAN1_PIN -1
#endif
#ifndef FAN2_PIN
  #define FAN2_PIN -1
#endif
#ifndef CONTROLLERFAN_PIN
  #define CONTROLLERFAN_PIN  -1
#endif

My setup is a Prusa i3 Steel with a Ramps 1.4 EEB (board 44) running 1.1.0 RC8 12-6-2016. It is a genuine BLTouch Classic from Antclabs. And like I said before, Pin D11 seems to be set high, I'm not sure if this has something to do with the BLTouch or not. I assume I am changing all of the right config and pin assignments.

Please let me know if there is anything I can try or anything I've done blatantly wrong. I am waiting until someone asks to upload full config.h and config_adv.h

Thank you very much,

James

Roxy-3D, 2017-05-16:
Have you tried just turning on #define BLTOUCH option Configuration.h ?  It should default everything to reasonable values for a RAMPS board.   You just plug in the wires to Servo-0 and with that option enabled, the 'Right' things should happen.

Bob-the-Kuhn, 2017-05-16:
Even worse than uploading the config files - I also want the pins_RAMPS.h file.  Please ZIP the three files and drop it on your reply.

I tried following your defines.  I ended up with SERVO0 on pin 1 and pin 11 as unused.

SpeedGP, 2017-05-16:
I had a feeling that was coming :)

Yes, 11 should be unused, that's one of the most confusing things... but not that big of an issue at this point.

I've changed those few items so many times, I may have written them backwards, but just as long as I can get pin D0 or D1 as the signal wire for the BLTouch, I'll be super happy!

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1005237/configs.zip)

Bob-the-Kuhn, 2017-05-16:
You have SERVO0 connected to pin 1.

The BLTouch signal should be attached to the Z_MIN endstop.

Bob-the-Kuhn, 2017-05-16:
You also need to set Z_MIN_ENDSTOP_INVERTING to "false".

Here's a write up on testing a BLTouch.

Testing a BLTouch with serial number:
1. Connect BLTouch, power up controller and upload the firmware
2. Blue LED will be ON if a servo signal is present.  Blue LED is faint.
3. M119 ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used.
4. M280 P0 S10 ; should deploy the probe and the orange LED will be OFF.
5. M119 ; should return OPEN
6. M280 P0 S90 ; should stow the probe and the orange LED will be ON
7. M280 P0 S60 ; puts it into the M119 test mode.
8. The probe should remain stowed and the blue LED should be OFF. If it's ON then the unit needs to be adjusted.
8. M119 ; should return TRIGGERED
10. M280 P0 S160 ; returns it to normal operation
11. M119 ; should return OPEN.

SpeedGP, 2017-05-16:
Maybe I'm using the wrong terminology. The three wires that are supposed to be connected to pin D11, 5v and Gnd, I'm referring to pin D11 as 'signal'. The issue is I already have something taking up all 4 "servo" connections, so I'd like to use "aux 1", ie 5v, gnd, D1 (or D0) for the BLTouch.

And you are right, right this second in the config, I have D1 setup, and I've tested this pin too, but it does not seem to change anything.

SpeedGP, 2017-05-16:
Trying inverting now, one sec

Bob-the-Kuhn, 2017-05-16:
Are the four items already attached to the servo connectors actually using servo signals?

Bob-the-Kuhn, 2017-05-16:
The three pin connector should go to gnd, +5V and D1.

The two pin connector should go to gnd & Z_MIN.

SpeedGP, 2017-05-16:
The three pin connector looks correct, the two pin connector looks correct. I just changed and uploaded "z_min_endstop_inverting" to false, then I ran M119 while the probe was up, it registered Open, then I pulled it down manually, and it still registered open.

Something is seriously messed up here... What do you suggest? Download the latest re-enter everything from scratch?

Only two of the 'servo' pins are used, the other two are just blocked by the 4x3 dupont connector

Bob-the-Kuhn, 2017-05-16:
The BLTouch normally issues a single 5mS pulse when it is triggered.  M119 won't see that.  M119 should always say open for the BLTouch.

Please provide a picture of how the 3 pin connector attaches to the board.

I'd also like you to fill this out for the BLTouch:
Brown wire - pin ?? on the RAMPs board
Red wire - pin ?? on the RAMPs board
Orange wire - pin ?? on the RAMPs board
White wire - pin ?? on the RAMPs board
Black wire - pin ?? on the RAMPs board

SpeedGP, 2017-05-16:
I just left for work. I will follow up on this a little later tonight.

For the record, thank you for your help. I do truly appreciate it.

James

-James--

SpeedGP, 2017-05-17:
Ok,

I have some extensions connected, green and black goes to the two-pin white and black, and the Red Black Blue connects to the brown red orange.

Brown wire - pin gnd aux1 on the RAMPs board
Red wire - pin 5v aux1 on the RAMPs board
Orange wire - pin D1 aux1 on the RAMPs board
White wire - pin z-min signal on the RAMPs board
Black wire - pin z-min gnd on the RAMPs board

[image]
[image]
[image]
[image]

It looks like there is a test mode I should play with, but I haven't read into it yet.

James

Bob-the-Kuhn, 2017-05-17:
You won't be able to get into the test mode until we can deploy & stow the probe.

As best I can tell it's wired up correctly.

+5 & ground have to be correct otherwise the orange light wouldn't be on and the pin wouldn't extend & retract on power up.

It's hard to tell from the picture but it looks like the servo signal is going to the D1 pin on AUX1.
[image]

Is the blue LED on?  It's much dimmer than the orange LED.  It's on the bottom end of the PCB on the right side.  You need to be looking for it to see it.

I'm leaning toward a bad connection.  
- Issue a M280 P0 S120 and then wiggle the connectors and mate/unmate them a few times. The pin will start cycling if you find the magic connection.
- Please check continuity from the blue D1 connector pin to the solder connection on the BLTouch.
- Remove & reseat the RAMPs board onto the Arduino Mega (yes, again).  Make sure there's no space between the mating connectors.
- Check continuity between the D1 input to the RAMPs board and the D1 pin on the AUX1 connector
- Really long shot - move the blue wire to A3/D57 or A4/D58 

[image]

SpeedGP, 2017-05-17:
Will do, thanks!

Quick random question, num_servos, it says it starts with 0, so if you had
two physical servos, would you type 2, or 1? Also, z_servo_nr (or whatever
it is called) should reflect the index, so 0 or 1, correct?

Roxy-3D, 2017-05-17:
> so if you had two physical servos, would you type 2, or 1?

If you had two...  You have servo 0 and servo 1

SpeedGP, 2017-05-17:
But servo_num would be 2?

SpeedGP, 2017-05-17:
Your longshot worked! It is currently using Pin D57/A3 and the blue light
came on! I am able to use the M280 commands, and complete a bed level. You
rock!!!!

One more tiny thing, this has been happening randomly the past 24 hours, do
you think it is the USB cable or possibly BAUD speed?

[image: Inline image 1]

-James--

Bob-the-Kuhn, 2017-05-17:
I'm really happy that it's working.  Wish I knew why D0 and D1 won't work.

Your image got thrown away somewhere in the ethers.  Please post it directly in this forum rather than as an email attachment.

I'm guessing you're getting random drops/faults in your prints.  USB com issues can be the cause.  I always print from the SD card because I have USB issues from my laptop.  My laptop will go to sleep in the middle of a print!

SpeedGP, 2017-05-17:
Actually no, it's just during connection. I use to have it drop out in the middle, but fixed it after getting a shorter, shielded cable with ferrite at both ends. I've been printing constantly from OctoPi for about a year now with zero issues.

This is in regards to the initial connection, sometimes it works fine, sometimes it hangs at connecting and displays this. I was assuming speed mismatch or noise on the line.

[image]

PS. The fan extender is still running pin 11 high (or so it seems). I'll be looking into this later in the week, but it's not super high on my priority list.

Seriously you rock! I'm an IT guy, I'm happy to send you some troubleshooting thoughts on your laptop :)

Bob-the-Kuhn, 2017-05-17:
You can use the **M43 I** command to see what the controller thinks pin 11 is doing.  This command will print out a list of all the pins, the signals/functions attached to them, if its an input, output or PWM and the current state/value.

To enable this command go to the end of Configuration_adv.h and enable PINS_DEBUGGING.

----

I have no idea why it sometimes hooks up immediately and sometimes doesn't.

SpeedGP, 2017-05-17:
Thanks! I'll give it a try!

for your laptop, first thing would be to look through power options in
control panel, there is an advanced section, look through hibernate, sleep,
turn hard disks off after, etc. the other place to look is in device
manager, you'll probably looking for your usb controller, right click, go
to power management tab, uncheck "allow this device to turn off to save
power".

You're welcome to close this ticket. thank you so much! (Unless you would
like to see results from M43)

oksakoor, 2017-05-20:
When i power on 3D Printer, then Bl Touch do selftest (moving pin up and down)
But after that cant move pin up or down and dont respond at all.

Roxy-3D, 2017-05-20:
I forget...  But I think the BL-Touch moves the pin when it gets power.   If that is true...  it is not being told by the firmware to move the pin.    It needs to be connected to a Servo connector.   Probably Servo-0

oksakoor, 2017-05-20:
You suggest activate this?

//#define Z_ENDSTOP_SERVO_NR 0   // Defaults to SERVO 0 connector.
//#define Z_SERVO_ANGLES {70,0}  // Z Servo Deploy and Stow angles

Roxy-3D, 2017-05-20:
If you have the #define BLTOUCH enabled, it should automatically set up
those values.

oksakoor, 2017-05-20:
When i was installing Bl Touch i enabled those

#define BLTOUCH
#define NUM_SERVOS 1

Roxy-3D, 2017-05-20:
That looks correct.   And the BL-Touch cable should be plugged into the Servo-0 connector.

oksakoor, 2017-05-20:
I plugged like this pictures shows. Is that Servo-0 connector?
[image]

Roxy-3D, 2017-05-20:
Yes.  That looks correct.   But you would be much better off giving it 5 volts from a separate power supply.
I don't think that is the problem you are fighting right now.

SpeedGP, 2017-05-20:
Have you tried M280 P0 S90 or M280 P0 S10

<https://cloud.githubusercontent.com/assets/28829211/26278464/e807bf22-3da3-11e7-8725-7c05c4844a8d.jpg>

<https://cloud.githubusercontent.com/assets/28829211/26278464/e807bf22-3da3-11e7-8725-7c05c4844a8d.jpg>

oksakoor, 2017-05-20:
I tried both codes and in lcd menu "BL Touch"- pin doesn't move

SpeedGP, 2017-05-20:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://3dator.com/wp-content/uploads/2016/04/AnthonyQuayle003.pdf&ved=0ahUKEwiLrMnkkP_TAhUS_mMKHVeWCl8QFggcMAA&usg=AFQjCNExLmy_geyECm2szqWBOGE6KKwELQ&sig2=jqjv88hrgejRbtqtIWgh9A

oksakoor, 2017-05-20:
I read this, but with Marlin 1.1.1 i need only enable those?

#define BLTOUCH
#define NUM_SERVOS 1

SpeedGP, 2017-05-20:
id look through that pdf and make sure your config is right
https://www.google.com/url?sa=t&source=web&rct=j&url=https://3dator.com/wp-content/uploads/2016/04/AnthonyQuayle003.pdf&ved=0ahUKEwiLrMnkkP_TAhUS_mMKHVeWCl8QFggcMAA&usg=AFQjCNExLmy_geyECm2szqWBOGE6KKwELQ&sig2=jqjv88hrgejRbtqtIWgh9A

closed 2018-02-06 by thinkyhead