MarlinFirmware/Marlin #6759: Mesh Invalidated

pmbroth, 2017-05-16:
Hello I downloaded the latest verison of code today, and get the following
[image]

The code as of 5/15th worked perfectly.  I did not make any changes since then.  

I have attached my config files from the 5/16 download

I don't know what changed from the other day to today.

Thank s

Brian

pmbroth, 2017-05-16:
[br config and adv.zip](https://github.com/MarlinFirmware/Marlin/files/1005442/br.config.and.adv.zip)

My files

Roxy-3D, 2017-05-16:
If you send a G29 P1 with UBL enabled...  You should see that message.   What is it not doing as you want?   The O option for a map is not T.

Tannoo, 2017-05-16:
O is now T

pmbroth, 2017-05-16:
When I usually send G29 P1, it does a level, and then starts creating the mesh.  With today's download, entering the command, does nothing except for the initial firts center level , then it just stops.  Hope that explains it.

Roxy-3D, 2017-05-16:
And the bed is homed?

pmbroth, 2017-05-16:
I just re-uploaded yesterdays code, since I archive everything, after entering G29 P1 , it moves to the center, then starts to build the mesh, here is some of the message I get
[image]

pmbroth, 2017-05-16:
Make Sense?

pmbroth, 2017-05-16:
Yes, the bed homes no problem, then it will just stop, and nothing else.  With code from yesterday and prior,it homes, then builds the mesh.

Roxy-3D, 2017-05-16:
Are you giving it commands via the LCD Panel?  Or are you using PronterFace to tell it what to do?

pmbroth, 2017-05-16:
I'm using prointerface and S3D  Both the same issues with today's code, but no issues with prior code

pmbroth, 2017-05-16:
I just sent another G29 P1 via S3D  no problem

Roxy-3D, 2017-05-16:
I'll have the code uploaded in a few minutes...    But why are you generating a new mesh all the time?   You should save your mesh...  And refine it to be perfect.   You really shouldn't be auto-probing all the time.

pmbroth, 2017-05-16:
I reuloaded the firmware, and create a new mesh whenever I upload  new firmware.  This helps me going through functionality testing.  Part of my checklist to make sure everything is working.  I'm in IT, so hardware and software testing for deployments always gets in the way.  :-)

Roxy-3D, 2017-05-16:
OK...  Because the mesh data is saved at the end of the EEPROM.  If you don't do something drastic like changing the size of your mesh, the data should be preserved.  Even if the EEPROM Configuration data changes, and you do a M502 / M500,  The mesh data should usually be safe.

But with that said, once you know how to operate the system...  Making a new mesh and fine tuning it (if necessary) should go very quickly.    I have the firmware compiling right now... In 3 or 4 minutes I'll be able to give it a G29 P1 and see what happens....

pmbroth, 2017-05-16:
OK good to know about the mesh.  I thought it was erased when I uploaded the firmware.

pmbroth, 2017-05-16:
cool, appreciate your quick response..

Roxy-3D, 2017-05-16:
No...   That was a design consideration.   Even if the data (or format of the data) in the EEPROM changes...  Your mesh data should be safe.

pmbroth, 2017-05-16:
that's great...  Thanks for the information...

Roxy-3D, 2017-05-16:
OK!  I duplicated what you are seeing...   What ever is wrong...  It isn't much.  But it might take 15 minutes to find it.

pmbroth, 2017-05-16:
ok, I'm not crazy - That is a good thing!  Thank you...

Tannoo, 2017-05-16:
BTW, I was trying to generate a new dementional mesh and yeah... it homed, then moved to the first probing position and stopped with the LCD status line saying "G29 UBL done!".

Roxy-3D, 2017-05-16:
Going through the steps to merge the correction now.    But if you want to jump ahead, you can replace this function in your ubl_G29.cpp file:
```cpp
  mesh_index_pair find_closest_mesh_point_of_type(const MeshPointType type, const float &lx, const float &ly, const bool probe_as_reference, unsigned int bits[16], const bool far_flag) {
    mesh_index_pair out_mesh;
    out_mesh.x_index = out_mesh.y_index = -1;

    // Get our reference position. Either the nozzle or probe location.
    const float px = RAW_X_POSITION(lx) - (probe_as_reference == USE_PROBE_AS_REFERENCE ? X_PROBE_OFFSET_FROM_EXTRUDER : 0),
                py = RAW_Y_POSITION(ly) - (probe_as_reference == USE_PROBE_AS_REFERENCE ? Y_PROBE_OFFSET_FROM_EXTRUDER : 0),
                raw_x = RAW_CURRENT_POSITION(X), raw_y = RAW_CURRENT_POSITION(Y);

    float closest = far_flag ? -99999.99 : 99999.99;

    for (uint8_t i = 0; i < GRID_MAX_POINTS_X; i++) {
      for (uint8_t j = 0; j < GRID_MAX_POINTS_Y; j++) {
        if ( (type == INVALID && isnan(ubl.z_values[i][j]))  // Check to see if this location holds the right thing
          || (type == REAL && !isnan(ubl.z_values[i][j]))
          || (type == SET_IN_BITMAP && is_bit_set(bits, i, j))
        ) {

          // We only get here if we found a Mesh Point of the specified type

          const float mx = pgm_read_float(&ubl.mesh_index_to_xpos[i]), // Check if we can probe this mesh location
                      my = pgm_read_float(&ubl.mesh_index_to_ypos[j]);

          // If using the probe as the reference there are some unreachable locations.
          // Also for round beds, there are grid points outside the bed that nozzle can't reach.
          // Prune them from the list and ignore them till the next Phase (manual nozzle probing).

//        if ((probe_as_reference && position_is_reachable_by_probe_raw_xy(mx, my)) || position_is_reachable_raw_xy(mx, my))
//          continue;
//
//        THE ABOVE CODE IS NOT A REPLACEMENT FOR THE CODE BELOW!!!!!!!
//
          bool reachable = probe_as_reference ?
                             position_is_reachable_by_probe_raw_xy( mx, my ) :
                             position_is_reachable_raw_xy( mx, my );
          if ( ! reachable )
            continue;
  
          // Reachable. Check if it's the closest location to the nozzle.
          // Add in a weighting factor that considers the current location of the nozzle.

          float distance = HYPOT(px - mx, py - my) + HYPOT(raw_x - mx, raw_y - my) * 0.1;

          /**
           * If doing the far_flag action, we want to be as far as possible
           * from the starting point and from any other probed points. We
           * want the next point spread out and filling in any blank spaces
           * in the mesh. So we add in some of the distance to every probed
           * point we can find.
           */
          if (far_flag) {
            for (uint8_t k = 0; k < GRID_MAX_POINTS_X; k++) {
              for (uint8_t l = 0; l < GRID_MAX_POINTS_Y; l++) {
                if (!isnan(ubl.z_values[k][l])) {
                  distance += sq(i - k) * (MESH_X_DIST) * .05
                            + sq(j - l) * (MESH_Y_DIST) * .05;
                }
              }
            }
          }

          // if far_flag, look for farthest point
          if (far_flag == (distance > closest) && distance != closest) {
            closest = distance;       // We found a closer/farther location with
            out_mesh.x_index = i;     // the specified type of mesh value.
            out_mesh.y_index = j;
            out_mesh.distance = closest;
          }
        }
      } // for j
    } // for i

    return out_mesh;
  }
```

pmbroth, 2017-05-16:
Thank you!  I will wait for the final when the fix is out in, and then move to a weekly firmware upload, let things bake in a little bit..  :-)

Tannoo, 2017-05-16:
I am trying it now.

Tannoo, 2017-05-16:
It's probing!

Tannoo, 2017-05-16:
My 1st 15x15 mesh generation.... I have time to make some coffee.

Roxy-3D, 2017-05-16:
@pmbroth   It's merged...

pmbroth, 2017-05-16:
Thank you!  I will test it!

pmbroth, 2017-05-16:
@Roxy-3D  Thank you!!  I just downloaded, configured, and uploaded to my printer.  All is working    Now for dinner, while it creates the mesh 15*15.  !

closed 2017-05-16 by pmbroth