MarlinFirmware/Marlin #6763: Lift Z when print stopped

gordo3di, 2017-05-16:
If you stop a print, is it possible to force the z axis to lift by 5mm or so. If you stop on the 1st layer and don't move the nozzle it could burn a hole depending on your bed. I'm not sure how to cleanly add it since killing the print disregards the buffer and just stops.

Just a thought. Thanks.

Roxy-3D, 2017-05-17:
Yes...    This is wanted by a lot of people... And more!   I want it to lift the nozzle, move away from the print and then turn off the nozzle.    But at that point it is getting very similar to the M600 Filament Change command.   So...   There is work in progress to let both ideas use the same support code:

https://github.com/MarlinFirmware/Marlin/pull/6407

Drayson80, 2017-05-22:
@Roxy-3D :-) seems this feature request get's quite popular.
Only one aditional request if considered in M600/M125: we need to make sure the documentation for each application/use case is foolproof so that it's simle to setup...

closed 2018-02-06 by thinkyhead