MarlinFirmware/Marlin #6765: Flow rate modification when printing seems to cause erratic movement

jdelahayes, 2017-05-16:
It seems that when i change the flow rate of an extruder while printing there is few seconds later erratic movements of the carriage. The carriage start to go left, then reprint quite well, then go to the back of the bed, then there is movements on the z axe causinf the head to crash on the bed. I can repeat that every time i change flow rate when printing. The same gcode print well if i don't touch the flow rate.

[image]

Maglin, 2017-05-19:
Just wanted to say I love the color of your Voron.  Have you recompiled and retired today?  A lot of changes have happened the last few days.

jdelahayes, 2017-05-19:
Thank you @Maglin. I tested last night babystepping and I had the same problems. I will update the firmware with the latest commits and retest with babystepping.

jdelahayes, 2017-05-19:
With the last bugfix-1.1.x commit i still have the same behavior. A line is skipped when the movement occurs.
[image]

jdelahayes, 2017-05-19:
The print stopped suddenly ... the screen was frozen and the menus were no longer accessible. Strange movements had occurred at least 5 or 6 times during printing. Any idea on how i can debug or help debuging ?
[image]

jdelahayes, 2017-05-19:
Aaaargh ... the current limit of XY stepper drivers wasn't egal and too low ... sometimes one motor did not have enought torque and the carriage go straight to the back or the right of the bed ! it's always good to check everything before thinking it's the other's work that not work :) it's not a bug. Thank you for your time.

Roxy-3D, 2017-05-19:
Yeah...  This is complicated stuff with a million variables...