MarlinFirmware/Marlin #6766: Some points are not probed with a more than 9 points matrix with AUTO_BED_LEVELING_BILINEAR

jdelahayes, 2017-05-16:
Some points are not probed when i set GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y to more than 3 (4 or 5) with AUTO_BED_LEVELING_BILINEAR.

The head move from point to point but the movement to probe the bed on Z axe is not done on all points, generally 2 points (or more, i can't remenber) are not probed.

My configuration is here https://github.com/jdelahayes/Marlin/blob/bugfix-1.1.x-voron/Marlin/Configuration.h

Roxy-3D, 2017-05-16:
The bed size is defined by where the nozzle can go.   But because the probe is often times offset from the nozzle, that means there are locations the probe can not reach.

jdelahayes, 2017-05-16:
There is no offset, it's a nozzle-based probe-switch and the my X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER are egal to 0. The head move at the point position, make a short stop and go to the next point without probing the bed. The missing point are not necessarly at the extremity of the matrix.

Bob-the-Kuhn, 2017-05-16:
I've seen that "move to a point, stop, do nothing, go on to the next position" when the probe was already triggered.

Maybe there's a cable issue or an interference problem.

Curiosity question - are you using a piezo system or a mechanical switch?

jdelahayes, 2017-05-17:
I will try to mesure the switch signal tonight. It's a mechanical switch. Thank you.

jdelahayes, 2017-05-18:
Yesterday, before re-testing, i pull the lastest changes for bugfix-1.1.X and since i cannot reproduce the problem ... i will re-re-test tonight ...

jdelahayes, 2017-05-19:
The problem came from a wrong adjustment of the screw adjusting the pressure of the switch ... sorry for the time wasted!

jdelahayes, 2017-05-19:
And thank you @Bob-the-Kuhn for the tips.

closed 2017-05-19 by jdelahayes