MarlinFirmware/Marlin #6767: SD card selection not found with 1.1.1

i3dprint365, 2017-05-17:
Please let me know if im just not turning the SD card on in the configuration before i flash the board I see 3 tabs that show code for sd card configurations. When I flash the board everything functions except the sd card. I have tried inserting before the printer is on and after no sd card function message even displays. There is also no print from sd card menu so, can you help please thanks.

Tannoo, 2017-05-17:
`#define SDSUPPORT` is what's needed.

Maglin, 2017-05-17:
@i3dprint365 I believe it used to be uncommented by default.  With the latest changes to the example config files it is now commented out.  I had the same issue last night.  Took me about 5 minutes of head scratching before I decided to look at the bottom of configuration.h.

i3dprint365, 2017-05-17:
found it thanks

closed 2017-05-17 by i3dprint365