MarlinFirmware/Marlin #6769: No z raise between probes if Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is defined

jodafo, 2017-05-17:
Hi,

I have the following issue:
If I use my z probe (bltouch) for z-homing via enabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, it z is not raised after probing. Z_SAFE_HOMING is enabled.

However, if I use my z_min switch endstop for homing z, and z_max for the z probe, everything works as expected, i.e. z is raised by Z_CLEARANCE_BETWEEN_PROBES between probes.

Bob-the-Kuhn, 2017-05-17:
Is the Z_MIN endstop disconnected when using the BLTouch in the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN mode?

To get details on what's happening please enable DEBUG_LEVELING_FEATURE in Configuration.h and then issue **M111 S32** before doing the G28, G29 sequence.  Copy the log into a TXT file and then drop the file onto your reply.

jodafo, 2017-05-17:
Well, sorry, indeed I didn't disconnect the Z_MIN endstop but instead just declared it as Z_MAX (for future usage). However, I didn't set Z_MAX_ENDSTOP_INVERTING to true so the max z endstop always triggered...

closed 2017-05-17 by jodafo

Bob-the-Kuhn, 2017-05-17:
Glad to see that you're up & running.

jodafo, 2017-05-18:
Thank you for your help!