MarlinFirmware/Marlin #6770: RCBugfix vs bugfix-1.1.x

jasoncross, 2017-05-17:
What is the difference between the "RCbugfix" branch and the bugfix-1.1.x branch?  I had been using RCBugfix, but when 1.1.0 went final, it disappeared and bugfix-1.1.x appeared.  Now they both exist, and both get updates.  Which branch should be used for the latest in-development updates?

FHeilmann, 2017-05-17:
I agree that RCBugFix should probably renamed to "devel" or something, since we are not updating RCs at the moment.

From the way I see it new commits are pushed both to bugfix-1.1.x and RCBugFix, but my best guess is that bugfix-1.1.x will stop receiving them at some point and RCBugFix will become the most bleeding edge again.

But maybe @thinkyhead can chime in what is plans are.

Roxy-3D, 2017-05-17:
Good guess!  But "No!"   Please check out:  https://github.com/MarlinFirmware/Marlin/issues/6742

jasoncross, 2017-05-17:
@Roxy-3D I'm not seeing how #6742 applies to this thread?

Roxy-3D, 2017-05-17:
Sorry...   Let me find the right number.   I goofed up a Cut & Paste....

jasoncross, 2017-05-17:
Ok, so bug-1.1.x is the one to use?

Roxy-3D, 2017-05-17:
Yes, https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x

FHeilmann, 2017-05-17:
Maybe we want to consider making these things a little more clear.

One Idea:

- `master`
- `maintenance`
- `rc`
- `devel`

The workflow would then be as follows:

- `master` is what everyone should grab for a stable version of marlin
- `maintenance` contains fixes for either `master` or `rc` depending on whether we are moving towards a minor or major release
-  `rc` contains the newest release candidate once we are moving towards a major release
-  `devel` contains the bleeding edge

After a major release, new features and major changes go to `devel`, `maintenance` is based on `master` and receives bugfixes.
During a minor release, `master` becomes `maintenance`.
Once a new major release is planned, `rc` and `maintenance` are moved up to `devel` and `rc` is released as the first release candidate. `maintenance`, at this point, receives bugfixes for `rc`.
During new release candidates,` rc` catches up with `maintenance`.
During a major release, `rc` and `master` become `maintenance`.

This would also solve one of the problems with current Marlin development: Feature Freezes. One could simply disallow anything but bugfixes against `maintenance` which would result in a feature freeze as soon as `rc` moves up to `devel` during the first release candidate. At the same time, `devel` could go wild all the time with new features.

fiveangle, 2017-05-17:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/README.md

> ## Contributing to Marlin
> 
> Click on the Issue Queue and Pull Requests links above at any time to see what we're currently working on.
> 
> _**To submit patches and new features for Marlin 1.1 check out the bugfix-1.1.x branch, add your commits, and submit a Pull Request back to the bugfix-1.1.x branch. Periodically that branch will form the basis for the next minor release.**_
> 
> Note that our "bugfix" branch will always contain the latest patches to the current release version. These patches may not be widely tested. As always, when using "nightly" builds of Marlin, proceed with full caution.

Roxy-3D, 2017-05-17:
The current RCBugFix should go away when everything associated with it is moved.   But it probably re-appears when another Release Candidate appears.

fiveangle, 2017-12-08:
Extraneous branches have been removed (or hidden) so this can be closed @jasoncross (or anyone with godmode powers 😊 )

closed 2017-12-09 by thinkyhead