MarlinFirmware/Marlin #6771: Feature request: printing QUEUE

batata004, 2017-05-17:
Hi,

As some of you advised me in this GITHUB community, sometimes I need to execute a bunch of GCODES right after turning on my printer. I dont want to execute these GCODEs before everyprint, just before the printer is turned on.

So I created a GCODE file that runs perfectly all the procedures the printer has to do. But often I have to go to PRINT FROM SD CARD (LCD) and print this initial file and I HAVE TO WAIT it finishes so I can choose the actual printing file that I wanna print.

I need to wait all the procedures to finish so only after that I can tell the printer to print the actual file.

Is there someway I can "queue" printing SD files?

fiveangle, 2017-05-17:
https://github.com/MarlinFirmware/Marlin/wiki/Autostart

batata004, 2017-05-17:
Grea great great!!!

batata004, 2017-05-18:
@fiveangle actually I was wrong. You suggestion may sound good but it's not. Indeed the printer starts printing the auto0 file right when I turn it on BUT it does not allow me to queue the final file that I indeed wold like to print.

I need to wait the auto0 procedures to be completely done so I can, after that, start printing my file. I need to be around, close to the printer till it finishes the procedures of auto0 so I can click PRINT FROM SD and choose the right file.

batata004, 2017-05-18:
Would it maybe be possible to me add a menu entry in the LCD that runs some gcodes? Is it possible to me create an entry in the LCD menus so when I click this entry it runs some gcode?

fiveangle, 2017-05-22:
No, you rename the file you want to print auto1...

closed 2018-02-06 by thinkyhead