MarlinFirmware/Marlin #6778: UBL should prohibit PROBE_MANUALLY

apelly, 2017-05-18:
The sanity check for UBL checks for a probe, but should exclude PROBE_MANUALLY.

Just making a note. I don't have time to fix it right now, but it causes compile errors.

Roxy-3D, 2017-05-18:
Yeah...  I don't even know off the top of my head what the sanity check currently does.   It won't be much.   

But the thing is...  We want to go the opposite direction.   UBL should work just fine with no probe.  We need to start relaxing constraints.   And let the user start with a G29 P0 (totally zero'ed mesh) or maybe G29 P0 and some G29 P2's.    And then let them jump to filling in the mesh with P3's.

thinkyhead, 2017-05-19:
Let's think about whether we can make `probe_pt` do a manual probe via the LCD when there's no electronic probe. Then all the code that calls `probe_pt` can stay the same. Manual adjustment via GCode is a little more tricky. See the ABL or MBL version of `G29` for the lowdown on how that should be implemented.

thinkyhead, 2017-05-19:
#6791 should catch this scenario now.

closed 2017-05-19 by thinkyhead