MarlinFirmware/Marlin #6779: BabyStepping not working

Maglin, 2017-05-18:
I'm not able to enter babystepping via double click nor is there a menu option for it.  I double checked and it is in my configuration_adv.h.  Here is my babystepping config out of my configuration_adv.h

```
#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define BABYSTEP_XY              // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false  // Change if Z babysteps should go the other way
  #define BABYSTEP_MULTIPLICATOR 1 // Babysteps are very small. Increase for faster motion.
  #define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping
  #define DOUBLECLICK_FOR_Z_BABYSTEPPING // Double-click on the Status Screen for Z Babystepping.
  #define DOUBLECLICK_MAX_INTERVAL 1250 // Maximum interval between clicks, in milliseconds.
                                        // Note: Extra time may be added to mitigate controller latency.
#endif
```

Roxy-3D, 2017-05-18:
That looks right...  Is your LCD Panel working other than the Baby Stepping stuff?

mtowara, 2017-05-18:
Does it work once the print is started and the Tune menu is available?

Maglin, 2017-05-18:
Sorry was late last night when I posted this.  Menu works.  I've only looked for babystepping after I've started a print.  The tune menu is available but I only have Z-Offeset as an option.  Did babystepping get changed to be called Z-Offset?  Or does using `#define BABYSTEP_ZPROBE_OFFSET`make it be called Z-Offeset?  That was the only option I had.  It didn't have the resolution that was set in babystepping and couldn't be accessed via double click.  It did work a few days ago before I started using UBL.

Roxy-3D, 2017-05-18:
> #define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping

I think it would be safest to turn this off until we get the baby stepping to work.    And it is a lot easier to see if it is working if you just home the printer and don't have it moving.   It maybe we also want to bump your    #define BABYSTEP_MULTIPLICATOR 15   so it is easier to see movement.

Can you try those things and report back???

Maglin, 2017-05-18:
@Roxy-3D I'll do this tonight.

Bergerac56, 2017-05-18:
Hello,
Experiencing with the V1.1.x bugfix, I just tried the babystep option. It is working fine for me. In "Tune" I get 3 options (babystep X, Y and Z). BTW is there a possibility to "baby-adjust" the Z offset manually from the LCD and without printing something? So I could adjust (and store hopefully) the Z_OFFSET (Z_PROBE_OFFSET_FROM_EXTRUDER) ?

Thanks

mtowara, 2017-05-18:
#define BABYSTEP_ZPROBE_OFFSET is supposed to do that, but I also had very limited success with that and reverted back to the classical babystepping

Maglin, 2017-05-18:
So with `#define BABYSTEP_ZPROBE_OFFSET` commented out, I can double click into babystepping at any time.  I've not checked but I'm sure I can access it through the Tune menu as well.  With `#define BABYSTEP_ZPROBE_OFFSET`  uncommented a double click would take me to ZProbe Offset but I couldn't change it with the encoder.  I had to go into the tune menu to do it which is only available during a print.

Roxy-3D, 2017-05-18:
> A question about babystepping. Does it save to EEPROM in anyway?

No.   And in fact...  I have my printers home a little bit too high.  I use Z-BabyStepping to exactly tune the nozzle height while it is printing the skirt.

Bergerac56, 2017-05-18:
When not printing, where do you find the babystep option? I found it only in "tune" But then, I have to be printing. I am surely missing something ;)

Roxy-3D, 2017-05-18:
I turn on the  #define DOUBLECLICK_FOR_Z_BABYSTEPPING option.   While at the status screen, you jsut double click the encoder wheel and you drop into Z-BabyStepping.

Bergerac56, 2017-05-18:
Ohhh noo. I forgot that option. Sorry It is late and ... :)

Maglin, 2017-05-18:
@Roxy-3D Thank you.  That is what I figured and kind of tested, but I was wanting confirmation.  👍

Roxy-3D, 2017-05-18:
The BabyStepping stuff is in the menu...   But I never use it.    And part of the problem you already discovered.  In order to be helpful...  Some options disappear when you can't use them.   I think it would be better to leave them in the menu but grey them out somehow so you know they are not available.

Maglin, 2017-05-24:
So been a few days.  I grabbed the latest bugfix-1.1.x and babystepping was working via doubleclick before I started a print.  After I started a print I could only access it via the Tune menu.  On a GLCD issue my display also blanks out on every menu change excpet going back to Info screen.  I have to move the encoder wheel to get the display to come up or display anything.  This maybe related.  I'm running the AnyCubic TriGorilla controller and have it set at a RAMPS1.4.  Looking at the pin definitions it appeared choosing RAMPS1.3 went to the same files, but maybe not.  BTW UBL with G29 P3.1 have given me a mesh I can print with out of the gate.  First time that has happened.  So great job everyone on the recent changes.

emartinez167, 2017-05-24:
I get the same screen blanking but on every menu move.

Xasin, 2017-10-29:
I'm having a bit of trouble using the `BABYSTEP_ZPROBE_OFFSET` feature.
I get the Z-Offset menu option, and can also reach it through a normal double-click.
**However**, while it does manipulate the Z-Probe offset, *it doesn't babystep*, which I think makes not really worth it.

I'm assuming the intended behaviour is to both babystep AND change the probe offset.

closed 2018-02-09 by thinkyhead