MarlinFirmware/Marlin #6781: MANUAL_HOME_POSITIONS not working correctly

coreytaylor211, 2017-05-18:
Hi the `MANUAL_HOME_POSITIONS` for the X axis is not working.  The Y axis is fine. after `G28` the X axis move a bit to check the min_endstop and hold on in position `X0.` The LCD shows `X100.` 
My Printer is an Prusa i3.

A part of my config `// The position of the homing switches
#define MANUAL_HOME_POSITIONS  // If defined, MANUAL_*_HOME_POS below will be used
//#define BED_CENTER_AT_0_0  // If defined, the center of the bed is at (X=0, Y=0)

//Manual homing switch locations:
// For deltabots this means top and center of the cartesian print volume.
#define MANUAL_X_HOME_POS 100
#define MANUAL_Y_HOME_POS 100
#define MANUAL_Z_HOME_POS 0
//#define MANUAL_Z_HOME_POS 402 // For delta: Distance between nozzle and print surface after homing.

//// MOVEMENT SETTINGS
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
#define HOMING_FEEDRATE {50*60, 50*60, 5*50, 0}  // set the homing speeds (mm/min)

// default settings
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,400,92.6}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {500, 500, 6, 25}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {9000,9000,50,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000   // X, Y, Z and E max acceleration in mm/s^2 for retracts

// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
// #define EXTRUDER_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
// #define EXTRUDER_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                10.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)`

plz help

mtowara, 2017-05-18:
> The LCD shows X100
sounds like its working... what behavior were you expecting?

Which branch are you using? The funny thing is that I see no reference to MANUAL_*_HOME_POS in the bugfix-1.1.x except when it is initially set to *_HOME_POS which is then in turn never used again except for deltas.

coreytaylor211, 2017-05-18:
I use the 1.0.1 branch. Only the Y axis working.  Here is an another part of code from Configuration_adv.h
`#ifdef MANUAL_HOME_POSITIONS  // Use manual limit switch locations
  #define X_HOME_POS MANUAL_X_HOME_POS
  #define Y_HOME_POS MANUAL_Y_HOME_POS
  #define Z_HOME_POS MANUAL_Z_HOME_POS`

that´s all, i can´t understand this

mtowara, 2017-05-18:
define "not working" / "working" please.
What are your bed dimensions? is your home position really 10cm apart from the endstop?

coreytaylor211, 2017-05-18:
my bed Dimensions are 200 x 200  and I would like my home position  to 100 x 100 . The Y axis works after `G28` command runs the y axis to the  endstop switch and then to y100, perfect. The X axis runs also to the endstop but after that stays on the x0 position.

The triggered stats are ok only the x_min_endstop and z_min_endstop is triggered. the other endstops shows open.

mtowara, 2017-05-18:
please scrap the remark about HOME_POS only being for delta, there is some Macro hacking going on which made my global search fail.

Do I understand you correctly that the printer thinks he is at x=100,y=100 (according to lcd) but actually is at x=0,y=100?

Is anything preventing you from updating to 1.1.x?
Its more than possible that this bug has actually been fixed in the past two years...

coreytaylor211, 2017-05-19:
> Do I understand you correctly that the printer thinks he is at x=100,y=100 (according to lcd) but actually is at x=0,y=100?
 Yes.

I did the upgrade to 1.1.0-RC8, the same issue. Now think the printer on both axis he is at x100 y100 but he is at x0 y0.

How can i flash my board, i would like to try a clean installation.

ps: what did you mean with 

> please scrap the remark about HOME_POS only being for delta, there is some Macro hacking going on which made my global search fail.

mtowara, 2017-05-19:
that now sounds like intended behavior to me.
Why do you want your home position to be in the middle of the bed?
If you really want that you could try to use:

```
#define BED_CENTER_AT_0_0 // If defined, the center of the bed is at (X=0, Y=0)
#define MANUAL_X_HOME_POS -100
#define MANUAL_Y_HOME_POS -100
```

Uploading the new firmware flashed everything except the EEPROM which you can reset with M502 + M500