MarlinFirmware/Marlin #6784: How to update my fork

oldmcg, 2017-05-18:
I'm still struggling with github.

From the "contributing code" document...

> Make a feature branch
> You should keep an unaltered copy of the bugfix-1.1.x branch that you can keep up to date from the main project. To make patches or add a new feature, make a copy of bugfix-1.1.x with a descriptive name.
> 

So I've got my fork and my branch off bugfix-1.1.x in my fork, now some changes have been made back in marlin's bugfix1.1.x.  Can someone please tell me how to "keep up to date from the main project" in my fork?  I want to "sync" the changes from marlin's bugfix-1.1.x into my fork's bugfix-1.1.x branch, and then I want to update my "feature" branch off that with those changes. 

I have the github web UI to my fork, and I have the github desktop app, and I have a local copy of the files of my feature branch.  Nothing modified yet, I just need to update to current before I paste my changes back into a changed source file because there are conflicts.

Thx.

Roxy-3D, 2017-05-18:
If you are using GitHub's Desktop program:    I believe you 'Select' your top level bugfix-v1.1.x branch.   And then select 'Sync'.    In theory...  I think that is supposed to get everything the same.   But I have not been successful yet having a top level master copy that I branch off of.   Instead...   I've been modifying my fork, and doing a Pull Request from that.

mtowara, 2017-05-18:
the git way of doing it: (sorry I only know the bare linux commandline commands):
in your cloned repo:
git fetch https://github.com/MarlinFirmware/Marlin.git bugfix-1.1.x
git merge FETCH_HEAD
(pray that you dont have merge conflicts, else resolve them)
git commit
git push if you want to update your fork in github

mtowara, 2017-05-18:
inside of github:
in your fork, create a new pull request, select compare across forks, choose base as your fork/branch and head fork as Marlin/branch, create pull request and merge.

oldmcg, 2017-05-18:
I think I accomplished it -- just not obvious. 

mtowara's method of creating a PR in the web UI is what I tried before, but I ended up creating a PR in marlin, not in my fork.  The web UI says "behind marlin/bugfix-1.1.x by 12 commits", but if you click the PR button right beside that which would think does something about the "behind by N commits" instead defaults to the wrong direction.  Changing the base around in the PR might have worked, but I was afraid of creating a new PR in marlin instead of in my fork, and there is nothing in the PR UI to indicate where the PR is headed.  IMHO, I shouldn't need to create a PR to update my own fork, but maybe that is only way from web UI.

From desktop app, I had to switch my branch to my local bugfix-1.1.x, then change the dropdown of what is being compared from MarlinFirmware/1.1.x to MarlinFirmware/bugfix-1.1.x (defaults to 1.1.x in desktop UI even though my fork has default branch of bugfix-1.1.x in web UI).  And change the dropdown below that to just plain bugfix-1.1.x.  Then click "sync", then click "Update from MarlinFirmware/bugfix-1.1.x".  Then change my branch again in the dropdown above that to "my feature", which apparently changes my local snapshot to that branch as well (git status changed).  Then select dropdown for compare my bugfix-1.1.x to my feature-branch, "sync", and "Update from bugfix-1.1.x".  

What was not clear to me before was that github desktop changes BOTH my local directory contents AND my github fork on the web.  Unlike the web UI that only changes fork on the web or command line that only changes local copy on my machine (except for push, which seems opposite of "pull request" for updating my fork).

Now I see why Roxy just deletes the fork and creates a brand new fork every time.  Maybe that is fewer steps.

Thank you both for your suggestions.

Roxy-3D, 2017-05-18:
I'll give your steps a try!    One big problem with deleting the fork and starting over is you can only have 1 thing going at a time.   So...  even if your way takes a few extra steps, if I can make it work, it will be very helpful to me.

mtowara, 2017-05-18:
and I thought plain old git was unintuitive :)

emartinez167, 2017-05-18:
Have you guys tried SourceTree from Atlassian?

Regards,

Ernesto.

Roxy-3D, 2017-05-18:
Trying it now...  But I'm lost....

emartinez167, 2017-05-19:
They have tutorials. I am going to have a look tonight.

Regards,

Ernesto.

Tannoo, 2017-05-19:
I have been successful with rebasing and squashing with these:

git rebase -i MarlinFirmware/bugfix-1.1.x

Pops up a window where you can select commits to be squashed into one.
If there are conflicts, got edit the files and save them.
After conflicts are resolved, git add for every resolved file:
git add <your base>/<resolved_filename.ext>

git push -f

All will then be updated and squashed into one commit.

Roxy-3D, 2017-05-19:
@emartinez167   Is SourceTree time bombed?  Does it quit working in 30 days if I don't buy it?

I think I'm starting to figure out a few things on it.    It is kind of backwards (compared to GitHub's Desktop) on how it displays the hierarchy of the branches.    That confused me for a little while.

emartinez167, 2017-05-19:
I think it is free? I can't recall having paid.

Regards,

Ernesto.

emartinez167, 2017-05-19:
It is actually free.

Regards,

Ernesto.

Roxy-3D, 2017-05-19:
Thank You!   I didn't want to spend a lot of time learning something just to have the time bomb go off and have them tell me I can't use it any more.

Bob-the-Kuhn, 2017-05-19:
Here's how I update my main branch.  I do this just before I fork a new branch from it.

First - I never do anything with the main branch except update it and fork new branches from it.  All my work is done on the forked branches.  This way you don't get any conflicts when updating the main branch.
```
open git desktop
select the main branch
open the branch in git shell & enter this sequence
git remote -v
git remote add upstream https://github.com/MarlinFirmware/Marlin.git
  only do this if the first command doesn't show this setting
git fetch upstream
git rebase upstream/bugfix-1.1.x
git push -f
```

When I want to bring one of the forked branches up to date I follow the wiki procedure.

oldmcg, 2017-05-19:
Thanks, I will try that.

Seems like a common enough scenario that there ought to be a button "update this branch from the #head of wherever it came from if there are no conflicts".  Or simply "sync" in every other SCM.

Bob-the-Kuhn, 2017-05-19:
Git can be pretty perverse.  

I just tried to update my main branch & it has conflicts.  I accepted all the requested changes and now it's asking for more.  I think it's going through the commits one at a time.

I either accept the pain now or when I try to submit the new branch later.  Now is better.

Roxy-3D, 2017-05-19:
For me...   It is simpler to set the directory aside, and go get the current stuff.   And then using a visual diff program, I know where my changes are.   I can cross those over fairly easily because they are only in a few places.

Of course...  The thing is the Source Code Control System should be making my life easier.  But it isn't.  It is making things harder.

fiveangle, 2017-05-19:
directions to make github slightly less frustrating:

1. Take Github Desktop
2. Throw in trash
3. Install GitKraken
4. Enjoy only having to learn a handful of command lines instead of a semester's worth

apelly, 2017-05-22:
Here are a few git resources I've found useful. Excuse the casual descriptions. These were my own notes. My rule 1 comes from the girl in the first video. "If in doubt, commit." You can always recover from here.

There is a swag of git lingo. I recommend embracing it instead of fighting it: Your google-fu will level up.

This video presents good background and gets onto some useful stuff. https://vimeo.com/46010208

This dude is some kind of git evangelist, somehow affiliated with github. He's on crack or something so the pace is up there. He throws in some useful reference links too. https://www.youtube.com/watch?v=ZDR433b0HJY

Neither of those are what I'd call advanced, but they cover the behind the scenes fundamentals and therefore the basics. Armed with this, and [git-cola](https://git-cola.github.io/), the world is pretty much your oyster. 

Do try git cola!

Also, [Meld](http://meldmerge.org/) is an awesome diff tool.

Both of these are cross platform I believe.

This last vid has good shit in it: From about 50 minutes in it covers stuff the other two glossed over. https://vimeo.com/35778382

Other random links
-
http://gitref.org/index.html
https://git-scm.com/book/en/v2
http://think-like-a-git.net/
http://www-cs-students.stanford.edu/~blynn/gitmagic/

thinkyhead, 2017-05-23:
If you have a bash command line, I've also included a bunch of [helper scripts](https://github.com/MarlinFirmware/Marlin/issues/3193) in the tree to make it faster to work with Marlin and Github. They are in the `buildroot/share/git` folder. You can use these commands as shortcuts. I will eventually do a (better) video demonstrating them. Even if you don't use them, it might give some insight to read them, as they do some clever stuff.

I keep my `bugfix-1.1.x` branch up to date with "upstream" using the `mfup` script, which also rebases whatever branch I am currently in on the latest bugfix. I manage my commits with `mfrb` which simply starts an "interactive rebase." I resolve any merge conflicts by hand. This works out fine. You have to examine each conflict carefully anyway.

From the command-line, I do much as @Bob-the-Kuhn does, except, well…
```
git fetch upstream
git checkout bugfix-1.1.x ;# my working copy
git reset --hard upstream/bugfix-1.1.x
```
This is faster and cleaner, and if you've modified the working copy then –poof– it's clean again.

Roxy-3D, 2017-05-23:
> If you have a bash command line, I've also included a bunch of helper scripts in the tree to make it faster to work with Marlin and Github. 

Using Github Desktop I can go to the upper right corner and click on the gear.  One of the options is "Open in Git Shell".   Does that constitute a 'bash command line'?    If so...   Do I need to have my main (unaltered) fork of Marlin selected before I do any of those scripts?

Tannoo, 2017-05-23:
In git desktop, you can also right-click on your repo name on the far left side. In that context menu will also be "Open in Git Shell"

Yes, that is a bash terminal.

Roxy-3D, 2017-05-23:
@thinkyhead are these safe for me to just type use?   If I type ./mfnew I don't want it accidently doing things up in MarlinFirmware/Marlin

- mfup script, which also rebases whatever branch I am currently in on the latest bugfix. 
- mfnew create a new copy of bugfix-1.1.x 
- mfrb  "interactive rebase." 
- mfpr  Pull Request

oldmcg, 2017-06-04:
@Bob-the-Kuhn, @thinkyhead, I think one of the pieces I was missing before was adding upstream reference...

> git remote -v
> git remote add upstream https://github.com/MarlinFirmware/Marlin.git
>   only do this if the first command doesn't show this setting
> git fetch upstream
> git rebase upstream/bugfix-1.1.x
> git push -f

Now I see the upstream references in remote -v but I also see "MarlinFirmware" references to the same thing.  Is "upstream" just a common naming convention or does it have special meaning?
```
>git remote -v
MarlinFirmware  https://github.com/MarlinFirmware/Marlin.git (fetch)
MarlinFirmware  https://github.com/MarlinFirmware/Marlin.git (push)
origin  https://github.com/oldmcg/Marlin.git (fetch)
origin  https://github.com/oldmcg/Marlin.git (push)
upstream        https://github.com/MarlinFirmware/Marlin.git (fetch)
upstream        https://github.com/MarlinFirmware/Marlin.git (push)
```
Before I never worried about accidentally doing something to marlin instead of my fork.  Now is there anything I can accidentally do that will affect marlin via this upstream reference?

And trying to understand the difference between rebase and reset --hard, it seems like reset --hard is essentially the same thing as deleting my forked bugfix-1.1.x branch and then re-adding it pointing at #head -- I get rid of all the "merged from" commit history each time I did a rebase.  But, since I have other working branches based off my fork's bugfix-1.1.x, will reset --hard invalidate those or cause me merge problems later because I've broken the merge chain that multiple rebase history showed?  I never do work in my bugfix-1.1.x branch directly, just sub-branches off of that.

Also, @thinkyhead, I don't understand that last comment

> (This also makes it track with "upstream" instead of your fork, which may be more convenient, or not.)

Does this mean I don't have to explicitly merge from marlin into my bugfix-1.1.x and then merge again into my working branch off of my bugfix-1.1.x before submitting a PR that is up to date with Marlin's bugfix-1-1.x #head?  What behavior is different by "track with 'upstream' instead of my fork"?

Thanks for all the help.

Roxy-3D, 2017-06-04:
> Before I never worried about accidentally doing something to marlin instead of my fork. Now is there anything I can accidentally do that will affect marlin via this upstream reference?

I can only answer this question (of the many you asked):  You won't hurt MarlinFirmware.  You don't currently have Write access to MarlinFirmware.   But that is why I asked:

> are these safe for me to just type and use? If I type ./mfnew in Git Shell I don't want it accidentally doing things up in MarlinFirmware/Marlin?

I do have write access and I don't fully understand Git.   I wanted to be sure I could use the scripts and not worry about clobbering MarlinFirmware repositories.   

The more complicated questions Thinky will have to answer.

thinkyhead, 2017-06-12:
> @thinkyhead are these safe for me to just type and use?

My guess is they won't work, because you'd need a fuller GNU environment. But perhaps they can be adapted to your environment…?

thinkyhead, 2017-06-12:
> If I type ./mfnew in Git Shell I don't want it accidentally doing things up in MarlinFirmware/Marlin

None of these will directly push to the upstream repository. You would want to use `mfinit` first.

thinkyhead, 2017-06-12:
> Is "upstream" just a common naming convention or does it have special meaning?

It's a common convention, and some of the `mf*` scripts expect it to be present. Since I don't find myself typing it very much these days, it might make sense not to use it, but I do find it more convenient than `MarlinFirmware` when I do want to use it.

thinkyhead, 2017-06-12:
> Before I never worried about accidentally doing something to marlin instead of my fork. Now is there anything I can accidentally do that will affect marlin via this upstream reference?

You can't push directly without more access privileges, so you're safe. Note that with `git reflog` you can recover something accidentally altered or deleted 99.9% of the time. Git doesn't throw away anything until you tell it to `git gc`, so you can always go back to a previous point in time through its commit ID.

thinkyhead, 2017-06-12:
> trying to understand the difference between `rebase` and `reset --hard`

Apples and oranges, really.

Rebase moves your commits to a new point in the commit history — and usually you use it to move your commits to the end, after the latest commit merged into `bugfix-1.1.x` for example. There's also `git rebase -i` which you can use to mess around with previous commits.

You use `git reset` to point the HEAD of the current branch at some other commit ID. `git reset --hard other_branch` makes the current branch identical to `other_branch`.

thinkyhead, 2017-06-12:
> Does this mean I don't have to explicitly merge from marlin into my bugfix-1.1.x and then merge again into my working branch off of my bugfix-1.1.x before submitting a PR that is up to date with Marlin's bugfix-1-1.x #head? What behavior is different by "track with 'upstream' instead of my fork"?

This is what `mfup` does — take a look at the source. It does a `git fetch upstream`, checks out `bugfix-1.1.x`, makes it point to the head of `upstream/bugfix-1.1.x`, then goes back to the current branch and hits it with a `git rebase bugfix-1.1.x` — at which point you might hit conflicts.

If you'd like to see how to deal with all this, I can do a screen-cast demo any old time.

thinkyhead, 2017-06-12:
> What behavior is different by "track with 'upstream' instead of my fork"?

Use `git branch -vv` to see details about your branches, including what "remote reference" it is tracking. For example, here's my current mess…
```
$ git branch -vv
  1.1.x                    b2c20a8c0 [upstream/1.1.x] Version 1.1.3
  RCBugFix                 515f05093 [upstream/RCBugFix] Version 1.1.2
  a3d_main_controller      7d604c4ac [origin/a3d_main_controller] A3D Main Controller configuration
  a3d_zaxis_controller     9b80edca7 [origin/a3d_zaxis_controller] Command handling via i2c
* bf_PCA9632               05a792407 [origin/bf_PCA9632] Combine Travis tests to shorten required time
  bf_chunk_support         9b1dcd204 [origin/bf_chunk_support] initial chunk support
  bf_config_CL-260         07f8da36d [origin/bf_config_CL-260] CL-260 configs
  bf_followups_friday      420e5f93a [origin/bf_followups_friday] Followup for I2C_POSITION_ENCODERS
  bf_laser_pwm_mode        070897bb7 [origin/bf_laser_pwm_mode] twibus to keep
  bf_prusa_multiplexer     1d1565b08 [origin/bf_prusa_multiplexer] Fix spacing in gcode.cpp
  bf_revert_draw_flags     3bb67ddea [upstream/bugfix-1.1.x: ahead 1, behind 176] Revert some drawing flags
  bf_teensy_fastio         6d829e1e5 [Bob-the-Kuhn/AT90USB---Teensy-&-FASTIO-convergence: ahead 19, behind 2] Tweaks to be more generalized
  blood_delta_bugfix-1.1.x 29c16f9c6 [origin/blood_delta_bugfix-1.1.x] Travis for personal configuration only
  bugfix-1.1.x             dce665a08 [origin/bugfix-1.1.x] Merge pull request #7025 from magnuspub/patch-1
  i3_millie_bugfix_1.1.x   31095b665 [origin/i3_millie_bugfix_1.1.x: ahead 64, behind 5] Debug leveling
  marcelo_original         6aa20d2dd [marcelogaio/bugfix-1.1.x: ahead 83] Fixed ## before endif
  marcelo_redo             0632eb7db [origin/marcelo_redo] Add Galvanometer and SLA printing support
  rcbugfix_plus_32bit      969fab895 [origin/rcbugfix_plus_32bit] fix: ocr_val overflow at slow feedrates, change feedrate increase condition from __SAM3X8E__ to CPU_32_BIT
  redo_a3d_main            fa7f9f797 [upstream/bugfix-1.1.x: ahead 4, behind 122] A3D Main Controller configuration
  release_113              71194155d [upstream/1.1.x: ahead 1, behind 4] Version 1.1.3
  restore_zrib             58b2c1fb1 [FATALHERMIT/RCBugFix: ahead 9, behind 920] placed them in the wrong folder
  test_millie              04bd4c00d config
```

closed 2019-02-20 by thinkyhead