MarlinFirmware/Marlin #6785: error compiling with CASE_LIGHT_PIN  defined

autonumous, 2017-05-19:
in configuration_adv.h, with 
`#define CASE_LIGHT_PIN 4`

get the following compiling error

```
/var/folders/1q/bzhhs7415d983x4v3_rrl6c40000gn/T//cc06OBFi.ltrans13.ltrans.o: In function `lcd_main_menu()':
cc06OBFi.ltrans13.o:(.text+0x27e): undefined reference to `case_light_on'
cc06OBFi.ltrans13.o:(.text+0x316): undefined reference to `case_light_on'
cc06OBFi.ltrans13.o:(.text+0x31c): undefined reference to `case_light_on'
cc06OBFi.ltrans13.o:(.text+0x322): undefined reference to `update_case_light()'
collect2: error: ld returned 1 exit status
```

suggesting case_light_on, update_case_light aren’t defined (only declared as extern)

i think it’s caused by  a recent change to the  the the HAS_CASE_LIGHT macro
`#define HAS_CASE_LIGHT (PIN_EXISTS(CASE_LIGHT) `   ->  ` #define HAS_CASE_LIGHT (PIN_EXISTS(CASE_LIGHT) && ENABLED(CASE_LIGHT_ENABLE))`

in https://github.com/MarlinFirmware/Marlin/commit/99c4900a98163c784ef210ec574fc052873e14e4

Tannoo, 2017-05-19:
Undo that change and see if it works for you.

Bob-the-Kuhn, 2017-05-19:
The problem is that the pins_xxx.h files that were just updated were created on a PR that assumed another PR would already be merged.  Both PRs are quite old so it looks like the dependency was forgotten.

Now that 1.1 is out there's probably enough time to revisit PR #5685

Bob-the-Kuhn, 2017-05-19:
I've moved the changes over to PR #6797 for inclusion in bugfix-1.1.x

closed 2018-02-06 by thinkyhead