MarlinFirmware/Marlin #6786: DIOSHIFT_CLK_DDR when verifying before upload

1629dill, 2017-05-19:
Received this error report when trying to compile before uploading to board.

Arduino: 1.8.2 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Todd Hansen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Todd Hansen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Todd Hansen\Documents\Arduino\libraries -fqbn=Sanguino:avr:sanguino:cpu=atmega1284p -ide-version=10802 -build-path C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130 -warnings=all -build-cache C:\Users\TODDHA~1\AppData\Local\Temp\arduino_cache_980043 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Todd Hansen\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Todd Hansen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Todd Hansen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Todd Hansen\Documents\Arduino\libraries -fqbn=Sanguino:avr:sanguino:cpu=atmega1284p -ide-version=10802 -build-path C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130 -warnings=all -build-cache C:\Users\TODDHA~1\AppData\Local\Temp\arduino_cache_980043 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Todd Hansen\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\Marlin.ino
Using board 'sanguino' from platform in folder: C:\Users\Todd Hansen\AppData\Local\Arduino15\packages\Sanguino\hardware\avr\1.0.2
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\Todd Hansen\AppData\Local\Arduino15\packages\Sanguino\hardware\avr\1.0.2\variants\sanguino" "C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\Marlin.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\Todd Hansen\AppData\Local\Arduino15\packages\Sanguino\hardware\avr\1.0.2\variants\sanguino" "-IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src" "C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\Marlin.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\G26_Mesh_Validation_Tool.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\M100_Free_Mem_Chk.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\MarlinSerial.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\Marlin_main.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\Sd2Card.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\SdBaseFile.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\SdFatUtil.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\SdFile.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\SdVolume.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\blinkm.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\cardreader.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\configuration_store.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\dac_mcp4728.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\digipot_mcp4018.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\digipot_mcp4451.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\endstops.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\hex_print_routines.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\least_squares_fit.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\mesh_bed_leveling.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\nozzle.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\planner.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\planner_bezier.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\printcounter.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\qr_solve.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\serial.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\servo.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\stepper.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\stepper_dac.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\stepper_indirection.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\stopwatch.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\temperature.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\twibus.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\ubl.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\ubl_G29.cpp
Using cached library dependencies for file: C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\ubl_motion.cpp
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\ [... cut ...]
C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\ultralcd.cpp:3748:7: note: in expansion of macro 'WRITE'

       WRITE(SHIFT_CLK, LOW); \

       ^

C:\Users\TODDHA~1\AppData\Local\Temp\arduino_build_813130\sketch\ultralcd.cpp:3828:11: note: in expansion of macro 'GET_BUTTON_STATES'

           GET_BUTTON_STATES(buttons_reprapworld_keypad);

           ^

Using library LiquidCrystal at version 1.0.5 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
'DIOSHIFT_CLK_DDR' was not declared in this scope

Maglin, 2017-05-19:
@1629dill did you use one of the new configuration examples?  I had a different liquidcrystal error and it was do to some changes in the configurations but I was trying to use my old one.  Just something to check out.  That is a wall of errors btw.

Tannoo, 2017-05-19:
Please redo that into a txt file or zip.

Tannoo, 2017-05-19:
But, it ssems to me that you have an issue with the IDE build environment.

Maybe uninstall the IDE and re-install it?

Bob-the-Kuhn, 2017-05-19:
I think he's got an outdated Sanguino IDE extension.

I just downloaded today's bugfix1.1.x & compiled it for Sanguino with no issues.  I changed the motherboard to SANGUINOLOLU_12, selected the RepRap discount full graphics controller and followed the instructions in the file **pins_SANGUINOLOLU_11.h**.

The new pins_SANGUINOLOLU_11.h has instructions on how to install a useable IDE for Sanguino.

1629dill, 2017-05-20:
Have tryed two computers and have the same problem.

Bob-the-Kuhn, 2017-05-20:
Have you tried the Sanguino IDE extension that's referenced in the **pins_SANGUINOLOLU_11.h** file?
[pins_SANGUINOLOLU_11.zip](https://github.com/MarlinFirmware/Marlin/files/1015860/pins_SANGUINOLOLU_11.zip)

I can try to get closer to your configuration & try to reproduce the problem.  To do that I'll need the following:
- Configuration.h
- Configuration_adv.h
- pins_SANGUINOLOLU_11.h
- **DETAILS on how I can get the IDE that you're using**
- snapshots of the Arduino settings you're using (similar to the ones below)

Please ZIP the three files up and then drop the ZIP onto your reply.

[image]

[image]

If you point me to the Sanguino IDE extension you're using and ZIP up your two config files and your

Bob-the-Kuhn, 2017-06-03:
Status?

closed 2018-02-07 by thinkyhead