MarlinFirmware/Marlin #6789: Feature Request - MKS-TFT32/TFT28 support

viperfan7, 2017-05-19:
Any chance we could see support for the MKS TFT32/TFT28 in the future?

Maglin, 2017-05-19:
Everyone wants this.  They are cheap but also kind of closed source so modifying the FW is a daunting task.  To my knowledge.

fiveangle, 2017-05-19:
Someone did start an alternative, very much in dev stage:

https://github.com/robotsrulz/MKS-TFT

RalphSch, 2017-10-20:
Finally got one to play with :) The MKS TFTs are the same display types like the PanelDue or Octoprint on an RaspberryPi, they are completely independent controllers with displays which are just talking gcode through a serial connection to the printer controller. A very cool alternative are the Nextion touch screens, there is already a project-file for an Marlin GUI.
The main question for all these kind of smart-serial-controllers is: Does Marlin support multiple serial connections at once (also interesting for stepper drivers like the TMC2208) to support such displays without losing direct USB connectivity?

AnHardt, 2017-10-20:
@RalphSch 
Even if a bit lengthy the current state is well described in #8004 .

tcm0116, 2017-10-20:
@viperfan7 What control board are you using? #8004 adds UART support for LPC1768-based controllers, which indirectly allows the MKS TFT32/28 to work if you configure Marlin to use the serial port the the MKS TFT is connected to. However, Marlin currently doesn't support having multiple host interfaces in Marlin, so you won't be able to have the MKS TFT connected to one UART and send commands from a PC via another UART or USB. That's something that's definitely on the TODO list.

My particular control board (a BIQU BQ111-A4) is LPC1768-bsed, but doesn't provide access to the native USB port on the LPC1768. Instead, it uses a UART to USB bridge and then runs the UART0 connection to the MKS TFT and to the UART to USB bridge in parallel. I can select the Disconnect option in the MKS TFT to allow me to send commands from the PC via the UART to USB bridge. It's a work-around, but actually works pretty well, except for I lose some of the added features provided by the native USB interface, such as being able to copy files to the system SD card directly from the PC.

Nutz95, 2017-12-09:
Hello, i'm also using a MKS-TFT display and i struggle to make it communicate to marlin.
I've tried the Aux port as it's usually working (it's supported by smoothieware).
And i've also tried the side UART (Re-Arm) port linked to the display.
They don't seem to speak to each other.

If it's a limitation for now to be able to use only 1 UART, On my side i'm not using the USB UART.
So is there a way to reroute the UART to the ReArm Side port?

marcio-ao, 2018-01-02:
I suppose this is off-topic, since it does not pertain on this particular screen in the FR, but we are working on a touchscreen implementation for Marlin. We are looking to manufacture the display boards as open source hardware. The screen controller is an FTDI FT81x which offloads most of the work from the microcontroller to a graphics engine. The SPI traffic is a lot less than using the RepRapDiscount boards.

thinkyhead, 2018-02-11:
Thanks to a helpful contributor we now have support for the controller used in the Malyan M200.

#9470 / #9485

thinkyhead, 2018-05-24:
@flankerzo — Thanks for the note. Malyan LCD is already working in `bugfix-2.0.x`, and I've just patched it to work in `bugfix-1.1.x` also. Download the latest code to continue.

You can choose where you want to connect the filament runout sensor. The most preferred connector to use is one of your unused endstop pins.

flankerzo, 2018-06-18:
Sorry for late update, it works, unfortunately my induction sensor fried Z min pin on mks gen L. i will fix it, 
Thank you :)

Psyc0Flyer, 2018-06-21:
I have the MKS TFT32_L V2.0 screen and I would like to use it on my PRUSA i3 MK3 with OctoPrint running on a Pi 3 B. Any advice?

thinkyhead, 2018-06-30:
> Any advice?

Try [MarlinKimbra](https://github.com/MKFirmware/MarlinKimbra).

boelle, 2019-02-18:
@viperfan7 is this one still relavant?

InsanityAutomation, 2019-02-18:
I would say given that these are external streamers and have no bearing in the firmware space, no. This should be closed.

boelle, 2019-03-06:
@thinkyhead i think we can close this one given what @InsanityAutomation said

closed 2019-03-15 by p3p