MarlinFirmware/Marlin #6790: Little Marlin 1.1.1 bug

edautz, 2017-05-19:
Upgrade my CTC Prusa I3 from 1.1.0-RC8 to 1.1.1.

I noticed a little bug. When using the manual bed level feature, after the start of this feature the text on my display vanished, after turning the rotary encoder the text appears. Same for the first calibration point. This bug was not present in 1.1.0-RC8.

I using a (GT)2560 rev A clone control board with a Amtel 1280 AVR and LCD2004 Reprap discount controller with rotary encoder.

thinkyhead, 2017-05-19:
The LCD code has gotten a tweak here and there. I'll look into this some time today.

Maglin, 2017-05-20:
I've seen the same behavior.  I never mentioned it though.  Using a Delta and UBL.

thinkyhead, 2017-05-21:
> When using the manual bed level feature…

`MESH_BED_LEVELING` or `PROBE_MANUALLY`?

edautz, 2017-05-21:
@thinkyhead

Manual mesh bed leveling. I also noticed this sometimes after starting print from SD card, right after the heating and the leveling tasks. Kust when the printong task starts. Turning or pushing the encoder trigger text again.

edautz, 2017-05-21:
@thinkyhead.

It is manual mesh bed leveling. I noticed sometimes after I start print from SD, after heating and homeing, printing starts and the text on the display also vanishes. Turn the encoder also displays the text again.

thinkyhead, 2017-05-22:
Are you saying that the Status Screen disappears also?

Roxy-3D, 2017-05-22:
> I noticed a little bug. When using the manual bed level feature, after the start of this feature through the display menu, the text on my display vanished, after turning the rotary encoder the text appears. 

Yeah...  This sounds similar to this:  https://github.com/MarlinFirmware/Marlin/pull/6822#pullrequestreview-39521334

That is why I added the 
```cpp
     lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
```

thinkyhead, 2017-05-22:
Hmm… well that indicates a bug that we're just hacking around. I'll re-examine the display logic some more. Is this the graphical or character display, or both?

thinkyhead, 2017-05-22:
Thing is, we must use `LCDVIEW_KEEP_REDRAWING` because it works on both.
```cpp
#if ENABLED(DOGLCD)
  bool drawing_screen = false;
  #define LCDVIEW_KEEP_REDRAWING LCDVIEW_CALL_REDRAW_NEXT
#else
  #define LCDVIEW_KEEP_REDRAWING LCDVIEW_REDRAW_NOW
#endif
```

thinkyhead, 2017-05-22:
See if #6830 heralds a return to normalcy.

Roxy-3D, 2017-05-23:
I saw the problem on a Graphical LCD Display.   I don't know that I've seen it on a 20x4 LCD Panel.

thinkyhead, 2017-05-23:
Graphical display would be the one that manifests the issue, if it is related to the choice of `LCDVIEW` flag.

edautz, 2017-05-23:
The screen is going complete blank, if this occurs with beginning of the print, the status screen is going complete blank. I use a lcd 20x4 char display.

Maglin, 2017-05-24:
I experience the same thing using a 20x4 GLCD.  On a AnyCubic TriGorilla controller.  I'm using the RAMPS 1.4 define for the MB.  The stock FW was using MB 33 which I think is RAMPS1.3.  Is there even a difference?  Could this be the cause?

Tannoo, 2017-05-25:
I see this now also after running the code from this morning.

renfrobr, 2017-05-25:
Same here using a 20x4 LCD. Screen blanks all over the place, any menu really. Touching the rotary encoder brings it back only to go blank again in the next screen/selection.

Roxy-3D, 2017-05-25:
So the problem is happening on both Graphical and 20x4 displays now???

bealparslan, 2017-05-26:
I made a video about this bug. Look the issue: #6856

gtavo, 2017-05-27:
I have the same problem, when trying to level the bed, the text vanishes, I have a REPRAP_DISCOUNT_SMART_CONTROLLER with 20x4

emartinez167, 2017-05-27:
This is now fixed with the latest patches.

renfrobr, 2017-05-27:
I can verify, it's fixed on my end

edautz, 2017-06-20:
Marlin upgrades to 1.1.3. Problem gone.

closed 2017-06-20 by edautz