MarlinFirmware/Marlin #6792: UBL Bugs

fr33dimension, 2017-05-19:
I run UBL successfully (Marlin 1.1) now for a few days and noticed the following.

- My start gcode raises the nozzle to 11mm and extrudes 10mm filament when temp is reached. With the UPL activated, It extrudes the 10mm way too fast, in terms of extrusion speed.

- when printing and the nozzle moves rapidly from one point to the other, the z axis makes a jerk up or down. This could be okay due to bed topology, but it looks and sounds kind of weird.

Roxy-3D, 2017-05-19:
How old is the code base that you are using?   An extruder and z movement bug were fixed a week and a half ago.   Your start gcode can specify the feedrate to use for the 10mm extrusion.   If nothing has been extruded yet, is it possible your default extrusion rate is too high?

fr33dimension, 2017-05-20:
Roxy, thanks good point about the speed, Just never had to do it and I use this start code since years. I downloaded 1.1 on May 12th, not sure how to check the code base date.

Roxy-3D, 2017-05-20:
Did your fade height go to 0.0 ?    The way the UBL State information is saved in EEPROM got changed.   And one thing I've noticed is if your have to do a M502 / M500 I lose my Fade Settings.   It seems to me it should default to 10.0  but it is defaulting to 0.0 until you set it.

fr33dimension, 2017-05-21:
Yes, noticed the same thing. And when it is set to 0.0, the extrusion speed is affected...weird. 

What's the command again to have the UBL activated at start up?

Roxy-3D, 2017-05-21:
Actually...  It is more of a 'state' than a command.   If you activate UBL with G29 A  and then do a M500 to save all the current settings...  It will power up in an active state.   You can do the opposite, G29 D and M500 to make it power up in a disabled state.

I'm going to be adding Z-Movement during the Mesh Editing today...  I'll try to get the Fade Height stuff to default to 10.0mm too...

fr33dimension, 2017-05-21:
Awesome! I installed it on a Ultimaker style machine with a proximity sensor and max z enstop, took me a while but it's working fine now.

fr33dimension, 2017-05-21:
I mentioned that earlier, but I noticed something like a z hop when using UBL, not sure what's causing it.

fr33dimension, 2017-05-24:
I am still seeing these weird z hops. Got UBL on two machines and both do it. I noticed when killing a print at the first layer I manually retract 10mm filament. When doing this with UBL active, it raises the z axis. I think ubl is actually reacting to retractions....is that it?

Roxy-3D, 2017-05-24:
I don't know...   Can you make a video of the Z-Hop ?   And maybe prior to doing that, turn on 'Debug Communications' in PronterFace so we know what lines of the GCode file are active.   Because if I can duplicate what you are seeing...   It is much easier to get it diagnosed and fixed.

fr33dimension, 2017-05-24:
I use repetier, the gcode is fine. When retracting in the 10mm of comp, it moves z up and back down when re-extruding. You don't experience that? Try that, home all, raise z 1mm, move x 70mm then retract 1mm and observe z. In the video you only see the last part of it, when i retract 1mm of filament, z moves. Can't attache it....anyway it's see if you can simulate it, doesn't make sense only I would see this.

Roxy-3D, 2017-05-24:
Do you have the firmware retraction turned on?

fr33dimension, 2017-05-24:
What do you mean? I am not aware of turning anything of that nature on, so no it's off unless default is on.

Roxy-3D, 2017-05-24:
Yes...  That is what I mean.  I never turn that on, so if that was on, that would explain why I don't see the problem.     Can I get you to do a 30 second video showing the problem.   That might give me a clue what is happening.

fr33dimension, 2017-05-24:
I will try but I couldn't upload the previous video, since the format was not supported. Zipped it, didn't work either...so if you have any other ideas let me know.

fr33dimension, 2017-05-24:
https://1drv.ms/v/s!AlouuPuoGfLkh6JEnOCwTtoNGq1QUA  see if you can view this. This is the earlier video.
LMK

Roxy-3D, 2017-05-24:
No.  I got to see your printer for 1/2 second.  And then a "There was a problem with this video" message appeared.

fr33dimension, 2017-05-24:
Well no, I am running 1.1.0 and that sounds very much like that....

fr33dimension, 2017-05-24:
What version should I be running? Thought 1.1 was it?

Roxy-3D, 2017-05-24:
the current bugfix_v1.1.0    But I'm trying to verify the bug fix is in there.   Several people have updated the file recently and I'm concerned that the bug fix got  removed...  Please give me a couple of minutes to resolve this.

fr33dimension, 2017-05-24:
NP thanks for the help.

fr33dimension, 2017-05-25:
Roxy,
That is it! That fixed that bug...question do other commands frequently change? After flashing the bugfix 1.1 the G29 O to display the mesh doesn't work anymore, used G29 T,  just curious.

Roxy-3D, 2017-05-25:
No...  The commands should not be changing much going forward.   The problem was G29 needed a lot of letters to do different things.   And some of those letters were 'Reserved' GCode letters.   People didn't think a 'G' was appropriate for Grid leveling.  They didn't want to see a G29 G command.   T was used for 3-Point leveling.  But that got combined with the various J modes and that freed up T to be used for Topographical Map.

fr33dimension, 2017-05-25:
Yes same here, no G29 in gcode, but needed to establish mesh. Thanks again....

closed 2017-05-27 by fr33dimension