MarlinFirmware/Marlin #6793: Question: Release schedule

ProfDrYoMan, 2017-05-19:
Hello,

are more frequent coordinated releases planned with the release of 1.1.0 (and 1.1.1) or should I start to use the bugfix-branch?

Since 1.1.0 it was a long time. I come from repetier and there "everybody" uses the dev-branch wich is close to never broken.

Roxy-3D, 2017-05-19:
Probably, you should use bugfix-v1.1.0   

>  I come from repetier and there "everybody" uses the dev-branch wich is close to never broken.

One way or the other...  We need the main branches to **_NEVER_** be broken.   There is no dispute on that.   The question is, what will it take to make that happen?

bgort, 2017-05-19:
I think there ought to be a 'must test before merging' guideline; merging something that just 'looks okay' should be frowned on.

Certainly bugs will always slip through here or there, but simple testing would have prevented most of the recent problems .. it would keep even the bugfix branch working 95+% of the time.

ProfDrYoMan, 2017-05-19:
Hmm, I see I started a discussion. :)

Roxy-3D, 2017-05-19:
>  it would keep even the bugfix branch working 95+% of the time.

If we can't keep careless, sloppy bugs from being merged...   I'm thinking I'll create "Yet Another Branch" labeled Rock-Solid-BugFix_v1.1.0     And in that branch, only code that has matured and proven itself for several weeks is eligible to be moved over to it and merged.   

I bet there are a lot of users that would be willing to help decide what 'fixes' are important enough and bug free enough to be moved over to it.   I would want to go even one step further...   THE USERS HAVE COMPLETE CONTROL OVER THE CONTENTS OF THAT BRANCH.   They get to decide what gets merged.

ProfDrYoMan, 2017-05-19:
http://nvie.com/posts/a-successful-git-branching-model/

What you talk about seems like hotfix branches.

Sticking to http://semver.org/ every hotfix might increase the third version number.

And yes. A way to separate feature/bugfix-branches and only get them into release if they are really stable seems a good way to go.

Don't know how to handle this with the lots of different hardware setup.
For me at first there needs to be a decent HAL-API.

bgort, 2017-05-19:
I agree, generally, that careless, sloppy bugs should have no place in code being used by users, but I'm not sure if creating another branch and the 'costs'/'overhead' associated with that -- merging into multiple branches, keeping track of what's where, etc. -- is worth it.  Also not sure how we'd implement a 'user approval' system, or if that could work.

Much better to just get everyone who can merge to test whatever code it is and state that they've tested a commit/PR -before- merging it, I think.  Maybe the way to do this is to configure the repo to require a review and approval from a member prior to allowing a merge?  I believe that's a configurable option....

bgort, 2017-05-19:
Good links, @ProfDrYoMan.  Thanks.

I actually posted the second one - the semver.org one - not too long ago.  We're pretty much using that system now, but I think we need to more clearly define what will increment the 'patch' number (the x in 1.1.x) .. and have a written/public plan for what the minor and major versions will include.  (That may exist, but it's only in @thinkyhead's head, as far as I know.)

Generally, I'd really like to see a written or graphic roadmap so it's easy to know what's going on, where things are headed, etc. .. even if it's just a basic/rudimentary one.  Maybe I'll take a stab at that when I have some free time, if others agree that that'd be useful.

ProfDrYoMan, 2017-05-19:
Glad I could help with the links. Using that model for my professional work as much as possible.

fiveangle, 2017-05-19:
The only way to maintain a truly stable "Release" is to have a set of identified tests on actual physical printers, and a set of individuals with those printers, and only after an RC has been actually signed off on by at least one of each of the required types, does the RC become R.  Anything less and it doesn't really matter what happens, so may as well just roll with the "*bugfix*"-du-jour wild west method :)

pmbroth, 2017-05-19:
I would keep it simple:  Stable Branch, RC Branch, Dev Branch

Maglin, 2017-05-20:
I'm with @pmbroth on this one.  If you want stable and bug free stick with the master stable branch.  Bugfix seems more like it should be called Dev.  And then once semi-throughly tested moved over to RC.  Does this mean RC will be bug free?  By no means can every scenario be tested by a small pool of people and machines, but it would make it easy to know what kind of code base you are putting onto your machine.  The devs would just have to ignore the "I NEED X FEATURE IN RC" comments because there will always be people that want bleeding edge but cry about any bugs.

That's my two cents.  It's probably not worth that much.  :)

closed 2018-02-06 by thinkyhead