MarlinFirmware/Marlin #6798: Delta M666 Issue

Gigglebear1, 2017-05-19:
Im attempting to manually level my bed by adjusting my endstops. I have enabled EEPOM in order to use the M500 M501 commands

i enter the following commands
M501  *EEPROM M666 X0 Y0 Z0
M666 X-5 Y-5 Z-5
M500
M501 *EEPROM M666 X0 Y0 Z0
M666 X-5
M500
M501 *EEPROM M666 X-5 Y0 Z0
M666 Y-6
M500
M501 *EEPROM M666 X-5 Y-6 Z0
M666 Z-4
M500
M501 *EEPROM M666 X-1 Y-2 Z0

I would assume that i can the last command M501 should be M666 X-5 Y-6 Z-4 but it seems as though z is being added to the x and y. This seems to be pretty consistent. I am not able to adjust each endstop individually. Am i doing something incorrect? Ive used the same procedure to calibrate my delta with the marlin v1.0.2 and it seems to work as expected. 

please help :)

Roxy-3D, 2017-05-19:
I don't use M666.   But to physically level the bed, one of the easiest and most effective ways is to bring up one of the Mesh based leveling systems, auto probe the bed...  And print out the Topology Map.   From that, it is very clear what has to happen.   And after the first iteration, do it a second time.  You should see the errors cut down in size and see what further refinement is necessary.

Gigglebear1, 2017-05-19:
i do not have a probe and im not able to get bed leveling to work(reference #6799). also i am not able to manually level my bed as it is fixed to the base platform (Delta printer)

Roxy-3D, 2017-05-19:
OK...    M666 is very small.   Most of its code is debug stuff.   I suggest we turn on the DEBUG_LEVELING_FEATURE in Configuration.h.  And then you activate it with M111 S32.  With that, we will easily get a lot more information.
```cpp
 /**
   * M666: Set delta endstop adjustment
   */
  inline void gcode_M666() {
    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) {
        SERIAL_ECHOLNPGM(">>> gcode_M666");
      }
    #endif
    LOOP_XYZ(i) {
      if (code_seen(axis_codes[i])) {
        endstop_adj[i] = code_value_linear_units();
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (DEBUGGING(LEVELING)) {
            SERIAL_ECHOPAIR("endstop_adj[", axis_codes[i]);
            SERIAL_ECHOLNPAIR("] = ", endstop_adj[i]);
          }
        #endif
      }
    }
    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING)) {
        SERIAL_ECHOLNPGM("<<< gcode_M666");
      }
    #endif
    // normalize endstops so all are <=0; set the residue to delta height
    const float z_temp = MAX3(endstop_adj[A_AXIS], endstop_adj[B_AXIS], endstop_adj[C_AXIS]);
    home_offset[Z_AXIS] -= z_temp;
    LOOP_XYZ(i) endstop_adj[i] -= z_temp;
  }
```

Gigglebear1, 2017-05-20:
thanks for the help. i enabled bed leveling debug and i get the following when I put 
M666 x-0.5 y-0.2 z-0.3

EEROM say
>>> gcode_M666
endstop_adj[X] = -0.50
endstop_adj[Y] = -0.20
endstop_adj[Z] = -0.30
<<< gcode_M666

yet when i M501 i get 
M666 X0.00 Y-0.10 Z-0.30

if I do M666 x-0.5 y-0.2 z-0.3
EEROM say
>>> gcode_M666
endstop_adj[X] = -0.50
endstop_adj[Y] = -0.20
endstop_adj[Z] = -0.30
<<< gcode_M666

the i M500 then m501 I get 
M666 X-0.30 Y-0.00 Z-0.10

Gigglebear1, 2017-05-20:
ITS normalize portion!!!!! it takes the extra and puts it on to the Z height. see a little counter  intuitive. i just have to also keep an eye on the Z height as well as the endstops.

ill post later on my findings

thanks for pointing that out @Roxy-3D

Maglin, 2017-05-20:
@Gigglebear1 I think you can probe manually with UBL.  I would get the same behavior as your with the LCD leveling.  Instead, choose fixed probe only.  Then run a UBL mesh prepare with G29 P2?  I'm about to try it out myself and see how it goes.  I just dropped my mesh down to 7x7 so as not to have to be touching off all night long.  My probe doesn't give me any good feelings inside when I view the mesh.  But it could be the interpolation that is causing me grief.

Roxy-3D, 2017-05-20:
> I just dropped my mesh down to 7x7 so as not to have to be touching off all night long. My probe doesn't give me any good feelings inside when I view the mesh. But it could be the interpolation that is causing me grief.

There is a 'U' option for G29 P1.    Instead of starting at the center point and working outward...  U is for an Unlevel bed.   It tries to probe the parameter far away from the starting point and away from other probed points.   So...  After just 5 or 10 probes...  You have enough of the perimeter probed to adjust the edges of your bed.

dbian, 2017-06-04:
You should use m665 hXXX together with m666.
PS, the manual leveling system is very useful, but now, it doesn't seem to work(reference #6799)

closed 2018-02-07 by thinkyhead