MarlinFirmware/Marlin #6799: manual bed leveling Delta

Gigglebear1, 2017-05-19:
using the Delta mini config 
i have the following #defines 

//#define USE_ZMIN_PLUG
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define Z_MIN_PROBE_ENDSTOP
//#define Z_MIN_PROBE_PIN Z_MAX_PIN  
#define PROBE_MANUALLY
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
#define DEBUG_LEVELING_FEATURE
#define LCD_BED_LEVELING

when i select  prepare > Level bed > Level Bed

it Will home and the LCD will say Homing XYZ then nothing....

if i click the knob the LCD will blank then if i hit the knob again it will say "moving..." but nothing will happen 

hitting the knob after this will have no affect 

do i have the incorrect setting or did i mess something up

Maglin, 2017-05-20:
Try UBL and define fix_probe with no offsets.  then do a G29 P2.  I think recently there was a gcode added to go to the next mesh point, but I can't remember it.  I'll have to look through all the comments that have been merged.

Roxy-3D, 2017-05-20:
> Retracted. G29 P2 won't do manual leveling.

Well...  It will...  But right now you need a real probe in the system.  But that constraint can be relaxed as soon as somebody as some time to do it.

povlhp, 2017-05-21:
Have the same issue on my MeCreator2. 
G29 works from gcode and moves nicely around and shows a table. M500 does not save the result. 
So multiple problems.

povlhp, 2017-05-23:
Looked at it. Seems like there is no code to actually do the moves in ultralcd. Will give it a shot, and see if I can get it working, and get the code back to the tree.
I have pins for the GT2560 (MeCreator2)  as well that I will create a pull request on soon.

Bob-the-Kuhn, 2017-05-24:
M500 won't save the UBL matrix/mesh.  G29 Sxx will save it to slot xx.  If no xx then it defaults to slot zero.

I've been using my LCD to do G29 P1, G29 P2 B, G29 P4 & G26.  It's under Prepare ..

povlhp, 2017-05-24:
But UBL seems to be sensor assisted levelling only ? We have the issue with pure manual leveling.
We get the nozzle to move to first probe position, then it just dies.

Tried to change      `_lcd_level_goto_next_point()`

Added the enqueue statement below, and now after adjusting Z at first position, it moves the 2nd position. But now it hangs there. I have not spent much time with the code, this was just a quick attempt to get it to move.  Looking around a bit more, it seems I need to define MESH_BED_LEVELING as well ? If so, this should be added to sanity checks (PROBE_MANUALLY without a method).

```
        #elif ENABLED(PROBE_MANUALLY)

          // Just wait for the G29 move to complete
          enqueue_and_echo_commands_P(PSTR("G29 S1"));
          lcd_synchronize();

        #endif

```

povlhp, 2017-05-24:
Well, just verified. I have MESH_BED_LEVELING enabled. 
Any ideas how to debug ? Should I just spread some` SERIAL_PROTOCOLLNPGM(x)`  in the code ? Been doing lots of print-debugging over the years.

Roxy-3D, 2017-05-24:
> M500 won't save the UBL matrix/mesh. G29 Sxx will save it to slot xx. If no xx then it defaults to slot zero.

bgort added this goodie...   You get to choose how this is handled.  But if enabled, an M500 will save the mesh to the current mesh slot.
```cpp
  // If this is defined, the currently active mesh will be saved in the
  // current slot on M500.
  #define UBL_SAVE_ACTIVE_ON_M500
```
>  If no xx then it defaults to slot zero.
```cpp
    if (code_seen('S')) {     // Store (or Save) Current Mesh Data
      g29_storage_slot = code_has_value() ? code_value_int() : state.storage_slot;
```
UBL remembers what mesh slot was last used for a load or save.  If not specified it defaults to that storage slot.

Bob-the-Kuhn, 2017-05-24:
I'm falling behind.  🙁

Roxy-3D, 2017-05-24:
> I'm falling behind.  :slightly_frowning_face:

No!   I almost didn't post that just because it comes off as a little bit critical.   It is impossible for one person to know even half of the details.   This is a huge, evolving fast code base.   It is actually the opposite.  In https://github.com/MarlinFirmware/Marlin/issues/6842#issuecomment-303812056  I raised an eye brow and said to myself "I've seen that in the code but never explored how to use it.  He is more in tune with things than me."

closed 2018-02-08 by thinkyhead