MarlinFirmware/Marlin #6800: homing issues

hardin162, 2017-05-19:
When I try to reverse the direction that the x direction is homing using --#define X_HOME_DIR 1-- (changed from -1 to 1) I am getting the following error 

Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
In file included from sketch\MarlinConfig.h:39:0,
                 from C:\Users\John\Desktop\New folder\standardMarlin\Marlin\Marlin.ino:37:
SanityCheck.h:847: error: #error "Enable USE_XMAX_PLUG when homing X to MAX."
     #error "Enable USE_XMAX_PLUG when homing X to MAX."
      ^
Multiple libraries were found for "U8glib.h"
 Used: C:\Users\John\Documents\Arduino\libraries\U8glib
 Not used: C:\Program Files (x86)\Arduino\libraries\U8glib
exit status 1
#error "Enable USE_XMAX_PLUG when homing X to MAX."

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

My z axis is moved through 2 m5 rods, they seem to move well when using Pronterface to move them but when trying to home the z axis they are not being supplied enough power to move and just make noise. Is there a way to increase the power it receives during homing. I tried to grease the rods and it did not help.

I would appreciate any help with either of the issues.

Maglin, 2017-05-20:
@hardin162 your issue is actually several.  Are you homing to Z min or Max?  That error is there to help you.  If you need to change your Z direction of travel you need to either change its direction in your configuration.h `#define INVERT_Z_DIR true`.  Whatever side of the axis your endstop is at is where you have to home it.ie.. if at the bottom of the Z then you use Z_MIN but it it's at top you use Z_MAX endstop pins.  You only change your homing direction to be what pins you are using.  And also to ensure you don't reverse your axis.  For the power, you most likely will have to adjust your driver's potentiometer which you can google for you specific controller board.  Also, you need to be able to easily move the leadscrews by hand without power supplied to the board.  If you can't then you need to fix the mechanical bind the axis or leadscrew is in.

hardin162, 2017-05-20:
Thanks. I was assuming that the printer would want to follow the right hand rule for the positive axes, but I changed the direction and it printed text so that was my bad. Turning up the pots worked, I just added an extra heat sink  and I have not had any thermal cuttofs thank you for the help.

closed 2017-08-22 by hardin162