MarlinFirmware/Marlin #6802: G33 isn't working

Maglin, 2017-05-20:
Just uploaded newest FW.  Now when I G33 I get the following.  Seems no matter what I do I get a massive std dev where the same setting would get me around 0.040 std dev.  Here is the output not that it helps but it seems like it's got broken somewhere.

```
>>>G33
SENDING:G33
echo:SD card ok
echo:Active Extruder: 0
echo:SD card ok
G33 Auto Calibrate
Checking... AC
.Height:318.83    Ex:-0.67  Ey:-0.45  Ez:+0.00    Radius:98.07
.Tower angle :    Tx:-0.73  Ty:-0.13  Tz:+0.00
Iteration : 01                                    std dev:-2147483.750
.Height:0.00    Ex:0.00  Ey:0.00  Ez:0.00    Radius:0.00
.Tower angle :    Tx:0.00  Ty:0.00  Tz:+0.00
Calibration OK                                    rolling back.
.Height:318.83    Ex:-0.67  Ey:-0.45  Ez:+0.00    Radius:98.07
.Tower angle :    Tx:-0.73  Ty:-0.13  Tz:+0.00
Save with M500 and/or copy to Configuration.h
```

Maglin, 2017-05-20:
Did a bunch of different testing.  I had changed my probe.  It's a mechanical probe that I install.  I had taken the lever off for better accuracy and added it's new offsets.  With the new offsets I would get that crazy deviation.  So here is what my offsets where and what they are now that works, but the probe is not under the nozzle atm.

```
//#define X_PROBE_OFFSET_FROM_EXTRUDER 20.0     // Z probe to nozzle X offset: -left  +right
//#define Y_PROBE_OFFSET_FROM_EXTRUDER -10.0     // Z probe to nozzle Y offset: -front +behind
//#define Z_PROBE_OFFSET_FROM_EXTRUDER -18.9     // Z probe to nozzle Z offset: -below (always!)

#define X_PROBE_OFFSET_FROM_EXTRUDER 0     // Z probe to nozzle X offset: -left  +right
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0     // Z probe to nozzle Y offset: -front +behind
#define Z_PROBE_OFFSET_FROM_EXTRUDER -18.9     // Z probe to nozzle Z offset: -below (always!)
```

MagoKimbra, 2017-05-20:
For @thinkyhead . The problem is with the new controls you put in position_is_reachable_raw_xy.
In that case, the result at some points is always false because the point calculating Y offset probe is outside.
In fact he had to put DELTA_CALIBRATION_RADIUS = DELTA_PRINTABLE_RADIUS - 20, where 20 is precisely the Y offset of the probe.
In my opinion, it could be resolved by becoming DELTA_CALIBRATION_RADIUS = DELTA_PRINTABLE_RADIUS - max (abs (X_PROBE_OFFSET_FROM_EXTRUDER), abs (Y_PROBE_OFFSET_FROM_EXTRUDER))
This way the probe points are always inside.

thinkyhead, 2017-05-23:
`position_is_reachable_raw_xy` ain't mine! Someone else adapted that from code that I had written for a branch elsewhere. I will see if I can fix it up. I seem to be spending most of my time cleaning up after others instead of working on the high level design stuff that so needs to be done.

thinkyhead, 2017-05-23:
@LVD-AC This looks like one for you.

LVD-AC, 2017-05-23:
Could you give the Bugfix branche a try and report back? There's already some fixes included there that could resolve that problem.

Of couse one of the disadvantages of having an eccentric probe is it reduces the radius the probe can reach without bumping into the towers by sqrt(sq(X_PROBE_OFFSET_FROM_EXTRUDER) + sq(Y_PROBE_OFFSET_FROM_EXTRUDER))

so you could change this setting to:

`#define DELTA_CALIBRATION_RADIUS ((DELTA_PRINTABLE_RADIUS) * 0.869 - sqrt(sq(X_PROBE_OFFSET_FROM_EXTRUDER) + sq(Y_PROBE_OFFSET_FROM_EXTRUDER)))`

PS: you can change the DELTA_CALIBRATION_RADIUS by `M665 Bx.xx `until you got it right and then hardcode it into Configuration.h.

bsabino, 2017-05-23:
I'm currently on the bugfix branch and executed G33:
echo:busy: processing
.      c:-0.88     x:-0.89   y:-0.90   z:-0.90
.                 yz:-0.90  zx:-0.90  xy:-0.89
Iteration : 01                                    std dev:0.894
.Height:304.19    Ex:+0.00  Ey:-0.01  Ez:-0.01    Radius:132.73

.      c:+0.08     x:+0.09   y:+0.08   z:+0.08
.                 yz:+0.08  zx:+0.08  xy:+0.09
Iteration : 02                                    std dev:0.083
.Height:304.10    Ex:+0.00  Ey:-0.01  Ez:-0.02    Radius:132.73

.      c:-0.00     x:-0.01   y:-0.01   z:-0.01
.                 yz:-0.01  zx:-0.01  xy:-0.01
Iteration : 03                                    std dev:0.009
.Height:304.10    Ex:+0.00  Ey:-0.01  Ez:-0.02    Radius:132.54

.      c:-0.01     x:-0.01   y:-0.01   z:-0.04
.                 yz:-0.03  zx:-0.04  xy:-0.01
Iteration : 04                                    std dev:0.027
.Height:304.10    Ex:-0.01  Ey:+0.00  Ez:-0.05    Radius:132.34

.      c:-0.02     x:-0.02   y:-0.02   z:-0.02
.                 yz:-0.02  zx:-0.02  xy:-0.02
Calibration OK                                    rolling back.
.Height:304.10    Ex:+0.00  Ey:-0.01  Ez:-0.02    Radius:132.54
Save with M500 and/or copy to Configuration.h
echo:busy: processing
echo:busy: processing

What I do not understand is how it get's de radius of 132.54 as my bed is definitly not bigger than 240 in diameter. So when saving it tries to print outside of the bed.. :)

LVD-AC, 2017-05-23:
The radius in the calibration routine is the delta radius, this is the horizontal distance of the diagonal rods; nothing to do with the printing limits and possible printing outside the bed which is set in the FW by PRINTABLE_RADIUS.

oldmcg, 2017-05-24:
I added position_is_reachable_raw_xy -- it prevents delta nozzle from going outside DELTA_PRINTABLE_RADIUS which can bind mechanics or bend rods.  There is also a _by_probe variant that for deltas makes sure that BOTH the nozzle and the probe don't move outside DELTA_PRINTABLE_RADIUS -- the nozzle so it doesn't exceed mechanical limits and the probe so it doesn't fall off the bed.  For delta probes with x/y offset from nozzle this makes the true probeable area on a delta look like a circle minus a crescent (on the side opposite the probe's x/y offset).

bsabino, 2017-05-24:
When running G33 I get the error regarding the radius: ?(M665 B)ed radius is implausible. For which I have the folowing in my config.h (What do I miss?):

  // Make delta curves from many straight lines (linear interpolation).
  // This is a trade-off between visible corners (not enough segments)
  // and processor overload (too many expensive sqrt calls).
  #define DELTA_SEGMENTS_PER_SECOND 100

  // Center-to-center distance of the holes in the diagonal push rods.
  #define DELTA_DIAGONAL_ROD 271.5 // mm 265.0

  // Horizontal offset from middle of printer to smooth rod center.
  #define DELTA_SMOOTH_ROD_OFFSET 186 // mm 198.0

  // Horizontal offset of the universal joints on the end effector.
  #define DELTA_EFFECTOR_OFFSET 31.0 // mm 30

  // Horizontal offset of the universal joints on the carriages.
  #define DELTA_CARRIAGE_OFFSET 20.6 // mm 35.0 

  // Horizontal distance bridged by diagonal push rods when effector is centered.
  #define DELTA_RADIUS (DELTA_SMOOTH_ROD_OFFSET - DELTA_EFFECTOR_OFFSET - DELTA_CARRIAGE_OFFSET) //mm  Get this value from auto calibrate

  // height from z=0.00 to home position
  #define DELTA_HEIGHT 303.20 // get this value from auto calibrate - use G33 P1 at 1st time calibration

  // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
  #define DELTA_PRINTABLE_RADIUS 116.0

  // Delta calibration menu
  // See http://minow.blogspot.com/index.html#4918805519571907051
  #define DELTA_CALIBRATION_MENU

  // set the radius for the calibration probe points - max DELTA_PRINTABLE_RADIUS*0.869 if DELTA_AUTO_CALIBRATION enabled
  #define DELTA_CALIBRATION_RADIUS ((DELTA_PRINTABLE_RADIUS) * 0.869) // mm

  // G33 Delta Auto-Calibration (Enable EEPROM_SETTINGS to store results)
  #define DELTA_AUTO_CALIBRATION
  #if ENABLED(DELTA_AUTO_CALIBRATION)
    #define DELTA_CALIBRATION_DEFAULT_POINTS 7 // set the default number of probe points : n*n (-7 -> +7)
  #endif

  // After homing move down to a height where XY movement is unconstrained
  #define DELTA_HOME_TO_SAFE_ZONE

Maglin, 2017-05-24:
Maybe try without `#define DELTA_HOME_TO_SAFE_ZONE` .  I haven't recompiled today but that is the only thing different from mine to yours.  Well, that and I set `#define DELTA_RADIUS` to its value after an auto calibrate.

LVD-AC, 2017-05-24:
Do not confuse between:
-the delta radius (M665 Rx.xx), this is the horizontal distance of the diagonal rods; 
-calibration radius (M665 Bx.xx), this the radius the probe can reach without bumping into the towers;
-printable radius, this is the radius the nozzle can reach without bumping into the towers.

When you set the calibration radius to high you get the` "?(M665 B)ed radius is implausible"` error.

oldmcg, 2017-05-25:
@LVD-AC, could you please explain the significance of 0.869 in scaling the printable radius to the calibration radius?  Can it not use full printable radius minus HYPOT(probe_off_x,y)?  Or is it trying to use some value closer to the rod delta radius?  Just trying to understand.

Thx.

LVD-AC, 2017-05-25:
The denser probe grids use different radius to probe:
P5: 0.95 - 1.05 * bed radius
P6: 0.9 - 1 - 1.1 * bed radius
P7: 0.85 - 0.95 - 1.05 - 1.15 * bed radius (hence the recommended limit of 1/1.15 = 0.869)

You can use G33 P4 up to scaling 1 (minus HYPOT(probe_off_x,y)), but only for the probe grids up to P4;

LVD-AC, 2017-06-08:
Issue may be closed now; solved in #6850