MarlinFirmware/Marlin #6803: 1284P with 1.1.1 restart loop

fred2088, 2017-05-20:
1284P with 1.1.1 release 
1. work fine by just define the board and TinyBoy2 - take up 72% of the program mem, 28% of global variable
2. restart loop when enabled SDSUPPORT - take up 77% of the program mem, 28% of global variable

Bob-the-Kuhn, 2017-05-20:
Please ZIP up your FAILING configuration files & post them.  I'll see if I can reproduce it on a Sanguino.

fred2088, 2017-05-20:
#define MOTHERBOARD BOARD_MELZI
 #define OLED_PANEL_TINYBOY2
#define SDSUPPORT

only changed 3 lines

Bob-the-Kuhn, 2017-05-20:
I'm getting a boot loop also with 1.1.1 but not when using bugfix-1.1.x.

Please switch to [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x )and see if that fixes the problem.

thinkyhead, 2017-05-20:
@Bob-the-Kuhn I'll do a 1.1.2 release later today, assuming we don't have any serious regressions.

Bob-the-Kuhn, 2017-05-21:
@fred2088 - better hold off on switching.  There's been reports on intermittent head crashes with the latest bugfix-1.1.x

fred2088, 2017-06-01:
still got restart loop on 1284p with 1.1.2 
if enabled both
#define sdsupport
#define U8GLIB_SSD1306
or
#define sdsupport
#define OLED_PANEL_TINYBOY2

Bob-the-Kuhn, 2017-06-02:
I forgot about this.  Sorry

Please try bugfix-1.1.x.

fred2088, 2017-06-13:
oh.! sorry, tried bugfix-1.1.x also got the same restart loop error !!!

fred2088, 2017-06-13:
either 
 #define sdsupport
 or
 #define OLED_PANEL_TINYBOY2
can work, but not both at the same time on bugfix 1.1.x and 1.1.2

Bob-the-Kuhn, 2017-06-14:
I tried changing just the three lines on today's bugfix-1.1.x and didn't have any problems.

Please post a copy of what the host screen/log shows from a power up through 2 or 3 restarts.

fred2088, 2017-06-14:
my bugfix.1.1.x is upto 14/5/2017, tried the lastest bugfix.1.1.x work fine

thanks, will try it out upto bugfix.1.1.x

fred2088, 2017-06-15:
fixed by comment the following line from pin_Sanguino11.H on 1.1.2, 1.1.3, bugfix1.1.x
  #elif ENABLED(OLED_PANEL_TINYBOY2)  || ENABLED(U8GLIB_SH1106) || ENABLED(U8GLIB_SSD1306)
    #define BTN_ENC             28
//    #define LCD_SDSS            -1

Bob-the-Kuhn, 2017-06-15:
I can't find that line in any of the pins_s*.* files.

fred2088, 2017-06-16:
it's already deleted from 1.1.2, 1.1.3 and the lastest bugfix.1.1x
while the code from this section will work fine with comment out #define LCD_SDSS -1 ...

Bob-the-Kuhn, 2017-06-16:
That's strange.  Usually a -1 will result in it being ignored.

closed 2017-06-20 by fred2088