MarlinFirmware/Marlin #6804: Typo in ubl_g29.cpp

christianh17, 2017-05-20:
Hi!

I just spent some time trying to understand ubl and x / y-areas. Unfortunately it is still not clear to me but I found a small typo in ubl_g29.cpp.
The line (139 in my last checkout):
`In Cartesian printers, mesh points within the X_OFFSET_FROM_EXTRUDER and Y_OFFSET_FROM_EXTRUDER`
should be:
`In Cartesian printers, mesh points within the X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER`
This helps with copy and search inside this file. 
Hope it's OK to write little things like this here, if not: please tell me!

Kind regards,
Christian

Roxy-3D, 2017-05-20:
Thanks...   I'll try to remember to get this updated the next time I do a push...

Bob-the-Kuhn, 2017-05-20:
There's a small graphic at the end of this [wiki page](http://marlinfw.org/docs/features/unified_bed_leveling.html) that explains about probeable areas.

christianh17, 2017-05-20:
I read this page so many times but still did not understand it completely. I have quite a big distance between my first extruder and the z-probe (X45/Y32). So I wanted to understand what I have to fill in UBL_MESH_MAX_X and ..Y (and of course MIN). Is it the position of the extruder or of the probe?

Roxy-3D, 2017-05-20:
When you do a G29 T you get a mesh map.  The .'s are places the probe is unable to get to.   So we need to fill those in one way or the other.   If you do a G29 P3 T   twice...   You will probably get all the values filled in.   They won't be highly accurate.   But they will be good enough to start fine tuning your mesh and getting it perfect.

Bob-the-Kuhn, 2017-05-20:
> Is it the position of the extruder or of the probe?

The coordinates are always for the nozzle EXCEPT when using the probe during G29 P1.  

Lets say your mesh locations are (10,10), (20,20), (30,30) & (40,40) and that the probe can only reach the first two.  During G29 P1 the **NOZZLE** is positioned so that the **PROBE** is at (10,10) and then (20,20).  During G29 P2 the **NOZZLE** is positioned at  (30,30) and then (40,40).  This way the mesh is populated with Z values obtained at the points.

christianh17, 2017-05-21:
Thanks for the offer helping me with my mesh.
My G29 T after one G29 P3 wich filled the left line:

Recv: (10,232)                                             (237,232)
Recv:  -0.063   -0.063    0.002    0.020    0.010   -0.050   -0.090
Recv:  -0.080   -0.080   -0.018    0.025    0.005   -0.090   -0.155
Recv:  -0.070   -0.070   -0.005    0.007   -0.013   -0.068   -0.163
Recv:  -0.088   -0.088   -0.028   -0.008   -0.023   -0.083   -0.155
Recv:  -0.120   -0.120   -0.063   -0.023   -0.038   -0.090   -0.170
Recv:  -0.145   -0.145   -0.098   -0.063   -0.070   -0.115   -0.168
Recv: [-0.200]  -0.200   -0.118   -0.093   -0.075   -0.138    0.007
Recv: (10,36)                                                 (237,36)

The resulting test print shows a print like on a tilted bed (on the left too low, on the right too high). I stopped the printing of the test pattern because on the right hand side it touched always the bed while printing. And: Please do noat have too close look at my capton - it definitely needs to be replaced. I had some trouble with my firmware settings... But: At first I want to have the ubl done, than I will replace the capton.
[image]

What shall I do? I have no idea- I even tilted the bed manually (of course before doing a new G29 P1) but the test print shows all the time the same result.

And: Regarding the configuration there are some questions left. I tried to understand the coding and tried a lot of the settings.

- In Conditionals_post.h the (X/Y/Z)_PROBE_OFFSET_FROM_EXTRUDER settings are set to 0. Why? This changes the logic afterwards.
- My x-probe is 45mm right of the first nozzle. The printing area starts at 28mm. So of course I can't probe till the left end of the bed. But: I do not know what to enter in UBL_MESH_MIN_X. Via try and error I ended with "4", this led to a measurement of the far left row at x=2 (seen in the graphics display). But: The test print printed on the left side "in the air". It does not really matter but I am sure this is not the intended way to use it.

So: I am confused while trying to configure my firmware.

It would be very kind if you could give me a hint where to search.
Regards
Christian

PS: Another screenie regarding the deactivation of the offset settings in Conditionals_post.h
[image]

Bob-the-Kuhn, 2017-05-21:
> UBL_MESH_MAX_X and ..Y (and of course MIN). 

These are the coordinates of where the NOZZLE can reach while printing.

If the G26 command is printing where you want it to print then you have these correct.

----

The highlighted items only happen if you do NOT have a probe.  They are NOT happening in your case.

----

Now that you have a G26 print you can start the fine tuning process.

In your case I suggest the following:
1) send **G29 P6 C0.15**  This will raise everything by 0.15mm 
2) do G26 again
3) send G29 P4 R49 to manually adjust every point in the mesh.  In my experience adjustments of about 0.050mm is best at this point.  Only use 0.10mm adjustments if the nozzle is scraping the bed or is printing in the air.
4) save your mesh
5) repeat steps 2, 3 and 4 as needed.

thinkyhead, 2017-05-21:
I've patched the typo.

closed 2017-05-21 by thinkyhead