MarlinFirmware/Marlin #6805: Floating point temperatures

Roxy-3D, 2017-05-20:
People seem to really want the decimal place back on the temperature displays.

http://forums.reprap.org/read.php?415,769199

Maybe we should change that back.  Except change the int to a double instead of a float ????

Grogyan, 2017-05-20:
Why double? A float should be fine.

I too would like to see the return of the decimal point precision of the
temperatures.

Roxy-3D, 2017-05-20:
https://github.com/MarlinFirmware/Marlin/issues/6728#issuecomment-301285735

teemuatlut, 2017-05-20:
All that's needed is to change degHotend (and degBed) to float and as far as I can tell, it is not called by an ISR. The temperature itself is already stored as float. The original PR doesn't explain why the change was made.

[image]

Roxy-3D, 2017-05-20:
What are the steps to get GitHub to generate a picture like that for other variables?

teemuatlut, 2017-05-20:
It's generated by Doxygen (and graphviz). While Marlin doesn't yet officially use it and provide it somewhere, it already has adopted the proper comment styling for Doxygen to work.

thinkyhead, 2017-05-20:
I recently changed all temperature handling code that didn't need `float` to use `int` — for performance and size. So don't just change it all back, please.

Just coerce the temperatures to `float` when they're printed or if they're being converted from sensor values, do that as `float`, but only for display. In most places temperatures can and should be handled as integers.

thinkyhead, 2017-05-20:
I'll review my changes and see where these temperatures with decimals were coming from, and I should have a patch pretty soon.

teemuatlut, 2017-05-21:
> where these temperatures with decimals were coming from

degHotend() - All this method is is a getter for "current_temperature[HOTEND_INDEX]". The variable already is a float and the method just returns it as int16.

> Just coerce the temperatures to float when they're printed

Can't just use type conversion if the decimal information is already lost in the method above. The easiest would be to have the M105 and LCD to call for the member variable directly, since that it made public by the class. Which I actually find odd, usually class variables are private and accessed by getters/setters, yes?

thinkyhead, 2017-05-21:
The getter has always been that way. Code that displayed the value as float in the past was merely bypassing the getter.

teemuatlut, 2017-05-21:
It's returning float on all the 1.1.0-RCx tags.

thinkyhead, 2017-05-21:
You're right, I'm mis-remembering. Anyway, I have a patch pending that should clear up this issue.

thinkyhead, 2017-05-21:
#6811 should have it covered.

closed 2017-05-21 by thinkyhead