MarlinFirmware/Marlin #6806: Help - Can't get Thermistor to work correctly?

coogle, 2017-05-20:
Hello, 

I'm hoping someone can help me - I'm having a heck of a time getting my extruder thermistor replaced. Due to a failed print that got out of control I broke the old one but thankfully I have a stack of these guys laying around:

https://www.amazon.com/gp/product/B016XGX3X0/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

According to them I should be compiling Marlin to use temp table 11 (TEMP_SENSOR_0 = 11). When I do however, I don't get the correct reading from the thermistor -- It reads 81 deg at room temp (where the heated bed reads 22).

I'm at a pretty serious loss here at this point, I've tried multiple different temp tables and I can't seem to find it. I've even tried a second thermistor to make sure I didn't have a bad one. I'm running Marlin 1.1.1 on a Anet v1.0 board (1284p). 

Suggestions?

Roxy-3D, 2017-05-20:
What type number was your old one?   Was it a 4.7K type number?  Because if it was a 1K type number, that would explain why it isn't working....

coogle, 2017-05-20:
Unfortunately I don't know. It was originally was this kit I picked up just to have a 2nd printer:

http://www.gearbest.com/3d-printers-3d-printer-kits/pp_343643.html

The old one was thrown away by mistake, so I can't say for certain anymore. What I can say is they sell this part separately so I can only hope/assume it's their replacement:

http://www.gearbest.com/3d-printer-parts/pp_228653.html

FWIW I did try table #51 as well without any luck. Perhaps I should try 52 and 55 as well?

Roxy-3D, 2017-05-20:
My guess is 52 won't work because that is a 200K thermistor.   But 55 would be worth a try.

Bob-the-Kuhn, 2017-05-20:
Maybe they're not 100K thermistors.  What is the resistance of the thermocouple at room temperature?

coogle, 2017-05-20:
I'll try 55 - I measured 90k ohm at room temp.

coogle, 2017-05-20:
No dice. (I meant 51 btw not 52)

Also I think you have a bug in your build process in thermistortables.h:92. For number 55 you include the file thermistortable_53.h (which doesn't exist)

Any other suggestions? I'm trying a different table now..

Roxy-3D, 2017-05-20:
```cpp
#if ANY_THERMISTOR_IS(52) // 200k ATC Semitec 204GT-2 (WITH 1kohm RESISTOR FOR PULLUP, R9 ON SANGUINOLOLU! NOT FOR 4.7kohm PULLUP! THIS IS NOT NORMAL!)
  #include "thermistortable_52.h"
#endif
#if ANY_THERMISTOR_IS(55) // 100k ATC Semitec 104GT-2 (Used on ParCan) (WITH 1kohm RESISTOR FOR PULLUP, R9 ON SANGUINOLOLU! NOT FOR 4.7kohm PULLUP! THIS IS NOT NORMAL!)
  _**#include "thermistortable_53.h"**_
#endif
#if ANY_THERMISTOR_IS(60) // Maker's Tool Works Kapton Bed Thermistor
  #include "thermistortable_60.h"
#endif
#if ANY_THERMISTOR_IS(66) // DyzeDesign 500°C Thermistor
  #include "thermistortable_66.h"
#endif
```

Roxy-3D, 2017-05-20:
No...   Mostly...  If you find a table that 'works', be careful and don't assume the numbers are accurate.

coogle, 2017-05-20:
Well I'm gonna quickly run out of tables here :) But I'll keep trying...

Bob-the-Kuhn, 2017-05-20:
Have you tried swapping the cables between the heated bed and the extruder thermal inputs?

I'm wondering if there is a cable issue or if you have a bad input.

I calculate that a 90k probe resistance would need about a 33k pull up resistor in order to read 80C using table 11.

coogle, 2017-05-20:
I can certainly swap the thermistors and give that a shot and see if there is something else wrong here. Good idea. Stay tuned.

coogle, 2017-05-20:
Looks like that's where the issue is. The known working heated bed thermistor is not working in the extruder plug. So it looks like the problem is my mainboard (ugh). Thanks for the suggestion that helped! Glad I found a bug at least along the way :)

closed 2018-02-06 by thinkyhead